esp8266/Arduino #921: Question about output of ESP.getChipId()?

EnvironmentMonitor, 2015-10-22:
Hi, 
when this is run it returns the last three portions of the mac address is this correct?

Serial.printf("  ESP8266 Chip id = %08X\n", ESP.getChipId());

Using ver. 1.6.5-1160-gef26c5f

Thanks....

igrr, 2015-10-22:
This is correct. By default, ESP sets its MAC address as a concatenation of
3 fixed bytes and 3 unique bytes burned into chip (which are read by
getChipId function).

EnvironmentMonitor, 2015-10-22:
Many thanks, I was trying to see if a version number could be returned,
maybe determine from that the -01,07 etc.

Thanks again......

closed 2015-10-22 by igrr

whyameye, 2016-07-21:
Sorry to reopen. I'm wondering about this statement: "ESP sets its MAC address as a concatenation of 3 fixed bytes and 3 unique bytes." I think you mean that the 1st 3 bytes of the MAC address are the same between all 8266s. However, looking at the MAC addresses of my 12Es, some have 1st 3 bytes 18:FE:34 and some have first 3 bytes are 5C:CF:7F. Moreover I just purchased a 12F and its 1st 3 bytes are 60:01:94.

I'm concerned about this because I have been using getChipId as a unique identifier for my 8266s. But now that I see that the 1st 3 bytes of the MAC address are perhaps not fixed after all, I wonder if it is possible that the last 3 bytes may not be unique between different runs of 12Es or between the 12E and 12F. Thoughts on this?

romelemperado, 2018-08-06:
Hi @igrr 

> ESP.getChipId() returns not related to the MAC ID of my NodMCU

My mac is: 84:F*:E*:52:F1:6C but it returns ESP5435756

duncan-a, 2018-08-06:
I've also noticed the three first bytes being different, wondered about the
significance and also questioned whether the last three bytes are, in fact,
truly unique across all ESP8266 devices or only in relation to the
groupings given by the first three bytes.

I think it would be very helpful to have an explanation of those first
three bytes and on the reality of the uniqueness of the last three bytes.

majenkotech, 2018-08-13:
The first three bytes are the OUI and are allocated to Espressif by IEEE. They currently have 23 allocated:

```
18-FE-34   (hex)                Espressif Inc.
24-0A-C4   (hex)                Espressif Inc.
24-B2-DE   (hex)                Espressif Inc.
2C-3A-E8   (hex)                Espressif Inc.
30-AE-A4   (hex)                Espressif Inc.
3C-71-BF   (hex)                Espressif Inc.
54-5A-A6   (hex)                Espressif Inc.
5C-CF-7F   (hex)                Espressif Inc.
60-01-94   (hex)                Espressif Inc.
68-C6-3A   (hex)                Espressif Inc.
84-0D-8E   (hex)                Espressif Inc.
84-F3-EB   (hex)                Espressif Inc.  <-- recognise it?
90-97-D5   (hex)                Espressif Inc.
A0-20-A6   (hex)                Espressif Inc.
A4-7B-9D   (hex)                Espressif Inc.
AC-D0-74   (hex)                Espressif Inc.
B4-E6-2D   (hex)                Espressif Inc.
BC-DD-C2   (hex)                Espressif Inc.
C4-4F-33   (hex)                Espressif Inc.
CC-50-E3   (hex)                Espressif Inc.
D8-A0-1D   (hex)                Espressif Inc.
DC-4F-22   (hex)                Espressif Inc.
EC-FA-BC   (hex)                Espressif Inc.
```

Given that the last three digits are supposedly "unique", that gives 256*256*256*28 possible MAC addresses Espressif have available to them currently.  That's 469,762,048 possible MAC addresses. These are across their entire range of products, of course, and some may be only used internally for testing, but it's already a large number. As they run out they can request another allocation from IEEE.

It's up to them if they do make them globally unique, though. Ideally, they should.

The full list of OUI allocations is available from http://standards-oui.ieee.org/oui.txt

YoNevelt, 2021-01-17:
I've got an ESP-WROVER-B (16MB flash) and this function gives me the MAC backwards by 2 digits:

- MAC: 40:F5:20:xx:xx:xx
- ChipID: xx:xx:xx:20:F5:40

By the way, I've seen a comment here, the Espressif Inc. has extended the registered MAC list :)