esp8266/Arduino #923: Purposes of high values for HTTP_MAX_DATA_WAIT and HTTP_MAX_CLOSE_WAIT in webserver

jpswensen, 2015-10-23:
I have noticed that when serving things like css and js files that the browser seems to hold connections open and thus it takes the full passing of HTTP_MAX_DATA_WAIT and HTTP_MAX_CLOSE_WAIT for the next file to be processed.

I was able to knock both these values down to 1 ms and everything seems to be working fine and I now have a blazing fast HTTP server. Can you explain:
1) why these waits were in there,
2) why they were set to high values, and
3) was my hack just an accident waiting to happen?

igrr, 2015-10-23:
1) These waits are an attempt to avoid actively closing TCP connection on the ESP side.
2) Values are pretty arbitrary. Ideally these should be turned into configurable parameters, so that user can set them when instantiating web server. 
3) It seems that in your case browser is not closing connections first as instructed by `Connection: close` header. This may be because there is some issue with HTTP headers (like duplicate headers), or Content-Length header is not being set to correct value. If you provide the source code, I can take a look.
Whether the change to timeouts you did is going to affect anything depends on the cadence at which requests arrive. When server actively closes TCP connection, the connection enters TIME_WAIT state and its protocol control block is kept in memory for 2*MSL period (which is 120 seconds). When this timeout expires, memory used by TCP PCB is freed. So if requests come in too fast (faster than `total_ram / sizeof(tcp_pcb) / 120` requests per second), ESP will run out of RAM.

So in general, we choose to increase latency by leaving it to the client to close the connection, and being able to free memory immediately without dropping into TIME_WAIT state.

My suggestion here is to figure out why in your setup the browser doesn't close the connection first.

jpswensen, 2015-10-23:
Thanks for the quick and very complete description. You can see the code I am running at 
https://github.com/jpswensen/ESP8266_WebGen
The sketch in AdvancedWebServerHuzzah should show the behavior I am talking about. It very well could be the way I implemented my sendBinaryFile so that I could write from PROGMEM. I did just notice now that the most recent from Github has a send_P function which makes my function unnecessary. Le me see if this solves my problem before you waste your time hunting down my problems.

P.S. when I started building using the latest from github vs the previous release, both my flash and RAM utilization dropped by about 10-15%. Nice!!! and Thanks!

igrr, 2015-10-23:
There is an issue indeed.
https://github.com/jpswensen/ESP8266_WebGen/blob/master/AdvancedWebServerHuzzah/WebFramework.cpp#L10
You are not giving any indication of content size, so the client doesn't know when all the data comes in and it is okay to close the connection.
Use `server.send_P` function if you're using the latest version from github, or call `server.setContentLength(realLen)` before calling `server.send(200, mimeType, "")`.

jpswensen, 2015-10-23:
Both of those solutions worked great. Thanks again for the quick replies!

closed 2015-10-23 by jpswensen