esp8266/Arduino #925: WiFi - ESP8266 - SX1276 - LoRa

grahamehorner, 2015-10-23:
has anyone see the https://www.lora-alliance.org/ it looks a perfect fit for IoT and using the ESP8266 as a bridging device between LoRa and WiFi see  https://hackaday.io/project/4758-sentrifarm/log/20300-brief-update-esp8266-plus-lora and If the community here is interested I would be willing to start R&D and produce a library and circuit for that purpose.

hallard, 2015-10-26:
Excellent ;-)

I had in my mind to update the [Gateway](https://github.com/hallard/NodeMCU-Gateway) hardware to support these modules

grahamehorner, 2015-10-26:
@hallard cool maybe we could work together on this; I'm not really a hardware person, but would love to get a LoRa arduino library working specifically with ESP8266.

BTW have you seen http://www.getfreevolt.com I'm looking/hoping to get some development units for testing; in hope that it can power/charge batteries for in the field IoT devices that use ESP8266 and/or SX1276/LoRa for long range communication.

hallard, 2015-10-26:
@grahamehorner 
Yes, no problem, I've just looked at freevolt, seems impressive ;-) Not sure it would be able to power Wifi module forever, these one are really consuming ...

marcobrianza, 2016-07-13:
Hello, I am testing a LoRa node on  www.thethingsnetwork.org in Milano with an RF95 radio and this stack:
https://github.com/matthijskooijman/arduino-lmic/blob/non-avr-printf/examples/ttn/ttn.ino
It builds correctly on the Arduino cores, but not on the ESP8266 :-(
If has these errors:

/var/folders/xt/636gqx4x4hz49yz9xgg79tkw0000gn/T/buildaf3861b8e88663e57f7270134668d526.tmp/libraries/arduino-lmic-non-avr-printf/hal/hal.cpp.o: In function `hal_enableIRQs':
/Users/marco/Dropbox/Arduino/libraries/arduino-lmic-non-avr-printf/src/hal/hal.cpp:202: undefined reference to `_impure_ptr'
/var/folders/xt/636gqx4x4hz49yz9xgg79tkw0000gn/T/buildaf3861b8e88663e57f7270134668d526.tmp/libraries/arduino-lmic-non-avr-printf/hal/hal.cpp.o:(.text._Z15hal_printf_initv+0xc): undefined reference to `fopencookie'
/var/folders/xt/636gqx4x4hz49yz9xgg79tkw0000gn/T/buildaf3861b8e88663e57f7270134668d526.tmp/libraries/arduino-lmic-non-avr-printf/hal/hal.cpp.o: In function `hal_printf_init()':
/Users/marco/Dropbox/Arduino/libraries/arduino-lmic-non-avr-printf/src/hal/hal.cpp:245: undefined reference to `fopencookie'
collect2: error: ld returned 1 exit status

I really have no idea how to handle such error.

The ESP8266 would be a great platform, on the pro-mini just compiling the example there is a 98% of flash use.

igrr, 2016-07-13:
https://github.com/matthijskooijman/arduino-lmic/blob/non-avr-printf/src/hal/hal.cpp#L231-L247
This won't work with the esp8266 core because it uses newlib, which has no such thing as fopencookie.

marcobrianza, 2016-07-13:
looking closer with at you hint I deducted it was just a serial print issue and not  a more serious problem. I was using the non-avr-printf branch because the master was not building on the Due. Switching back to the master branch it builds !!!
Next week I'll try to send my first lora packet with the ESP8266 :-)

hallard, 2016-07-13:
@marcobrianza 
Are you sure you disabled the LIMC printf constant in [config.h](https://github.com/matthijskooijman/arduino-lmic/blob/master/src/lmic/config.h#L26)

I'm running this stack with no problem on ESP8266

marcobrianza, 2016-07-13:
yes, I will pay attention to this, thank you

devyte, 2017-10-19:
Closing due to age and per #3655 .

closed 2017-10-19 by devyte