esp8266/Arduino #927: Define replacements in static pages

robertgregor, 2015-10-24:
Hello, is it possible to somehow serve the static pages like 
server.serveStatic("/st/", SPIFFS, "/");

But define in these static pages replacements strings like i.e. !!!user!!! and then the webserver will replace is with a variable, i.e.:
server.setStaticReplacement("!!!user!!!",user);
I think, this can be very handy....

Robert

sticilface, 2015-10-24:
Yes.

Ive written a lib that does exactly this. Just last week and still ironing
out some stuff. I'll put an example up today. But it works well. Serves
dynamic pages in 100ms, doesn't use any heap really, 8 bytes per variable,
and uses a different method to get content length correctly so the client
closes the connection.

sticilface, 2015-10-24:
Here you go
https://github.com/sticilface/HTTPPrinter/tree/master

closed 2015-11-03 by igrr