esp8266/Arduino #928: LmacRxBlk:1 mess

mytrain, 2015-10-24:
Hi everybody

I'm no struggling for more than two weeks getting rid of the 'LmacRxBlk:1' errors in my ESP web server.

My ESP is serving a dozen of files (html, js, png) for a total of about 130KB. When a client download the main page (index.html) I've got during download of some file the message 'LmacRxBlk:1' on  the logs (whith Serial.setDebugOutput(true)). On client side download of files where 'LmacRxBlk:1' appears appear to be corrupted (invalid file length). I assume the TCP socket died somewhere.

Reseting the ESP just changes the files impacted.

I've try to:
- my own code instead of using the streamFile fonction of webserver copy block of HTTP_DOWNLOAD_UNIT_SIZE bytes.
- add yield() in the loop between two blocks
- transmitting blocks out of the initial request (1 block max per entry in the 'loop' section)
- test with various clients (IPad Chrome, Window 7 Chrome, Windows 7 Firefox, Android Chome, ...)
- two ESP's (ESP12 on bread board and ESP12-F on my final PCB).

So if somebody can give me a tips about this ---- message, it's help me.

Links2004, 2015-10-24:
the problem comes from the fact that the webserver can only handle one connection at the time.
if a new connection is established by the browser it will be stored internally (backlog) if its not handled you get this error.
all modern browsers open multiple connections for loading images, css and js files in parallel.
its very easy to overload the ESP with all this request.
the maximum of TCP sockets is at 5 and the backlog is at 1.

you can try:
try to put all js files together or embed all images as base64 in one css file to reduse the parallel connections.
create js code inside the html that handle the loading of all other files in a non parallel way.

mytrain, 2015-10-24:
@Links2004 

Thank you for the piece of advise. I'll run some test and come back to you.

mytrain, 2015-10-26:
@Links2004 

Following your advice I re-engineered my html/javascript pages. That was not easy as there is no standard mechanism, I'm aware of, for avoiding parallel request from the browser.

Now it is working perfectly event with two client connected. For the moment ,it is a bit slow because I'm uploading only one block (1450) byte per cycle (loop invocation) but coming back to the original ESP webserver code I hope to get something faster.

If the community is interested in I may provide by html/javascript code as example on how to control upload of resources. For the ones interested I made a short video (sorry introduction is in FRENCH) : https://vimeo.com/143534223

me-no-dev, 2015-10-26:
you can have a small piece of javascript in your index that can load a list of files one after another. Just append the next script/css once the previous one is done. That will ensure that only one request is done at a time.

andig, 2016-01-18:
The problem with `LmacRxBlk` isn't that it only serves one request. The problem is that it leaks memory.

  - logfile: http://pastebin.com/yxHdYgJM
  - sketch: http://pastebin.com/AjHcmQAD

To put load on the system use this html file. in my case running on Windows PC- might be related to the delayed ACK issue?

	<!DOCTYPE html>
	<html>
	<body>
	<script src="https://code.jquery.com/jquery-2.1.4.min.js"></script>
	<script>
	$(document).ready(function() {
		window.setInterval(function() {
			$.getJSON("http://vzero-edd834.local", {
				timeout: 100
			});
		}, 100);
	});
	</script>
	</body>
	</html>

Edzelf, 2016-04-21:
I had the same experience.  In my ESP-Radio project I handle a continuous stream of MP3-data together with a ESP8266WebServer.   I also used ESP8266mDNS to get al local DNS-name for my radio.
The "LmacRxBlk:1" error disappeared when I removed the ESP8266mDNS!

andig, 2016-04-21:
I've not seen this error recently after moving to the async web server.

andig, 2016-04-21:
Mdns is unfortunately required for OTA?

Edzelf, 2016-04-21:
I don't think mDNS is required.  Just added basicOTA to my InternetRadio project and it woks perfectly.

andig, 2016-04-21:
> I don't think mDNS is required. Just added basicOTA to my InternetRadio project and it woks perfectly.

Its already required and started internally: https://github.com/esp8266/Arduino/blob/master/libraries/ArduinoOTA/ArduinoOTA.cpp#L110

I'm wondering- if thats true- how removing mdns can remove the `LmacRxBlk` problems? Worth investigating?

me-no-dev, 2016-04-21:
Going async is how you remove those ;)
they happen because while the included sync server is sending it's packets, the stack can not handle the async mdns packets that are coming (they are more than one since you can hear other's broadcasts).
Removing the mDNS does not fix the problem, but rather disables a useful otherwise service.

Edzelf, 2016-04-21:
andig: In my project ArduinoOTA is only started (from the webserver) if OTA is required.
me-no-dev & andig: I will switch to async webserver anyway.
Thank you both for your feedback.

devyte, 2017-10-19:
I can't reproduce this error with latest git. Closing.

closed 2017-10-19 by devyte

d-a-v, 2017-10-19:
```LmacRxBlk:1``` can also be solved by lwip update #3362. 
It was in fact the trigger.