esp8266/Arduino #929: ESP.shutdown()

rouhu, 2015-10-25:
Would it possible to have command ESP.shutdown().

So that after device has done its work it would shutdown and it would reboot only after manually turning off/on the power. ESP.deepsleep() is not an option here as device should never wake up.

asetyde, 2015-10-25:
You can done all work in setup or broke while in loop and after use button or circuit or power cut to reset ..

I think no control by sdk

Links2004, 2015-10-25:
you can use ESP.deepsleep().
normaly GPIO16 is connected to RESET an is low in case of sleep.
if you create your own circuit that Reset is low and keep it low until you remove the power you can sleep as long you want.
may simply using a Flipflop.

igrr, 2015-10-29:
I think the easiest way to achieve this is using a single P-MOS transistor.

rouhu, 2015-10-29:
Using ESP01 and 0 and 2 Pins are taken but if I connect TX pin to CH_PD and put it HIGH on boot and LOW on shutdown. Should work ?

igrr, 2015-10-29:
ESP will output some stuff to TX pin on startup, so this won't work with TX
pin.

rouhu, 2015-10-29:
All right, would RX pin work instead ?

rouhu, 2015-11-03:
Tried but CH_PD needs to be VCC before even any code runs. Anyway, solved it by letting it sleep 21 000 years,  ESP.deepSleep(999999999*999999999U, WAKE_NO_RFCAL);
Now I only have to wait and see if it works....

closed 2015-11-03 by rouhu

mag-hub, 2019-12-15:
Rather than sleeping for time, You can use

`ESP.deepSleep(0);`

It will wake up only on reset or restart..

boly38, 2023-01-23:
in fact a function that implement this another idea: [Very DeepSleep](https://arduinodiy.wordpress.com/2021/01/15/very-deepsleep-and-energy-saving-on-esp8266-part-6-power-down-up/) (= switch off)

Bighoneypot, 2023-01-23:
Very deepsleep?? Remove power😁

weeblebiker, 2024-03-25:
I'm looking at this because my ESP32-s3 is connected to a battery. I want it totally off to conserve the battery when two buttons are pressed, not just in deep sleep.

d-a-v, 2024-03-25:
@weeblebiker 
You are asking on the esp8266 repository.