esp8266/Arduino #930: About wifi

asetyde, 2015-10-26:
About said and password , i'm not sure cost car is best choice for stability in eprom , is possible use char said[] and char password[]  ? in future ?

i've some problem with refresh and data transformation

Links2004, 2015-10-26:
```char *``` can easy be cast to ```const char *```

example:
```cpp
char test[] = "1234";
const char * ctest1 = (const char *) test;
const char * ctest2 = (const char *) &test[0];
```

what exactly the problems? can you provide example when the problem is shown?

asetyde, 2015-10-28:
Thx for reply , 

In particular about said and password that i store on my eprom , however with many flash this strings are first to be damaged and i must do a eprom reset.
Also to convert data and send on network char data[] is more reliable than char* data .

const char * ssid
const char* password   

I ll prefer to use char ssid[10] and char password[10] but is not permitted at now , how can i do ?

Links2004, 2015-10-28:
```char data[];``` is another writing for ```char * data;``` it does internal the same.

```char data[10] = "1234";```  the compiler will allocate 10 bytes in flash and ram 
```const char data[10] = "1234";```  the compiler will allocate 10 bytes in flash
```const char data[] = "1234";```  the compiler will allocate 5 bytes in flash

```cpp
char test[10] = "1234";
const char * ctest = (const char *) &test[0];
```

the function will take a pointer the app / developer has decide how much memory is needed for the SSID an PASS.
SSID can have up to 32 Byte
PASS up to 64 Byte

begin can be used with ```const char *``` and ```char *```
https://github.com/esp8266/Arduino/blob/esp8266/hardware/esp8266com/esp8266/libraries/ESP8266WiFi/src/ESP8266WiFi.h#L58-L59

I not understand you what you mean by reliable. for the compiler its the same.
the flash live time has nothing to do with the compiler or data type itself.
also the ESP has no EEPROM we emulate it in the SPI flash and there a wirte will always erase a whole segment and write it again.

you only need to write when the SSID or PASS is changed and normally an flash can handle easy 100000 write cycles, if you need to change the SSID or PASS that often I would be very surprised.

asetyde, 2015-10-28:
I can use char ssid[10] with  begin ? 

I put this ssid and password in string and this is put in database ... All data is put with sprintf %s

Links2004, 2015-10-28:
yes ```WiFi.begin((char *) &ssid[0]);```

asetyde, 2015-10-28:
Tomorrow  I'll try , many thanks Markus

asetyde, 2015-10-29:
This a lot markus , all work . Without use in eprom const char , when i flash no damage on eprom is seen . More safe ,i think , use char data[number] fixed value.

Another quest , i use : 
   uint8_t MAC_array[6];
    WiFi.macAddress(MAC_array);
	for (int i = 0; i < sizeof(MAC_array); ++i) {
		if (i < 5) { sprintf(mind.mac_Address, "%s%02x:", mind.mac_Address, MAC_array[i]); }
		else { sprintf(mind.mac_Address, "%s%02x", mind.mac_Address, MAC_array[i]); }
	}
mind.mac_Address is a string 
for mac 
------
But for ip ? , i prefer not use inet arpa library for 1 conversion that i use sometimes in my cycle, i need to put on mind.ip  (char ip[40]) , if someone can help me , 

IPAddress sample = WiFi.localIP();

It's only return value ? another way ? can i sue a print to punt on string

asetyde, 2015-10-29:
all done if can be useful @igrr  take 

			  IPAddress sample = WiFi.localIP();
			  byte oct1 = sample[0];
			  byte oct2 = sample[1];
			  byte oct3 = sample[2];
			  byte oct4 = sample[3];
			  char s[16];
			  sprintf(s, "%d.%d.%d.%d", oct1, oct2, oct3, oct4);

closed 2015-10-29 by asetyde