esp8266/Arduino #935: TCP Socket crashing

conkerkh, 2015-10-28:
I'm having a bit of problem, basically I have 2 TCP Sockets opened and one UDP if I flood UDP socket with packets it immediately crashes TCP sockets (WebSocket to be precise) pretty much when UDP socket is flooded with packets TCP doesn't work at all or is very slow, on the other hand ESP receives UDP packets and processes them just fine. When i lower frequency on UDP packets to like 10p/s everything works better however it still crashes after some time what could be the cause is it SDK issue or rather my code?

igrr, 2015-10-29:
This would depend on how you are handling incoming UDP packets. If you let them pile up, this will most certainly make the device unresponsive over TCP. Ideally, UDP and TCP packets should be processed as soon as they arrive. However due to the blocking nature of Arduino network APIs, this is not always easy to do. 

If you have really high packet load over UDP, I would suggest using LwIP APIs.
Another option is internal UdpContext API. It's not documented, but it's used in mDNS and SSDP libraries. It has a method to specify callback function which will be called when packet arrives, making it possible to process UDP asynchronously. Take a look at the code, it's mostly self-explanatory.

The benefit of using LwIP APIs is that these will not change, so your code will be based on stable APIs.

UdpContext may change, potentially, but it might be slightly easier to use (if you are more C++-inclined).

conkerkh, 2015-10-29:
I looked at Wireshark log there is really weird thing going on and looks like a library bug, basically just before connection is lost something like this happens: 
-client sends ACK packet nr x
-esp replies to ACK nr X with ACK
-client sends ACK packet nr x+1 
-esp for some reason replies with old ACK in wireshark marked as retransmission 

I'm using very low frequency when sending packets trust me just 10 UDP packets/s. Apart from this 10 WebSocket packets are transmitted every second. Whole main loop executes in <40 microseconds.

igrr, 2015-10-29:
This should be something in LwIP stack then.

conkerkh, 2015-10-29:
Any way how I can debug this??

conkerkh, 2015-10-30:
Problem Solved, sorry didn't try your suggestion earlier. Adding callback to onRX solved the issue. I'm not sure if it's the bug but clearly looks like that handling UDP inside loop on every iteration doesn't necessarily work as it should in my case I used it as in documentation so:
````
void UDPHandle() {
    char recieveBuffer[128];
    int packetSize = Udp.parsePacket();
    if (packetSize) {
      last_packet = millis();
      // read the packet into packetBufffer
      int len = Udp.read(recieveBuffer, 128);
      parseBuffer(recieveBuffer, len);
      //Serial.println(recieveBuffer);
}
````

This caused all the problems (I believe it's a problem in the way parsePacket() is handled when there are no packets). In the end in ended up adding this code to WiFiUDP: 

````
typedef std::function<void(void)> rxhandler_t;

void WiFiUDP::regCallback(rxhandler_t handler) {
	_ctx->onRx(handler);
}

```
So far everything looks to be stable and smooth. This Probably means that when using UDP and TCP at the same time there should be callback when UDP packet is received rather than just calling parsePacket on every loop.

igrr, 2015-10-30:
You might be missing a `while` loop in your first code block. There might be multiple packets in queue and you need to handle all of them.

conkerkh, 2015-10-30:
Good point fixed it, thanks for help we can close this issue.

closed 2015-10-30 by conkerkh