esp8266/Arduino #938: quick question guys

FrankOcean11, 2015-10-29:
i own this model 
http://www.ebay.com/itm/191607430420?_trksid=p2057872.m2749.l2649&ssPageName=STRK%3AMEBIDX%3AIT 
is this consider a generic model ?  so that i can follow schematic as shown in your https://github.com/esp8266/Arduino/blob/esp8266/hardware/esp8266com/esp8266/doc/boards.md#generic-esp8266-modules ??

also im using this FTDI http://www.ebay.com/itm/181385609653?_trksid=p2057872.m2749.l2649&ssPageName=STRK%3AMEBIDX%3AIT will this be good enough ?> and will it work according to the your schematic in your Supported boards ? thanks in advanced!

igrr, 2015-10-29:
Yes, you can use "Generic ESP8266 module" option with this board. FTDI should also be fine.
In the future, please use the http://esp8266.com/arduino forum for questions. This issue tracker should be used to submit bugs and feature requests for the core and libraries. Thanks.

closed 2015-10-29 by igrr

FrankOcean11, 2015-10-29:
i kno i know im so sorry i just thought it be much quicker but thanks so much for your support!