esp8266/Arduino #940: DHT library error

pbecchi, 2015-10-29:
I get a compilation error using dht.h lib:

Arduino:1.6.5 (Windows 8.1), Scheda:"NodeMCU 0.9 (ESP-12 Module), 80 MHz, 115200"

d:/users/pbecc/appdata/roaming/arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/lib\libm.a(lib_a-w_sqrt.o):(.literal+0x8): undefined reference to `__ieee754_sqrt'
d:/users/pbecc/appdata/roaming/arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/lib\libm.a(lib_a-w_sqrt.o): In function `sqrt':
d:\ivan\projects\arduinoesp\toolchain\dl\esp-newlib\build\xtensa-lx106-elf\newlib\libm\math/../../../../../newlib/libm/math/w_sqrt.c:63: undefined reference to `__ieee754_sqrt'
collect2.exe: error: ld returned 1 exit status

The library call the sqrt() math function : this give the error( no error commenting out the line!)

10to7, 2015-10-30:
The compiler doesn't support sort, log or pow which are used in the dew point part of this library. If you remove the dewpoint.h and reference the DHT11.h directly to get the relative humidity and temperature it will work.

I believe a less precise approximation of dew point temperature in Centigrade is temp - ((100 - relative humidity)/5)

asetyde, 2015-11-04:
My dht library on nodemcu 1.0 work perfectly . My library not call sqrt

pbecchi, 2015-11-04:
The DHT library contain a module:

float DHT::computeHeatIndex(....
This module call sqrt and pow math function. 
Of course if you don't use it there is no problem with unsupported math!

As 10to7 said, there is simpler way to calculate dew point that don't involve the use of those functions.

I used the library example that call "computeHeatIndex" : I dont normally use this function.

To simplify the library it may be possible to rewrite Dewpoint calculation and also to use integer ( in term of 0,1 °F or °C) rather than floating values!!

asetyde, 2015-11-04:
i use heatindex on bodemcu with dht library , with 0 problems

pbecchi, 2015-11-04:
I use the Adafruit DHT library! It look like this:

float DHT::computeHeatIndex(float temperature, float percentHumidity, bool isFahrenheit) {
  // Using both Rothfusz and Steadman's equations
  // http://www.wpc.ncep.noaa.gov/html/heatindex_equation.shtml
  float hi;

  if (!isFahrenheit)
    temperature = convertCtoF(temperature);

  hi = 0.5 * (temperature + 61.0 + ((temperature - 68.0) * 1.2) + (percentHumidity * 0.094));

  if (hi > 79) {
    hi = -42.379 +
             2.04901523 * temperature +
            10.14333127 * percentHumidity +
            -0.22475541 * temperature*percentHumidity +
            -0.00683783 * pow(temperature, 2) +
            -0.05481717 * pow(percentHumidity, 2) +
             0.00122874 * pow(temperature, 2) * percentHumidity +
             0.00085282 * temperature*pow(percentHumidity, 2) +
            -0.00000199 * pow(temperature, 2) * pow(percentHumidity, 2);
     
    if((percentHumidity < 13) && (temperature >= 80.0) && (temperature <= 112.0))
      hi -= ((13.0 - percentHumidity) * 0.25) */* sqrt*/((17.0 - abs(temperature - 95.0)) * 0.05882);

    else if((percentHumidity > 85.0) && (temperature >= 80.0) && (temperature <= 87.0))
      hi += ((percentHumidity - 85.0) * 0.1) * ((87.0 - temperature) * 0.2);
  }

  return isFahrenheit ? hi : convertFtoC(hi);
}

asetyde, 2015-11-04:
do you use last version ?

pbecchi, 2015-11-04:
Your library is probably different..... look at the your source!

asetyde, 2015-11-04:
Use version 1.1.1 and i think ti will go

pbecchi, 2015-11-04:
Thank's!!
It is important to know that those math functions cannot be used!

asetyde, 2015-11-04:
ok start from this and go up , after says to community ( i don’t try because in my 3000 lines of code i’m afraid :)) but if some new version work is a good idea upgrade

asetyde, 2015-11-07:
https://github.com/adafruit/DHT-sensor-library/issues/45

asetyde, 2015-11-07:
last supported release 1.2.0 , at now i ll try to find a solution

asetyde, 2015-11-07:
please reopen issue

asetyde, 2015-11-07:
https://github.com/asetyde/DHT-sensor-library/pull/2
https://github.com/asetyde/DHT-sensor-library/pull/1

I resolve all error and i realize stable esp8266 dht 1.2.3 version without use of sqrt and i've added sqrt house made function

asetyde, 2015-11-07:
@igrr with my mod , at now library can work and if is necessary sproot function can be used DHT.sqroot(double value);  (return double value)

i'm waiting to library creator , but we can use library DHT 1.2.3 with more hex accuracy and force instantly system

igrr, 2015-11-07:
I think sqrt issue should be fixed in libm build. Or would you rather fix
every piece of code which happens to use sqrt?

asetyde, 2015-11-07:
For dht library it's not problem , but I understood it's not in library , or it's not implemented

pbecchi, 2015-11-07:
Indipendently from the Adafruit DHT library (it is the only one that use complex equation to calculate dewpoint) I think will be nice to have the math functions available in libm.

For the DHT case we can use the Rob Tillaart one that is the official Arduino library: https://github.com/RobTillaart.(no use of fancy math!).

But for other calculations it may be usefull to have those functions!!

asetyde, 2015-11-08:
i prefer adafruit version , more function

asetyde, 2015-11-24:
I probably found a bug , with my mod dht library 1.2.3 can block device , now investigating , i restore dht standard 1.1.1 and try . my device has 2  type of sensor dht and analogo photo-resistor , i think it's impossibile that photo-resistor can block device , i suppose this . 

i ll give news as soon as possible

asetyde, 2015-11-26:
Library 1.2.3 mod that i've done has a big problem , probably sqroot function  cause a block of device .. now investigating

asetyde, 2015-11-27:
@igrr  sqrt is ok in 2.0.0 ?

igrr, 2015-11-27:
Not a Windows user so can't check myself. Could someone running Windows check please?

asetyde, 2015-11-27:
yes i’ve also windows , how check ?

pbecchi, 2015-11-27:
Somebody on Arduino forum said that sqrt is working with 2.0.0!

I will install it and test!

asetyde, 2015-11-27:
ok i’m wait because my code crash with new 2.0.0

pbecchi, 2015-11-27:
Confirmed!
 SQRT  on 2.0.0 no compilation error , no execution errors :
 SQRT(4.0)  result 2.0!!

igrr, 2015-11-27:
sqrt of a compile-time constant is evaluated by the compiler. You need to do something like
```c++
String s = "4.0";
float f = s.toFloat();
Serial.println(sqrt(f));
```
to verify.

asetyde, 2015-11-27:
yes, it goes. I will try new dht library but last time wifi and mysql not work with last staging , i think i stay in stable  , when i come back to 1.1.1 dht library in stable my wifi go ok before stopped random ..

asetyde, 2015-11-27:
thanks pbecchi

pbecchi, 2015-11-27:
tttThe code I use is

pbecchi, 2015-11-27:
The code I used is:
void setup() {
  // put your setup code here, to run once:
double e=4.;
Serial.begin(9600);
Serial.println(sqrt(e));
}
I will verify now with DHT library.

pbecchi, 2015-11-27:
DHT Adafruit library compile with no problem.... 
I think issue could be closed!

asetyde, 2015-11-27:
I Think not , my issue library work but after some time , board is blocked .. Need some time test with network monitor, you code is so simple.

closed 2015-11-30 by igrr

asetyde, 2016-01-06:
@igrr  stable 2.0.0 with DHT 1.2.3 is full compatible , please note on readme . thx