esp8266/Arduino #942: Weird behaviour with EEPROM.begin()

marvinroger, 2015-10-30:
I would like to call `EEPROM.begin()` with the exact amount of storage needed to save memory.
I want to `put` a struct composed of:

- 1 `boolean` (1 byte)
- 1 `byte` (1 byte)
- 5 `char[33]` (165 bytes)
- 1 `unsigned int` (4 bytes)
- for a total of 1 + 1 + 165 + 4 = 171 bytes

So calling `begin` with 171 should be okay. But it's not. It does not crash or anything, it just seems the data doesn't get written. But calling it with at least 173 does work.

Here is the test code:

```cpp
#include <Arduino.h>
#include <EEPROM.h>

const int EEPROM_OFFSET = 0;
const int EEPROM_OFFSET_SETUP = EEPROM_OFFSET;
const int EEPROM_LENGTH_SETUP = 1;
const int EEPROM_OFFSET_BOOT_TYPE = EEPROM_OFFSET_SETUP + EEPROM_LENGTH_SETUP;
const int EEPROM_LENGTH_BOOT_TYPE = 1;
const int EEPROM_OFFSET_HOSTNAME = EEPROM_OFFSET_BOOT_TYPE + EEPROM_LENGTH_BOOT_TYPE;
const int EEPROM_LENGTH_HOSTNAME = 32 + 1;
const int EEPROM_OFFSET_WIFI_SSID = EEPROM_OFFSET_HOSTNAME + EEPROM_LENGTH_HOSTNAME;
const int EEPROM_LENGTH_WIFI_SSID = 32 + 1;
const int EEPROM_OFFSET_WIFI_PASSWORD = EEPROM_OFFSET_WIFI_SSID + EEPROM_LENGTH_WIFI_SSID;
const int EEPROM_LENGTH_WIFI_PASSWORD = 32 + 1;
const int EEPROM_OFFSET_MQTT_HOST = EEPROM_OFFSET_WIFI_PASSWORD + EEPROM_LENGTH_WIFI_PASSWORD;
const int EEPROM_LENGTH_MQTT_HOST = 32 + 1;
const int EEPROM_OFFSET_MQTT_PORT = EEPROM_OFFSET_MQTT_HOST + EEPROM_LENGTH_MQTT_HOST;
const int EEPROM_LENGTH_MQTT_PORT = 4;
const int EEPROM_OFFSET_MQTT_BASE_TOPIC = EEPROM_OFFSET_MQTT_PORT + EEPROM_LENGTH_MQTT_PORT;
const int EEPROM_LENGTH_MQTT_BASE_TOPIC = 32 + 1;
const int EEPROM_SIZE = EEPROM_OFFSET_MQTT_BASE_TOPIC + EEPROM_LENGTH_MQTT_BASE_TOPIC;

enum BootType : byte {
  NORMAL = 1,
  CONFIG = 2
};

struct ConfigStruct {
  bool configured;
  BootType boot_type;
  char hostname[EEPROM_LENGTH_HOSTNAME];
  char wifi_ssid[EEPROM_LENGTH_WIFI_SSID];
  char wifi_password[EEPROM_LENGTH_WIFI_PASSWORD];
  char mqtt_host[EEPROM_LENGTH_MQTT_HOST];
  unsigned int mqtt_port;
  char mqtt_base_topic[EEPROM_LENGTH_MQTT_BASE_TOPIC];
};

void setup() {
  Serial.begin(115200);
  Serial.print("EEPROM of ");
  Serial.print(EEPROM_SIZE);
  Serial.println(" bytes");
  EEPROM.begin(EEPROM_SIZE);
  
  ConfigStruct config;
  config.configured = true;
  
  EEPROM.put(EEPROM_OFFSET, config);
  EEPROM.commit();

  ConfigStruct config_read;
  EEPROM.get(EEPROM_OFFSET, config_read);
  if (config_read.configured) {
    Serial.println("OK");
  } else {
    Serial.println("Not OK");
  }
}

void loop() {
  
}
```

Maybe i've made a mistake somewhere, but I just don't understand what happens.
Thanks.

igrr, 2015-10-30:
Writes are only possible in 4-byte words, so when you request 171 bytes to be written, the library actually needs to write 172 bytes (43 words). But since you have called begin requesting just 171, i suppose the write fails in bounds checking code.

There was a change (https://github.com/esp8266/Arduino/commit/afd0ca23a09a2896a73c65a4c0aa014d3602c093) which made sure size requested in a call to `begin` would be rounded up to a word boundary. Are you using stable or staging package?

igrr, 2015-10-30:
I suggest you to call `EEPROM.begin(sizeof(ConfigStruct))` which will take alignment into account and you won't need to calculate EEPROM_SIZE manually.

marvinroger, 2015-10-30:
Oh, I see. That's why the minimum for begin is 4.

I am using the latest staging package. If I call begin with 171 `(EEPROM_SIZE + 3) & (~3)` does return 172, but still it doesn't work. If I call begin with 173, `(EEPROM_SIZE + 3) & (~3)` returns 176 and then it works.

Forget what I said, `sizeof(ConfigStruct)` returns 176... Why? Is there an overhead of the Struct?

marvinroger, 2015-10-30:
For those wondering, a Struct may add some padding depending on alignment rules. See http://stackoverflow.com/a/5435890

So the only reliable way to get the size of a Struct is obvisouly through `sizeof(Struct)`. Thanks @igrr for pointing it out.

closed 2015-10-30 by marvinroger

igrr, 2015-10-30:
mqtt_port is aligned to 4-byte boundary, so it's offset is 136 bytes, not 133.
Apart from `sizeof` there is also `offsetof(type, member)` macro which will calculate offsets automatically for you, taking alignment rules into account.