esp8266/Arduino #948: Deprecated recipe.ar.pattern

probonopd, 2015-11-01:
Arduino 1.6.6 hourly build 201510300355 suggests to consider upgrading this core:

```
./Arduino_1.6.6_201510300355
Picked up JAVA_TOOL_OPTIONS: 
Warning: platform.txt from core 'ESP8266 Modules' contains deprecated recipe.ar.pattern="{compiler.path}{compiler.ar.cmd}" {compiler.ar.flags} {compiler.ar.extra_flags} "{build.path}/{archive_file}" "{object_file}", automatically converted to recipe.ar.pattern="{compiler.path}{compiler.ar.cmd}" {compiler.ar.flags} {compiler.ar.extra_flags} "{archive_file_path}" "{object_file}". Consider upgrading this core.
```

igrr, 2015-11-08:
Apparently changing this to `{archive_file_path}` removes the warning but breaks compatibility with 1.6.5.
I'm considering to leave the warning in place for now.

fredericplante, 2015-11-08:
Ah! The Arduino IDE again. MdR. ;)

Is there any solution to fix that problem?

Warning: platform.txt from core 'Arduino ARM (32-bits) Boards' contains deprecated recipe.c.combine.pattern="{compiler.path}{compiler.c.elf.cmd}" {compiler.c.elf.flags} -mcpu={build.mcu} "-T{build.variant.path}/{build.ldscript}" "-Wl,-Map,{build.path}/{build.project_name}.map" {compiler.c.elf.extra_flags} -o "{build.path}/{build.project_name}.elf" "-L{build.path}" -mthumb -Wl,--cref -Wl,--check-sections -Wl,--gc-sections -Wl,--entry=Reset_Handler -Wl,--unresolved-symbols=report-all -Wl,--warn-common -Wl,--warn-section-align -Wl,--warn-unresolved-symbols -Wl,--start-group "{build.path}/syscalls_sam3.c.o" {object_files} "{build.variant.path}/{build.variant_system_lib}" "{build.path}/{archive_file}" -Wl,--end-group -lm -gcc, automatically converted to recipe.c.combine.pattern="{compiler.path}{compiler.c.elf.cmd}" {compiler.c.elf.flags} -mcpu={build.mcu} "-T{build.variant.path}/{build.ldscript}" "-Wl,-Map,{build.path}/{build.project_name}.map" {compiler.c.elf.extra_flags} -o "{build.path}/{build.project_name}.elf" "-L{build.path}" -mthumb -Wl,--cref -Wl,--check-sections -Wl,--gc-sections -Wl,--entry=Reset_Handler -Wl,--unresolved-symbols=report-all -Wl,--warn-common -Wl,--warn-section-align -Wl,--warn-unresolved-symbols -Wl,--start-group "{build.path}/core/syscalls_sam3.c.o" {object_files} "{build.variant.path}/{build.variant_system_lib}" "{build.path}/{archive_file}" -Wl,--end-group -lm -gcc. Consider upgrading this core.
Warning: platform.txt from core 'Arduino ARM (32-bits) Boards' contains deprecated recipe.ar.pattern="{compiler.path}{compiler.ar.cmd}" {compiler.ar.flags} {compiler.ar.extra_flags} "{build.path}/{archive_file}" "{object_file}", automatically converted to recipe.ar.pattern="{compiler.path}{compiler.ar.cmd}" {compiler.ar.flags} {compiler.ar.extra_flags} "{archive_file_path}" "{object_file}". Consider upgrading this core.
Board generic (platform esp8266, package esp8266) is unknown

Error compiling.

igrr, 2015-11-08:
@fredericplante please double-check which board you have selected in Tools > Board menu.

`core 'Arduino ARM (32-bits) Boards' ` and `Warning: platform.txt from core 'Arduino ARM (32-bits) Boards' contains deprecated recipe.ar.pattern` makes me think that you have some ARM board selected.

fredericplante, 2015-11-08:
I don't know what to say.

[image]

fredericplante, 2015-11-08:
When i reaload the right board, i get 

Arduino: 1.6.6 (Windows XP), Board: "Generic ESP8266 Module, Serial, 80 MHz, 40MHz, DIO, 921600, 4M (3M SPIFFS)"

Warning: platform.txt from core 'Arduino ARM (32-bits) Boards' contains deprecated recipe.c.combine.pattern="{compiler.path}{compiler.c.elf.cmd}" {compiler.c.elf.flags} -mcpu={build.mcu} "-T{build.variant.path}/{build.ldscript}" "-Wl,-Map,{build.path}/{build.project_name}.map" {compiler.c.elf.extra_flags} -o "{build.path}/{build.project_name}.elf" "-L{build.path}" -mthumb -Wl,--cref -Wl,--check-sections -Wl,--gc-sections -Wl,--entry=Reset_Handler -Wl,--unresolved-symbols=report-all -Wl,--warn-common -Wl,--warn-section-align -Wl,--warn-unresolved-symbols -Wl,--start-group "{build.path}/syscalls_sam3.c.o" {object_files} "{build.variant.path}/{build.variant_system_lib}" "{build.path}/{archive_file}" -Wl,--end-group -lm -gcc, automatically converted to recipe.c.combine.pattern="{compiler.path}{compiler.c.elf.cmd}" {compiler.c.elf.flags} -mcpu={build.mcu} "-T{build.variant.path}/{build.ldscript}" "-Wl,-Map,{build.path}/{build.project_name}.map" {compiler.c.elf.extra_flags} -o "{build.path}/{build.project_name}.elf" "-L{build.path}" -mthumb -Wl,--cref -Wl,--check-sections -Wl,--gc-sections -Wl,--entry=Reset_Handler -Wl,--unresolved-symbols=report-all -Wl,--warn-common -Wl,--warn-section-align -Wl,--warn-unresolved-symbols -Wl,--start-group "{build.path}/core/syscalls_sam3.c.o" {object_files} "{build.variant.path}/{build.variant_system_lib}" "{build.path}/{archive_file}" -Wl,--end-group -lm -gcc. Consider upgrading this core.
Warning: platform.txt from core 'Arduino ARM (32-bits) Boards' contains deprecated recipe.ar.pattern="{compiler.path}{compiler.ar.cmd}" {compiler.ar.flags} {compiler.ar.extra_flags} "{build.path}/{archive_file}" "{object_file}", automatically converted to recipe.ar.pattern="{compiler.path}{compiler.ar.cmd}" {compiler.ar.flags} {compiler.ar.extra_flags} "{archive_file_path}" "{object_file}". Consider upgrading this core.
Board generic (platform esp8266, package esp8266) is unknown

Error compiling.
Invalid library found in C:\Documents and Settings\gust\Local Settings\Application Data\Arduino15\packages\esp8266\hardware\esp8266\1.6.5-1160-gef26c5f\libraries\esp8266: C:\Documents and Settings\gust\Local Settings\Application Data\Arduino15\packages\esp8266\hardware\esp8266\1.6.5-1160-gef26c5f\libraries\esp8266
Invalid library found in C:\Documents and Settings\gust\Local Settings\Application Data\Arduino15\packages\esp8266\hardware\esp8266\1.6.5-1160-gef26c5f\libraries\esp8266: C:\Documents and Settings\gust\Local Settings\Application Data\Arduino15\packages\esp8266\hardware\esp8266\1.6.5-1160-gef26c5f\libraries\esp8266

  This report would have more information with
  "Show verbose output during compilation"
  enabled in File > Preferences.

flimshaw, 2015-11-10:
+1, also seeing this error

dmkanter, 2015-11-10:
+1 seeing this too. downgraded to 1.6.5 for now and able to compile & flash just fine.

bachi76, 2015-11-16:
@igrr - this change seems to break building from Eclipse Arduino: The auto-generate makeFile will contain

```
Starting combiner
...xtensa-lx106-elf-gcc: error: arduino.ar: No such file or directory
makefile:89: recipe for target 'ESP_Configurable_Wifi_Connection.elf' failed
```

If I manually replace all instances of core.a with arduino.ar in the makefile and run make from the command line it will build. If I re-run from Eclipse, makefile gets rewritten and the build breaks.

Disclaimer: I don't know much about make and the auto-generation of it. I found 'arduino.ar' in some pref setting (environment/project/it.baeyens.arduino.core.toolChain.release.2078438648/A.ARCHIVE_FILE/value=arduino.ar) but that was also auto-reset when manually replacing it with 'core.a'.

closed 2015-11-30 by igrr

probonopd, 2016-01-16:
Using Arduino 1.6.7 Hourly Build 2015/12/17 04:29 I installed http://arduino.esp8266.com/stable/package_esp8266com_index.json, then compiled a sketch, then installed http://arduino.esp8266.com/staging/package_esp8266com_index.json, and am now getting 

`Board generic (...) is unknown`

davidbuzz, 2016-01-24:
+1 to "Board generic (platform esp8266, package esp8266) is unknown" when trying to compile anything ( eg blink example ) on 1.6.6 or 1.6.7 releases of IDE ( I'm using OSX).    Version 1.6.4, and 1.6.5r5 both work fine.

Links2004, 2016-01-24:
for switching between stable and staging a manual delete of the other version is needed or the IDE not know what to use.
delete all esp stuff in the Arduino dir in APP data, and install the esp core new.

davidbuzz, 2016-01-24:
@Links2004 I usually rename the entire /Applications/Arduino.app to a different name, which should achieve the same thing, no?

Links2004, 2016-01-24:
yes, but keep in mind the IDE changed the path somewhere from 1.6.5 to 1.6.7.
best you can do is  add a dir called "portable" where the "arduino.exe" is,
then the IDE will store all stuf inside there. 
by this you can have more then one IDE with multiple versions without renaming :)

davidbuzz, 2016-01-24:
On OSX this worked for me: 
1. close the IDE first. 
2. renaming the directory ~/Library/Arduino15  to ~/Library/Arduino15-old
3. copying the preferences.txt from the old one to the new one
4. opening 1.6.7 IDE ( latest ) , and entering Tools->Board xx ->Board Manager 
5. In Board Manager, re-installing the addons I previously had . ( in particular the esp8266 one ) 
6. Unders Tools->Board, select the correct esp8266 board( for me it's the generi one )
7. compile something that didn't work in 1.6.7 before, and it works fine now.

sja, 2016-01-27:
@davidbuzz That helped, thanks. :+1: