esp8266/Arduino #953: installing new package by boards manager: java.lang.NullPointerException

Juppit, 2015-11-01:
After building the new repository with cygwin I got an error in the Arduino IDE when I try to install the package (esp8266-71347b8) with following message:
```
java.lang.RuntimeException: java.lang.NullPointerException
	at cc.arduino.contributions.packages.ui.ContributionManagerUI$4.run(ContributionManagerUI.java:182)
	at java.lang.Thread.run(Thread.java:745)
Caused by: java.lang.NullPointerException
	at java.io.File.<init>(File.java:360)
	at cc.arduino.contributions.packages.ContributionInstaller.install(ContributionInstaller.java:154)
	at cc.arduino.contributions.packages.ui.ContributionManagerUI$4.run(ContributionManagerUI.java:176)
```
I removed \Users\[user_name]\AppData\Roaming\Arduino15 without success.

At the moment I am back on staging version.

igrr, 2015-11-01:
Yeah, I'm getting this as well. It happens because right now we don't have any tags, and version is simply a commit number (71347b8). Arduino IDE expects version number to be semver-compliant, and runs into an error because it is not. I have just verified this by creating a local tag (2.0.0-dev1). With this tag created, new package was installed just fine. I'm keeping this issue open for now, it will be resolved once I tag the release candidate.
For now if you want to use git version, I suggest you to clone this repository into Arduino hardware directory — see instructions in Readme. This is much easier than generating a package and installing it, and you can update your version simply by doing a git pull.

Juppit, 2015-11-01:
Txs for the quick answer.
For now I will find a workaround with your explanations.

On 01.11.2015 at 19:39 wroteIvan Grokhotkov:

Juppit, 2015-11-09:
@igrr Due to the problems with Arduino 1.6.6 I'm afraid we will have to wait for Milestone 2.0.0.
That is why I propose that, we could make a litle change in build_board_manager_package.sh:
 ver=0.0.0-`git describe --tags --always`
This ugly version string will be semver-compliant until milestone 2.0.0

Juppit, 2015-11-18:
Yea, so I'll close it.

closed 2015-11-18 by Juppit