esp8266/Arduino #954: Setting the AP channel does not work

sticilface, 2015-11-01:
Setting this 
```
WiFi.softAP(ssid, password, 5, 0);
```
in the AP demo lib, results in a lib being created with channel 1 (at least according to my mac, wireless diagnostics.  Is this behaviour known?

igrr, 2015-11-02:
Did you call `WiFi.mode(WIFI_AP)` explicitly before calling `WiFi.softAP`? If the ESP is already connected as STA, it will use the same channel for AP (because it can not do two channels at the same time).

sticilface, 2015-11-02:
It is already connected to STA, so that explains it then. 

Thanks

closed 2015-11-02 by sticilface

prabbit237, 2018-06-20:
I realize this is an old issue but the following solution is still pertinent. 

Take two esp8266s, one to connect to the router in station mode and one to provide softAP to the other equipment. Then use serial-to-serial to pass the info back and forth. The first esp8266 could be a esp-01 to reduce cost (but these days you can get the nodemcu boards for under $2.45.) 

The serial-to-serial might be low-bandwidth but for IoT devices, they don't need to do screaminn, make yer nose bleed speeds. And apparently you can come (after overhead) close to 4mbps (page 12 https://www.ctr-electronics.com/downloads/pdf/4A-ESP8266__AT_Instruction_Set__EN_v0.40.pdf "Baudrate range: 110 to 115200*40 (4.608 Mega)")