esp8266/Arduino #956: VisualStudio complier failing

grahamehorner, 2015-11-02:
I'm using the VisualStudio plugin to build a custom library, however everything I've tried thus far gives errors with regards header files etc. note output below; any suggestions greatly appreciated. 

Arduino.h:28:26: fatal error: avr\pgmspace.h: No such file or directory
:#include <avr\pgmspace.h>
:compilation terminated

Compiling 'WebServer' for 'NodeMCU 1.0 (ESP-12E Module)'
Build folder: file:///C:/Users/grahame.horner/AppData/Local/V.Micro/Arduino/Builds/WebServer/nodemcuv2
Summary: Header=1 Prototypes=3 Imports=0
Additional Defines: 
Architecture Tools: C:\Users\grahame.horner\AppData\Roaming\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2/bin/
GCC: 4.8.2
Sketchbook: file:///D:/DEV/git.hub/arduino
Core Include Paths
Include Path 'D:\DEV\git.hub\arduino\WebServer'
Include Path 'C:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino'
Include Path 'C:\Program Files (x86)\Arduino\hardware\esp8266-1.6.5-947\esp8266-1.6.5-947-g39819f0\variants\nodemcu'
"C:\Users\grahame.horner\AppData\Roaming\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\Program Files (x86)\Arduino\hardware\esp8266-1.6.5-947\esp8266-1.6.5-947-g39819f0/tools/sdk//include" -c -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11 -MMD -DF_CPU=80000000L -DARDUINO=165 -DARDUINO_ESP8266_ESP12 -DARDUINO_ARCH_ESP8266-1.6.5-947-G39819F0 -DESP8266 -I"D:\DEV\git.hub\arduino\WebServer" -I"C:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino" -I"C:\Program Files (x86)\Arduino\hardware\esp8266-1.6.5-947\esp8266-1.6.5-947-g39819f0\variants\nodemcu" "C:\Users\grahame.horner\AppData\Local\V.Micro\Arduino\Builds\WebServer\nodemcuv2\WebServer.cpp" -o "C:\Users\grahame.horner\AppData\Local\V.Micro\Arduino\Builds\WebServer\nodemcuv2\WebServer.cpp.o"
:ISO C99 requires whitespace after the macro name [enabled by default]
WebServer.cpp:In file included from
Arduino.h:28:26: fatal error: avr\pgmspace.h: No such file or directory
:#include <avr\pgmspace.h>
:compilation terminated
"C:\Users\grahame.horner\AppData\Roaming\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\Program Files (x86)\Arduino\hardware\esp8266-1.6.5-947\esp8266-1.6.5-947-g39819f0/tools/sdk//include" -c -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11 -MMD -DF_CPU=80000000L -DARDUINO=165 -DARDUINO_ESP8266_ESP12 -DARDUINO_ARCH_ESP8266-1.6.5-947-G39819F0 -DESP8266 -I"D:\DEV\git.hub\arduino\WebServer" -I"C:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino" -I"C:\Program Files (x86)\Arduino\hardware\esp8266-1.6.5-947\esp8266-1.6.5-947-g39819f0\variants\nodemcu" "C:\Users\grahame.horner\AppData\Local\V.Micro\Arduino\Builds\WebServer\nodemcuv2\WebServer.cpp" -o "C:\Users\grahame.horner\AppData\Local\V.Micro\Arduino\Builds\WebServer\nodemcuv2\WebServer.cpp.o"
:ISO C99 requires whitespace after the macro name [enabled by default]
WebServer.cpp:In file included from
Arduino.h:28:26: fatal error: avr\pgmspace.h: No such file or directory
:#include <avr\pgmspace.h>
:compilation terminated
Error compiling

asetyde, 2015-11-02:
library include ! not avr\pgmspace.h
#include <pgmspace.h> for esp

grahamehorner, 2015-11-02:
@asetyde sorry don't quite follow; I don't include this in my code anywhere; this is coming from other core .h file that I referenced like Arduino.h, I shouldn't have to reference #include <pgmspace.h> explicitly?

grahamehorner, 2015-11-02:
is the order of the include path correct?

Core Include Paths
 Include Path 'D:\DEV\git.hub\arduino\WebServer'
 Include Path 'C:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino'
 Include Path 'C:\Program Files (x86)\Arduino\hardware\esp8266-1.6.5-947\esp8266-1.6.5-947-g39819f0\variants\nodemcu'

igrr, 2015-11-02:
`C:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino` should not be in include paths, not sure why Visual Studio is doing that for you.

grahamehorner, 2015-11-02:
@igrr ok, this must be an issue with the current version of the Visualmicro plugin as I've simply created a new Arduino Project and built the project for Arduino 1.6 - NodeMCU 1.0 (ESP-12E Module)

asetyde, 2015-11-02:
Which  version of visual micro ?

grahamehorner, 2015-11-02:
@igrr issue logged @asetyde 1510.30.0 - 31/10/2015

asetyde, 2015-11-02:
I'm same version , I found some problems of cleaning files . You can uninstall and try to use old version end October  .
My work , therefore something not work, I remember there are a tool to clean visual micro

grahamehorner, 2015-11-02:
@asetyde I've done a fresh install on a VM of Windows10 with VS2015 and AtmelStudio 7 with a version of Visualmicro from end oct.  both VS2015 and AtmelStudio have the same issue, if I remove the  C:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\Arduino path from the directory list, it automatically gets re-added?

VisualMicro, 2015-11-03:
Hi,

Sorry for the confusion. There seems to be a bug if both a Board Manager core is installed and also a core has been installed into C:\Program Files (x86)\Arduino\hardware\...

You should only need to use one of them so either uninstall the Board Manager version or remove C:\Program Files (x86)\Arduino\hardware\esp8266-1.6.5-947\esp8266-1.6.5-947-g39819f0

Of course this bug will be fixed in a future update but the instruction above will correct the problem in the short term.

Thanks

closed 2015-11-15 by igrr