esp8266/Arduino #958: Use without Arduino IDE / arduino-builder problem?

marvinroger, 2015-11-02:
I would like to code on Atom and get rid of the Arduino IDE, kind of limited.
Is there a way to build without using it?

I tried arduino-builder with `/opt/arduino-builder/arduino-builder -compile -hardware /opt/arduino-builder/hardware/ -hardware ~/.arduino15/packages/esp8266/hardware/ -tools ~/.arduino15/packages/esp8266/tools/ -libraries ~/Arduino/libraries -fqbn esp8266:1.6.5-1160-gef26c5f:nodemcuv2 ~/Arduino/Sketch/Sketch.ino`, but `ARDUINO_ARCH_ESP8266` doesn't get defined so libraries like OneWire fail. When changing from `ARDUINO_ARCH_ESP8266` to `ESP8266` it passes, but I get `fork/exec {runtime.tools.ctags.path}/ctags: no such file or directory`.

So something is not right, seems like it doesn't parse parameters correctly. I saw in this issue tracker that you would maybe use `arduino-builder` to do CI, so I guess it should be compatible. Maybe I am doing it wrong? Or there is a problem in arduino-builder?

igrr, 2015-11-03:
@pgollor was able to make use of arduino-builder, perhaps you can take a look at his scripts?
https://github.com/pgollor/esp8266-arduino-autobuilder/

marvinroger, 2015-11-03:
Thanks, I'll take a look at it. But from version 1.5.0 arduino has its own official CLI, which makes things even simpler: `/opt/arduino/arduino --verify --board esp8266:esp8266:nodemcuv2 Sketch.ino`.

By the way Arduino just released IDE 1.6.6 which notably brings library to library dependency (no need to import every library used by dependencies in the .ino file). Just wanted to ping you ;)

igrr, 2015-11-03:
Arduino CLI is what we are using for Travis CI builds, you may check the
scripts in this repository.

Regarding 1.6.6, yeah, it actually breaks a few things here.

Rickgg, 2015-11-03:
I have used [Platformio](http://platformio.org) to program my Arduino in Atom, in conjunction with [Platomformio](https://atom.io/packages/platomformio) and It works very well. I haven't tried to use platformio to program the ESP, but supposedly it works.

marvinroger, 2015-11-03:
I heard about it, but there is only support for ESP-01 (for now) and I am
using nodemcu. Thanks for the tip!

sej7278, 2015-11-04:
it seems 1.6.6 uses arduino-builder to do the compilation so is totally broken for esp8266, it seems to be due to the naming scheme for platform/package (they shouldn't both be "esp8266")

marvinroger, 2015-11-04:
Arduino CLI still exists. (see https://github.com/arduino/Arduino/blob/099260600718a9c664df6050e7bd934b87f661a4/build/shared/revisions.txt#L11).

closed 2015-11-07 by marvinroger

steve-superiontech, 2015-11-12:
I am also seeing the error message: fork/exec {runtime.tools.ctags.path}/ctags: no such file or directory.

Did you resolve this? How?

sej7278, 2015-11-12:
@steve-superiontech i believe the issue is with 1.6.5 upgrades - nuke your arduino install (including ~/.arduino15) and install 1.6.6 from scratch, works for me using my Makefile:

```make
ARDUINO_PATH = $(HOME)/arduino-1.6.6
SKETCHBOOK   = $(HOME)/sketchbook
ESPTOOL		 = $(SKETCHBOOK)/hardware/esp8266com/esp8266/tools/esptool/esptool
SKETCH 		 = $(notdir $(CURDIR)).ino
TARGET_DIR   = $(CURDIR)/build-esp8266
MONITOR_PORT = /dev/ttyUSB0

all:
	@ mkdir -p $(TARGET_DIR)

	$(ARDUINO_PATH)/arduino-builder -compile -logger=machine \
	-hardware "$(ARDUINO_PATH)/hardware" \
	-hardware "$(SKETCHBOOK)/hardware" \
	-tools "$(ARDUINO_PATH)/tools-builder" \
	-tools "$(ARDUINO_PATH)/hardware/tools/avr" \
	-built-in-libraries "$(ARDUINO_PATH)/libraries" \
	-libraries "$(SKETCHBOOK)/libraries" \
	-fqbn=esp8266com:esp8266:generic:UploadTool=esptool,CpuFrequency=80,FlashFreq=40,FlashMode=dio,UploadSpeed=115200,FlashSize=512K64 \
	-ide-version=10606 \
	-build-path "$(TARGET_DIR)" \
	-warnings=none \
	-prefs=build.warn_data_percentage=75 \
	-verbose "$(SKETCH)"

flash:
	$(ESPTOOL) -v -cd ck -cb 115200 -cp $(MONITOR_PORT) -ca 0x00000 -cf $(TARGET_DIR)/$(SKETCH).bin 

upload: all flash

clean:
	rm -rf $(TARGET_DIR)

monitor:
	screen $(MONITOR_PORT) 9600
```

marvinroger, 2015-11-12:
I've seen this Makefile somewhere... Do you have the link of the original post? I can't find it anymore.

sej7278, 2015-11-12:
its my Makefile you won't have seen it anywhere, most of the "all" recipe is just the output from the ide that i've parameterized

steve-superiontech, 2015-11-12:
Thanks for that. I’ve tried using this as a template, though I’m not doing esp8266. I now get a different error:

===Running: {0} ||| [PrintUsedAndNotUsedLibraries]
===Running: {0} ||| [PrintUsedLibrariesIfVerbose]
Rel: can't make tmp_build_dir/core/core.a relative to tmp_build_dir
*errors.errorString Rel: can't make tmp_build_dir/core/core.a relative to tmp_build_dir
/home/jenkins/jenkins/jobs/arduino-builder-linux64/workspace/src/arduino.cc/builder/utils/utils.go:278 (0x469e82)
/home/jenkins/jenkins/jobs/arduino-builder-linux64/workspace/src/arduino.cc/builder/phases/linker.go:55 (0x4c2fe9)
/home/jenkins/jenkins/jobs/arduino-builder-linux64/workspace/src/arduino.cc/builder/builder.go:155 (0x440159)
/home/jenkins/jenkins/jobs/arduino-builder-linux64/workspace/src/arduino.cc/builder/builder.go:115 (0x43f555)
/home/jenkins/jenkins/jobs/arduino-builder-linux64/workspace/src/arduino.cc/builder/builder.go:186 (0x440709)
/home/jenkins/jenkins/jobs/arduino-builder-linux64/workspace/main.go:315 (0x403554)
/opt/go/src/runtime/proc.go:63 (0x415bf3)
/opt/go/src/runtime/asm_amd64.s:2232 (0x43bad1)

Where tmp_build_dir is my TARGET_DIR.

My build script is:

$ cat foo
rm -r ./tmp_build_dir
mkdir ./tmp_build_dir
arduino-builder \
-logger=machine \
-hardware=$ARDUINO_ROOT/hardware \
-tools=$ARDUINO_ROOT/tools-builder \
-tools=$ARDUINO_ROOT/hardware/tools/avr \
-fqbn=arduino:avr:uno \
-built-in-libraries=$ARDUINO_ROOT/libraries \
-build-path=./tmp_build_dir \
-ide-version=10606 \
-prefs=build.warn_data_percentage=75 \
-debug-level=10 \
Digital_IO.ino

From: Simon John
Reply-To: esp8266/Arduino
Date: Thursday, November 12, 2015 at 3:18 PM
To: esp8266/Arduino
Cc: Steve Phillips
Subject: Re: [Arduino] Use without Arduino IDE / arduino-builder problem? (#958)

@steve-superiontech<https://github.com/steve-superiontech> i believe the issue is with 1.6.5 upgrades - nuke your arduino install (including ~/.arduino15) and install 1.6.6 from scratch, works for me using my Makefile:

ARDUINO_PATH = $(HOME)/arduino-1.6.6
SKETCHBOOK   = $(HOME)/sketchbook
ESPTOOL      = $(SKETCHBOOK)/hardware/esp8266com/esp8266/tools/esptool/esptool
SKETCH       = $(notdir $(CURDIR)).ino
TARGET_DIR   = $(CURDIR)/build-esp8266
MONITOR_PORT = /dev/ttyUSB0

all:
    @ mkdir -p $(TARGET_DIR)

    $(ARDUINO_PATH)/arduino-builder -compile -logger=machine \
    -hardware "$(ARDUINO_PATH)/hardware" \
    -hardware "$(SKETCHBOOK)/hardware" \
    -tools "$(ARDUINO_PATH)/tools-builder" \
    -tools "$(ARDUINO_PATH)/hardware/tools/avr" \
    -built-in-libraries "$(ARDUINO_PATH)/libraries" \
    -libraries "$(SKETCHBOOK)/libraries" \
    -fqbn=esp8266com:esp8266:generic:UploadTool=esptool,CpuFrequency=80,FlashFreq=40,FlashMode=dio,UploadSpeed=115200,FlashSize=512K64 \
    -ide-version=10606 \
    -build-path "$(TARGET_DIR)" \
    -warnings=none \
    -prefs=build.warn_data_percentage=75 \
    -verbose "$(SKETCH)"

flash:
    $(ESPTOOL) -v -cd ck -cb 115200 -cp $(MONITOR_PORT) -ca 0x00000 -cf $(TARGET_DIR)/$(SKETCH).bin

upload: all flash

clean:
    rm -rf $(TARGET_DIR)

monitor:
    screen $(MONITOR_PORT) 9600

steve-superiontech, 2015-11-12:
It appears to work if I use the full path to the build directory instead of a relative path from where the program is executed. In other words, I replaced this line in the build script:

-build-path=./tmp_build_dir \

With the following:

-build-path=`pwd`/tmp_build_dir \

Weird.

From: Steve Phillips
Date: Thursday, November 12, 2015 at 5:32 PM
To: esp8266/Arduino, esp8266/Arduino
Subject: Re: [Arduino] Use without Arduino IDE / arduino-builder problem? (#958)

Thanks for that. I’ve tried using this as a template, though I’m not doing esp8266. I now get a different error:

===Running: {0} ||| [PrintUsedAndNotUsedLibraries]
===Running: {0} ||| [PrintUsedLibrariesIfVerbose]
Rel: can't make tmp_build_dir/core/core.a relative to tmp_build_dir
*errors.errorString Rel: can't make tmp_build_dir/core/core.a relative to tmp_build_dir
/home/jenkins/jenkins/jobs/arduino-builder-linux64/workspace/src/arduino.cc/builder/utils/utils.go:278 (0x469e82)
/home/jenkins/jenkins/jobs/arduino-builder-linux64/workspace/src/arduino.cc/builder/phases/linker.go:55 (0x4c2fe9)
/home/jenkins/jenkins/jobs/arduino-builder-linux64/workspace/src/arduino.cc/builder/builder.go:155 (0x440159)
/home/jenkins/jenkins/jobs/arduino-builder-linux64/workspace/src/arduino.cc/builder/builder.go:115 (0x43f555)
/home/jenkins/jenkins/jobs/arduino-builder-linux64/workspace/src/arduino.cc/builder/builder.go:186 (0x440709)
/home/jenkins/jenkins/jobs/arduino-builder-linux64/workspace/main.go:315 (0x403554)
/opt/go/src/runtime/proc.go:63 (0x415bf3)
/opt/go/src/runtime/asm_amd64.s:2232 (0x43bad1)

Where tmp_build_dir is my TARGET_DIR.

My build script is:

$ cat foo
rm -r ./tmp_build_dir
mkdir ./tmp_build_dir
arduino-builder \
-logger=machine \
-hardware=$ARDUINO_ROOT/hardware \
-tools=$ARDUINO_ROOT/tools-builder \
-tools=$ARDUINO_ROOT/hardware/tools/avr \
-fqbn=arduino:avr:uno \
-built-in-libraries=$ARDUINO_ROOT/libraries \
-build-path=./tmp_build_dir \
-ide-version=10606 \
-prefs=build.warn_data_percentage=75 \
-debug-level=10 \
Digital_IO.ino

From: Simon John
Reply-To: esp8266/Arduino
Date: Thursday, November 12, 2015 at 3:18 PM
To: esp8266/Arduino
Cc: Steve Phillips
Subject: Re: [Arduino] Use without Arduino IDE / arduino-builder problem? (#958)

@steve-superiontech<https://github.com/steve-superiontech> i believe the issue is with 1.6.5 upgrades - nuke your arduino install (including ~/.arduino15) and install 1.6.6 from scratch, works for me using my Makefile:

ARDUINO_PATH = $(HOME)/arduino-1.6.6
SKETCHBOOK   = $(HOME)/sketchbook
ESPTOOL      = $(SKETCHBOOK)/hardware/esp8266com/esp8266/tools/esptool/esptool
SKETCH       = $(notdir $(CURDIR)).ino
TARGET_DIR   = $(CURDIR)/build-esp8266
MONITOR_PORT = /dev/ttyUSB0

all:
    @ mkdir -p $(TARGET_DIR)

    $(ARDUINO_PATH)/arduino-builder -compile -logger=machine \
    -hardware "$(ARDUINO_PATH)/hardware" \
    -hardware "$(SKETCHBOOK)/hardware" \
    -tools "$(ARDUINO_PATH)/tools-builder" \
    -tools "$(ARDUINO_PATH)/hardware/tools/avr" \
    -built-in-libraries "$(ARDUINO_PATH)/libraries" \
    -libraries "$(SKETCHBOOK)/libraries" \
    -fqbn=esp8266com:esp8266:generic:UploadTool=esptool,CpuFrequency=80,FlashFreq=40,FlashMode=dio,UploadSpeed=115200,FlashSize=512K64 \
    -ide-version=10606 \
    -build-path "$(TARGET_DIR)" \
    -warnings=none \
    -prefs=build.warn_data_percentage=75 \
    -verbose "$(SKETCH)"

flash:
    $(ESPTOOL) -v -cd ck -cb 115200 -cp $(MONITOR_PORT) -ca 0x00000 -cf $(TARGET_DIR)/$(SKETCH).bin

upload: all flash

clean:
    rm -rf $(TARGET_DIR)

monitor:
    screen $(MONITOR_PORT) 9600

marvinroger, 2015-11-12:
@sej7278 no offense! :) Actually I did see it somewhere: [on your blog](http://blog.the-jedi.co.uk/tag/esp8266/)!

sej7278, 2015-11-13:
@marvinroger - lol i forgot i put it on there too! so people do read my blog, i was starting to think it was becoming my personal wiki :smile: 

@steve-superiontech - yes i found that relative paths seem to break arduino-builder, it was reported [here](https://github.com/arduino/arduino-builder/issues/21) but that's only for libraries, i found it broke pretty much everything, which is why i made the Makefile really, got fed up of typing huge paths all the time.