esp8266/Arduino #959: OTA password

jeffryr, 2015-11-03:
When trying to upload a sketch via OTA from the IDE, the following dialog box pops up:
```
Type board password to upload a new sketch
```
Clicking on Upload without entering a password results in the following error message:

```
Sketch uses 259,828 bytes (59%) of program storage space. Maximum is 434,160 bytes.
Global variables use 39,458 bytes (48%) of dynamic memory, leaving 42,462 bytes for local variables. Maximum is 81,920 bytes.
processing.app.debug.RunnerException: Unable to connect to 192.168.100.59
	at cc.arduino.packages.uploaders.SSHUploader.uploadUsingPreferences(SSHUploader.java:136)
	at cc.arduino.UploaderUtils.upload(UploaderUtils.java:78)
	at processing.app.Sketch.upload(Sketch.java:1178)
	at processing.app.Sketch.exportApplet(Sketch.java:1152)
	at processing.app.Sketch.exportApplet(Sketch.java:1124)
	at processing.app.Editor$DefaultExportHandler.run(Editor.java:2452)
	at java.lang.Thread.run(Thread.java:745)
Unable to connect to 192.168.100.59
```
Trying to access the monitor via OTA a similar dialog box pops up.  This may be related to the REST API used by the Arduino Yun which by default is password protected.

Links2004, 2015-11-04:
its look like you using the Version 1.6.6 of the Arduino IDE,
in this version they change the OTA handling to SSH which is not possible with the ESP.
try to use version 1.6.5.

jeffryr, 2015-11-04:
This was with Version 1.6.5-r5 on linux, I'll try it with 1.6.5 and let you know.

jeffryr, 2015-11-04:
Oops, my bad, errant symbolic link was left pointing to some random flavor of 1.6.6.

jeffryr, 2015-11-05:
Tried it with 1.6.5 as well:
```
head -1 arduino-1.6.5/revisions.txt
ARDUINO 1.6.5 - 2015.06.15
cd arduino-1.6.5/hardware/esp8266com/esp8266
git log | head -6
commit deb68987face14f29553127c6f15223f0daf173f
Merge: 49bc812 d57315b
Author: Ivan Grokhotkov <igrr@users.noreply.github.com>
Date:   Wed Nov 4 18:53:03 2015 +0300

    Merge pull request #955 from pgollor/master
```
Still prompts for a password.  Clicking on Upload without entering a password results in the following error message:
```
Unable to connect: wrong password?
```

webguy16, 2015-11-05:
1.6.5/Windows10, the password box shows up for me only when i try to use to the serial monitor, clicking 'upload' (on serial monitor password prompt) generates the same error.

me-no-dev, 2015-11-08:
Serial Monitor for networked ESPs will not work yet. Arduino uses SSH for that and changes to the IDE are necessary. If the Arduino Team accepts out changes to enable OTA with the Upload button, next we will push one that will enable SerialMonitor over sockets so that we can use it on the ESP. Hopefully soon it will all work a breeze :)

mralexgray, 2015-11-22:
This issue persists, in even the latest Arduino 1.6.7 nightlies / 2.1.x WIP branches. Tested on Mac OS X.  It's a deal-breaker, ladies!  I don't understand why Arduino is ignoring the upload directives in `platform.txt`.  Also, I am having a hard time reproducing this bug to investigate why.. even though I _couldn't avoid it, earlier_!  Humph.

igrr, 2015-11-22:
This issue persists in nightlies because the pull request hasn't been merged by the Arduino team (yet).

mralexgray, 2015-12-05:
@igrr I don't see an open request re: issue #959  which number is it?

danielfabro, 2015-12-11:
I'm having this same problem. The ultimate, which the solution?

irwanayeah, 2015-12-29:
yes I have same problem. any solution?

me-no-dev, 2015-12-29:
Update to Arduino IDE 1.6.7

danielfabro, 2016-01-07:
I already use IDE 1.6.7.

krzychb, 2016-01-07:
@danielfabro,

Please follow [this example](https://github.com/esp8266/Arduino/blob/master/doc/ota_updates/ota_updates.md#arduinoota) and your should not see any window asking for a "password to upload a new sketch". If there are issues then check [Troubleshooting](https://github.com/esp8266/Arduino/blob/master/doc/ota_updates/ota_updates.md#troubleshooting) section as well.

If this not helps then check [this post](http://www.esp8266.com/viewtopic.php?p=37371#p37371) and the whole following thread for an issue that may be similar to what you have.

Krzysztof

torntrousers, 2016-01-28:
I can get OTA uploads working ok with 1.6.7 but  i wondered about [the comment](https://github.com/esp8266/Arduino/issues/959#issuecomment-154802702) from @me-no-dev back in Nov suggesting Serial Monitor might be made to work remotely too, which would be really cool. It prompts for a password if i try though, is it possible for remote serial monitor to work yet?

me-no-dev, 2016-01-28:
not yet, but i'm thinking about it :) whatever we propose to the Arduino team it needs to be generic and implementable across the boards. While a simple TCP Socket is not a problem, there needs to be an option for authentication and that is the hard part. Have secure authentication that is implementable on all networked arduinos (that do not have the capability to run SSH)

tuxedo0801, 2016-02-25:
From my point of view:

Debug-Code should not be present in production code. So hardcoding wifi credentials etc. just for debugging purpose would be ok for me. But that might conflict with scenarios where wifi is also used for other things and not just debug.

closed 2016-02-27 by igrr

danielperez505, 2016-05-14:
I have the same problem, I found this tutorial the guy successfully connect to the OTA, I will try it, he said that we need to install python I'm not sure if that's the solution. 
http://educ8s.tv/esp8266-ota-wemos-d1/

danielperez505, 2016-06-15:
Here the solution : 
https://github.com/esp8266/Arduino/blob/master/doc/ota_updates/readme.md#password-protection

elgs, 2017-08-01:
Same problem for 1.8.3 installed from homebrew.

danielperez505, 2017-08-02:
Just be sure  that you follow the steps, install phyton and disable your FW

csmcvea, 2018-02-23:
Have the same error
macOS 10.13.3 Arduino IDE 1.8.5
On uploading the error The dialog **Type board password to upload a new sketch**
I enter the password as defined in the BasicOTA script. and get the error

`python /Users/csmcvea/Library/Arduino15/packages/esp8266/hardware/esp8266/2.4.0/tools/espota.py -i 192.168.1.8 -p 8266 --auth=mypass -f /var/folders/2k/qfnbmhxd0plfydqbkrr_phlr0000gn/T/arduino_build_404073/BasicOTA.ino.bin 
Authenticating...FAIL
12:58:59 [ERROR]: ERR: ERROR[11]: Invalid bootstra
`
I enter nothing in the dialog and this error 
`python/Users/csmcvea/Library/Arduino15/packages/esp8266/hardware/esp8266/2.4.0/tools/espota.py -i 192.168.1.8 -p 8266 --auth= -f /var/folders/2k/qfnbmhxd0plfydqbkrr_phlr0000gn/T/arduino_build_404073/BasicOTA.ino.bin 
Authenticating...FAIL
13:03:39 [ERROR]: Authentication Failed`
 The thing is when I first uploaded it ran perfectly, so I figured I'd just tweet the password, the port, the setHostname. And never ran again. I tried flashing the ESP8266 again and loaded the BasicOTA again but this time I only changed the ssid and password. No joy!

If you can help it's much appreciated.
csmcvea

danielperez505, 2018-02-23:
Is the fw in toy computer

csmcvea, 2018-02-24:
UPDATE:
Seems I can upload to esp8266 after I run 
BasicOTA.py
Reset esp8266
upload file to esp8266
this works but is not acceptable because OTA suggests can't reach the esp8266 because it's remote.
Still a conundrum!

csmcvea, 2018-02-24:
@danielperez505 
I don't understand your comment 
`Is the fw in toy computer`
What do you mean toy?
What is `fw`
and if toy means that is esp8266 after I upload BasicOTA.py then how do I check `fw`

devyte, 2018-02-24:
After flashing over serial, you have to physically reset the esp before trying an OTA update. Did you do that?

csmcvea, 2018-02-24:
You can see from my comment earlier:
UPDATE:
Seems I can upload to esp8266 after I run
BasicOTA.py
Reset esp8266
upload file to esp8266
this works but is not acceptable because OTA suggests can't reach the esp8266 because it's remote.
Still a conundrum!

mrWheel, 2019-03-25:
I'm on core 2.5.0 with ArduinoIDE 1.8.8 (macOS)
I have uploaded numerous firmware to a lot of ESP boards. All without problems .. until today :-(
I can upload the basicOTA sketch and then upload the basicOTA or my own sketch using OTA. But the firmware I'm working on will not upload OTA (configuring OTA is exactly the same as the basicOTA) a second time after first uploading the basicOTA. Keep getting the popup to enter a Board Password which I have not set (and even if I set one, it will not work). Occasionally I get a same popup if I try to open the Serial Monitor..
I can though upload OTA using the command line espota.py program. That works all the time.

What am I missing???