esp8266/Arduino #960: ESP8266 - problem with warning msg and uploading a blank sketch

mojgithub, 2015-11-03:
I have a problem with warnings before upload and when uploading blank sketch.

----------------------------
My sistem:
1.6.6 Hourly Build 2015/10/23
Usb Serial RS232 PL2303HX - driver 2008
ESP-13
---------------------------

Even before i press button for upload i got this msg:
1)
Warning: platform.txt from core 'ESP8266 Modules' contains deprecated recipe.ar.pattern="{compiler.path}{compiler.ar.cmd}" {compiler.ar.flags} {compiler.ar.extra_flags} "{build.path}/{archive_file}" "{object_file}", automatically converted to recipe.ar.pattern="{compiler.path}{compiler.ar.cmd}" {compiler.ar.flags} {compiler.ar.extra_flags} "{archive_file_path}" "{object_file}". Consider upgrading this core.
WARNING: Category '' in library OneWire is not valid. Setting to 'Uncategorized'

How to fix this?

2) 
Sketch uses 450 bytes (1%) of program storage space. Maximum is 32,256 bytes.
Global variables use 9 bytes (0%) of dynamic memory, leaving 2,039 bytes for local variables. Maximum is 2,048 bytes.

This is normal msg.

3) 
Invalid library found in C:\Users\MYNAME\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\1.6.5-947-g39819f0\libraries\esp8266: C:\Users\MYNAME\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\1.6.5-947-g39819f0\libraries\esp8266
... it repeats like 15 times ...

---------------------------------------------------------
WHEN UPLOADING:

When i press button for upload i got this msg:
-Blank sketch
-Generic ESP2866 module
EN(CH_PD), 3.3V --> GND
IO0 --> GND
TX --> RX 
RX --> TX
IO15 -->GND

1)
Warning: platform.txt from core 'ESP8266 Modules' contains deprecated recipe.ar.pattern="{compiler.path}{compiler.ar.cmd}" {compiler.ar.flags} {compiler.ar.extra_flags} "{build.path}/{archive_file}" "{object_file}", automatically converted to recipe.ar.pattern="{compiler.path}{compiler.ar.cmd}" {compiler.ar.flags} {compiler.ar.extra_flags} "{archive_file_path}" "{object_file}". Consider upgrading this core.
WARNING: Category '' in library OneWire is not valid. Setting to 'Uncategorized'
WARNING: Category '' in library EEPROM is not valid. Setting to 'Uncategorized'
WARNING: Category '' in library ESP8266httpUpdate is not valid. Setting to 'Uncategorized'
WARNING: Category '' in library Hash is not valid. Setting to 'Uncategorized'
WARNING: Category '' in library SPI is not valid. Setting to 'Uncategorized'
WARNING: Category '' in library Ticker is not valid. Setting to 'Uncategorized'
WARNING: Category '' in library Wire is not valid. Setting to 'Uncategorized'

2)
Sketch uses 201,034 bytes (46%) of program storage space. Maximum is 434,160 bytes.
Global variables use 44,592 bytes (54%) of dynamic memory, leaving 37,328 bytes for local variables. Maximum is 81,920 bytes.

3)
warning: espcomm_sync failed
error: espcomm_open failed

4)
Invalid library found in C:\Users\MYName\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\1.6.5-947-g39819f0\libraries\esp8266: C:\Users\MYName\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\1.6.5-947-g39819f0\libraries\esp8266

..repeats 3 times

what should i do ...

pgollor, 2015-11-04:
I had some similar problems and my solution was to use the latest stabile Arduino 1.6.5 build. maybe you can give it a try?

mojgithub, 2015-11-04:
ok, i have tried 1.6.5 version and get this msg when i try to upload a blank:

Sketch uses 201,034 bytes (46%) of program storage space. Maximum is 434,160 bytes.
Global variables use 44,592 bytes (54%) of dynamic memory, leaving 37,328 bytes for local variables. Maximum is 81,920 bytes.

warning: espcomm_sync failed
error: espcomm_open failed

--------------------------
Usb Serial RS232 PL2303HX - driver 2008
ESP-13
EN(CH_PD), 3.3V --> GND
 IO0 --> GND
 TX --> RX 
 RX --> TX
 IO15 -->GND

Links2004, 2015-11-04:
check you wiring:

https://github.com/esp8266/Arduino/blob/master/doc/boards.md#minimal-hardware-setup-for-bootloading-and-usage

CH_PD shut be have a pullup to 3V3.

connecting the Pins direct to 3V3 or GND is not a good idea we have seen more then once problems with this.

mojgithub, 2015-11-04:
error: failed reading byte
warning: espcomm_send_command: cant receive slip payload data
error: failed reading byte
warning: espcomm_send_command: cant receive slip payload data
error: failed reading byte
warning: espcomm_send_command: cant receive slip payload data
error: failed reading byte
warning: espcomm_send_command: cant receive slip payload data
error: failed reading byte
warning: espcomm_send_command: cant receive slip payload data
error: failed reading byte
warning: espcomm_send_command: cant receive slip payload data
error: failed reading byte
warning: espcomm_send_command: cant receive slip payload data
error: failed reading byte
warning: espcomm_send_command: cant receive slip payload data
error: failed reading byte
warning: espcomm_send_command: cant receive slip payload data
warning: espcomm_sync failed
error: espcomm_open failed

pgollor, 2015-11-04:
Try to check thr folloeing points:
* your esp8266 runs in uploading mode or your programming device does it
* your UART Adapter works correctly
* your 3,3V powersupply delivers enough power

Vor you could try an ftdi UART USB adapter. I had some trouble with the PL23... series

Links2004, 2015-11-04:
do you see the bootloader messages @ 74880Baud when you reset the ESP?

for manually boot loading it shut look like this:
```
ets Jan 8 2013,rst cause:2, boot mode:(1,7)
```

the normal boot:
```
ets Jan 8 2013,rst cause:2, boot mode:(3,7)

wdt reset
load 0x40100000, len 30000, room 16
tail 0
```

mojgithub, 2015-11-04:
where can i see that?

Links2004, 2015-11-04:
open a serial terminal at 74880 Baud.
there are many tools you can use for this, for example RealTerm, Hterm, Putty...

mojgithub, 2015-11-05:
I have managed to upload my sketch(blink example works)!

i used this link to hardwire setup (ESP to Serial) https://github.com/esp8266/Arduino/blob/master/doc/boards.md#minimal-hardware-setup-for-bootloading-and-usage 

Thank all for help.
-----------------------------------------------------------------------------------------

Another question(ESP - AT commands):

I would like to check command "AT" in serial monitor but it seems that nothing happens.
I have removed IOO and turned off on power.
I have opened my SMonitor and set bound to (Bound rate i have tried: 9600, 115200)
I typed "AT" and press send

what i am missing here?

Links2004, 2015-11-05:
AT commands has noting to do with the ESP Arduino port :)

igrr, 2015-11-05:
For the further discussion of AT commands, please use esp8266.com forum.
Make sure you flash AT firmware to the module. Uploading the sketch to the module replaces AT firmware.

closed 2015-11-05 by igrr

xtalker, 2015-11-12:
I had similar problems with:

Invalid library found in C:\Users\MYName\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\1.6.5-947-g39819f0\libraries\esp8266: 

but downgrading the Arduino IDE from 1.6.6 to 1.6.5 seems to have solved them.

Brisdo, 2016-11-03:
This might explain the issue for some folks:
https://www.bountysource.com/issues/29336127-one-wire-library-still-in-v2-3-0-causes-useless-warning