esp8266/Arduino #962: ESP8266-12E with  Ethernet lib (w5100) or enc28j60

funvill, 2015-11-04:
Board: ESP8266-12E using a NodeMCU v1.0 dev board. 
Library: ESP8266 Arduino library version: 1.6.5-947-g39819F0
IDE: Arduno 1.6.6 or visual studios with visual micro

I am following up on this issue closed but incomplete issue. 
https://github.com/esp8266/Arduino/issues/574

I was able to alter the EthernetClass::begin function and compile.
https://github.com/esp8266/Arduino/issues/574#issuecomment-123416888

But when the board runs i get the following error on the serial port. 

    Exception (28):
    epc1=0x40204e27 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000
    ctx: cont 
    sp: 3ffe9db0 end: 3ffea090 offset: 01a0
    >>>stack>>>
    3ffe9f50:  00000000 3ffe8ea0 3ffe8000 4020362c  
    3ffe9f60:  00000000 00000000 00000000 00000000  
    3ffe9f70:  3ffe9fa8 3ffe9fa0 3ffe9fb0 00000000  
    3ffe9f80:  00000000 00000000 00000000 3ffea008  
    3ffe9f90:  3ffea000 3ffe8ea0 3ffe8000 402036d8  
    3ffe9fa0:  3ffe8ad0 00ffffff 3ffe8ad0 0101a8c0  
    3ffe9fb0:  3ffe8ad0 0101a8c0 3ffe8ad0 b101a8c0  
    3ffe9fc0:  3ffe8ad0 00ffffff 00000000 00000000  
    3ffe9fd0:  3ffea010 00000000 00000000 00000000  
    3ffe9fe0:  00000000 00000000 00000000 3ffea0bc  
    3ffe9ff0:  3fffdc20 3ffe8ea0 3ffe9034 40203718  
    3ffea000:  3ffe8ad0 0101a8c0 3ffe8ad0 0101a8c0  
    3ffea010:  3ffe8ad0 b101a8c0 3ffe9034 0101a8c0  
    3ffea020:  3ffe8b18 00000000 3ffe9034 40203742  
    3ffea030:  3ffe8ad0 0101a8c0 3ffe8ad0 b101a8c0  
    3ffea040:  00002580 0101a8c0 3ffe9034 4020631c  
    3ffea050:  3fffdc20 00000000 3ffe9034 40201d40  
    3ffea060:  3ffe8ad0 b101a8c0 00000000 00000000  
    3ffea070:  00000000 00000000 3ffea0b4 40201863  
    3ffea080:  00000000 00000000 3ffe9070 40100118  
    <<<stack<<<
 
I added in some serial port debug messages and I have tracked down this exception to line 44 of the WebServer example.

    Ethernet.begin(mac, ip);

I started breaking it down line by line by coping code out of the Ethernet.begin() function into the Arduino sketch and adding serial port debug messages. 

    W5100Class::init()

At this point its out of my depth and I am not sure where to go from here.

Should I keep trying to get the  Ethernet lib (w5100) working with the ESP8266 chip or should I abandoned it?
Has anyone got this working with a different Ethernet chip?

pgollor, 2015-11-04:
I'm not sure if i understand your Problem. But why do you want to use another Ethernet library with the esp8266?
We habe a own Ethernet class for the esp8266.

Links2004, 2015-11-04:
you have a null ptr problem ```excvaddr=0x00000000```

you have done all changes like here?
https://github.com/Links2004/Arduino/commit/10c5dec92a00c544a4b9f800bba590fc309fa016

funvill, 2015-11-04:
> But why do you want to use another Ethernet library with the esp8266?

In this case I want to query a device that only has a wired Ethernet connection for information and make it available to a wifi connected device via a softAP on the ESP. Doing this all without a router on site. 

> you have a null ptr problem excvaddr=0x00000000
> you have done all changes like here? Links2004@10c5dec

Thank you that worked for compiling the code. I feel foolish, I didn't see the code changes until you pointed them out to me. 

I will report back on the functionality today. 
Thank you.

pgollor, 2015-11-04:
Ah okay. A ethernet wifi bridge would be very nice. :-)

funvill, 2015-11-05:
Nothing good to report i am afraid. 
I am using the updates to the Ethernet library found in this commit Links2004@10c5dec 

I am using a unknown version of the Arduino Ethernet shield that uses a Wiznet W5100 chip.
https://www.arduino.cc/en/Main/ArduinoEthernetShield

I am using the following test code to help me debug the Ethernet library 
https://gist.github.com/funvill/6d5d5a3b3026f7551241

This code causes the esp8266 to crash and restart. 

    localIP=4294967295 subnetMask=4294967295 subnetMask=4294967295
    begin... ret=3
    , Socket ID=0 Status=255, Socket ID=1 Status=255, Socket ID=2 Status=255, Socket ID=3 Status=255
    Sending...
    beginPacket... ret=1
    write... ret=11

    ctx: cont 
    sp: 3ffe9e40 end: 3ffea0c0 offset: 01b0

    >>>stack>>>
    3ffe9ff0:  00000801 00008000 3ffe8eb4 40205463  
    3ffea000:  0000000b 3ffe8eb4 00000004 402051cc  
    3ffea010:  000f4240 00000001 3ffe8ab9 40205743  
    3ffea020:  40207064 00000031 00000004 4020511d  
    3ffea030:  0000000b 00000001 3ffe9064 402064a5  
    3ffea040:  00000000 00000000 3ffe9064 3ffe8e60  
    3ffea050:  0000000b 3ffea0ec 402017ce 3ffea0ec  
    3ffea060:  0000000b 3ffe8e60 3ffe9064 402044a0  
    3ffea070:  3ffe8a70 3ffe8e60 3ffe9064 40202367  
    3ffea080:  3fff4048 0000000f 0000000f 3fff4030  
    3ffea090:  00000002 00000002 00000000 3ffea0ec  
    3ffea0a0:  3fffdc20 00000000 3ffea0e4 4020184a  
    3ffea0b0:  00000000 00000000 3ffe90a0 40100118  
    <<<stack<<<

I don't think that the ESP8266 is correctly communicating with the Wiznet W5100 chip. I suspect this because the Ethernet library can not seem retrieve an IP address from the Wiznet W5100 chip

After looking though the changes made to the Ethernet library I see that the SPI clock is set to a fixed number instead of a clock frequency divide. 

    #define SPI_ETHERNET_SETTINGS SPISettings(14000000, MSBFIRST, SPI_MODE0)

Why is this not a clock frequency divide instead of a magic number? 
Why was 14000000 chosen?

I also see that its different then the default 4000000 that is used by the Arduino default code. 

I am novice using a scope but I checked the SCK out from the ESP chip and found that the frequency was distorted and didn't look right to me. 
http://imgur.com/3xkvsX1

So I set the clock frequency to '''1000000''' the Arduino SPI library default. This produced a much cleaner data in the scope. 
http://imgur.com/Og3K6wy

I also tried a few other frequency all producing the same result. 

I then tested the SPI bus on the ESP8266 to ensure that it is working. I set up an Arduino to send out a message and use the ESP8266 to receive the message and print it the message. This worked with no issues, I also tried the reversed of this where the ESP8266 send the message and the Arduino received it. No issues. 

I started reading the Wiznet W5100 datasheet https://www.sparkfun.com/datasheets/DevTools/Arduino/W5100_Datasheet_v1_1_6.pdf looking for the simplest command I could send from the ESP8266 to the Wiznet W5100 and get a response. I am not an expert in SPI bus and I am struggling with this.

I am still working on this but I am not having much success.

Links2004, 2015-11-05:
the max clock Frequenz for the W5100 is 70ns --> 14.286 Mhz
[image]

that the ESP not output the exact frequency that is configured is normal, the SPI code tries to come as close as possible to the frequency with the internal clock dividers. 

you debug out looks like there is only 0xFF coming back from the W5100

have you net get a line like 
```
Exception (28):
epc1=0x40204e27 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000
```
when it stops working?

you can try to use:
https://gist.github.com/Links2004/341307404b135e0277f8

the Exception lines give you the exact address in the code where the problem is.
to get a file to match the hex address to code run:
```
xtensa-lx106-elf-objdump -S arduino_file.elf 
```

funvill, 2015-11-05:
> the bad looking signal may comes from you logic analyzer if I understand to GUI right you sample at 8000K samples per second. with this config its impossible to see the 14Mhz signal in a good way.
you need at minimum 14000K samples better 30000K samples.

I agree with you, I also set up a test sketch to ensure that the SPI bus is working on the esp8266 using a loop back connection (MISO <==> MOIS) 
https://gist.github.com/funvill/e1399a0928034671803f

> you debug out looks like there is only 0xFF coming back from the W5100

Agreed. I am not sure if the W5100 is actually sending the 0xFF or if the line is just being held high and the ESP detecting that value. 

> have you not get a line like 
>> Exception (28):
epc1=0x40204e27 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000 

> when it stops working?

No I have not. 

I reverted my changes to my Ethernet library and I am using your example code.  https://gist.github.com/Links2004/341307404b135e0277f8 

I get the following output on the serial port. 

	[SETUP] Reset Info: Fatal exception:4 flag:3 (SOFT_WDT) epc1:0x4020251b epc2:0x00000000 epc3:0x00000000 excvaddr:0x00000000 depc:0x00000000
	localIP=4294967295 subnetMask=4294967295 subnetMask=4294967295
	begin... ret=3
	, Socket ID=0 Status=255, Socket ID=1 Status=255, Socket ID=2 Status=255, Socket ID=3 Status=255
	Sending...
	beginPacket... ret=1
	write... ret=11

	ctx: cont 
	sp: 3ffe9fe0 end: 3ffea260 offset: 01b0

	>>>stack>>>
	3ffea190:  00000801 00008000 3ffe9034 40205577  
	3ffea1a0:  0000000b 3ffe9034 00000004 402052e0  
	3ffea1b0:  00d59f80 00000001 3ffe8c49 40205857  
	3ffea1c0:  40207ab4 00000031 00000004 40205231  
	3ffea1d0:  0000000b 00000001 3ffe91ec 40206b25  
	3ffea1e0:  00000000 00000000 3ffe91ec 3ffe8ff8  
	3ffea1f0:  0000000b 3ffea28c 402017ce 3ffea28c  
	3ffea200:  0000000b 3ffe8ff8 3ffe91ec 402045b4  
	3ffea210:  00000001 3ffe8ff8 3ffe91ec 40202477  
	3ffea220:  3fff41d8 0000000f 0000000f 3fff41c0  
	3ffea230:  00000002 00000002 00000000 3ffea28c  
	3ffea240:  3fffdc20 00000000 3ffea284 4020184a  
	3ffea250:  00000000 00000000 3ffe9240 40100118  
	<<<stack<<<

I do not know how to read this new line in the serial buffer. Is there a tool that will convert it from the hex address to a line in code? 

> Reset Info: Fatal exception:4 flag:3 (SOFT_WDT) epc1:0x4020251b epc2:0x00000000 epc3:0x00000000 excvaddr:0x00000000 depc:0x00000000

I have ordered more ESP8266 to ensure that there is no issues with the board I currently have.

Links2004, 2015-11-05:
the soft watch dog is triggert (SOFT_WDT) the pace ist was trigger is 0x4020251b (memory address)
to finde the code that is at this address you need to run 
```
xtensa-lx106-elf-objdump -S <filename_here>.elf
```
the elf file is int the temp dir from the arduino IDE.

but its most likely a while which is waiting for a correct status.

the bigger problem is the 0xFF reading, always 0xFF is not good.
can you compare the AVR and ESP SPI signal with the logic analyzer?

funvill, 2015-11-05:
I got it working! This was all my fault and bad documentation on Arduino's site. No changes need to be made to the esp8266 Arduino project. 

I am using the Arduino Ethernet Shield for the W5100 chip break out. I had connected to the SPI pins on the right hand side of the board, instead of the SPI header under the Ethernet board. 
[image]

The documentation on the Arduino Ethernet Shield reports that it is using (pins 10, 11, 12, and 13 on the Uno) this is NOT true, The Arduino Ethernet Shield actually doesn't connect (11, 12, and 13 ) and uses the SPI header instead. I figured this out by loading the schematics into eagle and viewing the board layout. 
https://www.arduino.cc/en/Main/ArduinoEthernetShield

I will write up a good post on this entire process and post it to my blog. Then I will add a link to this issue for people that have issues with this in the future. 

Thank you for your help. 
I will get my employer to donate some money to this project.

Links2004, 2015-11-05:
so the library is working with the changes like this: https://github.com/Links2004/Arduino/commit/10c5dec
if yes I will add it to the repository so every one can simply use it :)

pgollor, 2015-11-05:
@funvill Could you post your example sketch with your "mini router" code? :)

funvill, 2015-11-05:
> so the library is working with the changes like this: Links2004@10c5dec

Yes

> Could you post your example sketch with your "mini router" code? :)

I will post as much of it as I can. 
I will have to leave out the section that deals with communicating with my proprietary device. This may take me one or two weeks to develop.

pgollor, 2015-11-05:
@funvill Okay, previously much thanks for it.

Vikingfr, 2015-11-12:
Hi,

main difference between SPI port and direct pins: CS (chip select) is PIN 14 on SPI header, not PIN 10.

closed 2015-11-15 by igrr

mkeyno, 2016-05-15:
@funvill  any success to bind ESP & w5100

FernandoGarcia, 2016-11-04:
Hi @funvill @Links2004! 

Is possible give us a working basic example for this modified library?
The default examples doesn't work without changes.

Best regards.

littleoz2, 2017-02-03:
Hello, has anyone got this working?, been attempting this and the Ethernet.begin just returns 0, any help would be much appreciated

tusharvb19, 2017-04-20:
Has anyone successful with Wiznet W5100 arduino shield and NodeMCU(ESP8266 12E)?
Can someone give the details of hardware configuration?
Help needed.

tusharvb19, 2017-07-19:
I got ping working but the example code web-server is not!(sic) 

I have used the normal SPI pins on NodeMCU which are:

#define D0 16
#define D1 5 // I2C Bus SCL (clock)
#define D2 4 // I2C Bus SDA (data)
#define D3 0
#define D4 2 // Same as "LED_BUILTIN", but inverted logic
#define D5 14 // SPI Bus SCK (clock)
#define D6 12 // SPI Bus MISO
#define D7 13 // SPI Bus MOSI
#define D8 15 // SPI Bus SS (CS)
#define D9 3 // RX0 (Serial console)
#define D10 1 // TX0 (Serial console)

When I fired up browser I am not getting the desired webpage!
for the ip 192.168.0.177 I got this on serial monitor(for every reset 1 line):
⸮⸮=s⸮⸮⸮server is at 0.168.0.0
⸮(⸮0r⸮⸮⸮server is at 0.168.0.0
$⸮%⸮⸮5Q⸮⸮⸮server is at 0.168.0.177
⸮!⸮⸮-2⸮⸮⸮server is at 0.168.0.177
⸮⸮=s⸮⸮⸮server is at 192.168.0.240
⸮'⸮⸮=1⸮⸮⸮server is at 0.168.0.177
$⸮*⸮⸮5Q⸮⸮⸮server is at 192.168.0.177

tusharvb19, 2017-07-19:
Also sometimes it is causing soft wdt reset while handling data with the browser!

devyte, 2017-07-20:
How long is each transaction taking? There's a limit to how long you can be
away from the wifi maintenance. Try some yield()s here and there.
What does mem usage look like? If your eth interface requires additional
buffers, you could be running out of heap.
What is your worse case stack usage? There's a limit to how much you can
allocate there.

Awesome progress, I suspect you need just some tweaking to get it stable.