esp8266/Arduino #963: MFRC522

asetyde, 2015-11-04:
I'VE OPEN ON GITHUB LIBRARY ABOUT COMPABILITY WITH ESP8266

I think also spi code is not fully ok , because often esp need many reboot to flash or sometimes both not work correctly.

Links2004, 2015-11-04:
the SPI library has noting to do with the flash process or the SPI flash access.
the flash process is handled by a bootloader that is hardcoded in the Hardware.
the SPI flash access is handled by the SDK.

the SPI of the ESP not supports Mode 2 or 3 may this is part of your problem.
see: https://github.com/esp8266/Arduino/blob/master/doc/reference.md#spi

asetyde, 2015-11-04:
ok but at now if i do a simply sketch , work but with other library DHT , use with eeprom and common espwifi give problems. Reference is a bit poor

Links2004, 2015-11-04:
the SPI works link the one for the AVR, excepted the written Mode problem.
if you want to see all possible functions take a look here:
http://links2004.github.io/Arduino/da/d4e/class_s_p_i_class.html

this one for DHT https://github.com/adafruit/DHT-sensor-library will work with the ESP.

the EEPROM usage is a bit different from the AVR since we not have a real EEPROM on the ESP we emulate one.

asetyde, 2015-11-04:
Ok DHT and eeprom it’ok .

About rfid i want only read , i will try to make it work next week

thk Markus

asetyde, 2015-12-17:
on holidays i'll try , help wanted

ricardojlrufino, 2016-05-29:
please close it is resolved

closed 2016-10-28 by asetyde