esp8266/Arduino #965: Arduino 1.6.6 shows  Error after installing ESP-Boards

tommyfranke, 2015-11-04:
After installing the ESP-Stable Package with Boardmanager in the new Arduino IDE 1.6.6 it shows the following Error:
Warning: platform.txt of Kernel 'ESP8266 Modules' contains abandoned recipe.ar.pattern

If i load any script it shows five times this error:
Invalid Library D:\Arduino\runtimeIDE\arduino-1.6.6\portable\packages\esp8266\hardware\esp8266\1.6.5-947-g39819f0\libraries\esp8266 at D:\Arduino\runtimeIDE\arduino-1.6.6\portable\packages\esp8266\hardware\esp8266\1.6.5-947-g39819f0\libraries\esp8266 detected

I can't compile anything because ot these Errors. With Arduino 1.6.5 everything was fine!!!?????

Links2004, 2015-11-04:
1.6.6 is currently not supported.

linked to: #959 #948

asetyde, 2015-11-04:
i use 1.6.8 .. but work , therefore is good idea come back to 1.6.5 ?

Links2004, 2015-11-04:
if its work for your cases you can use it, but OTA for example is sure broken.
the ```esp8266``` library is no real library is only used for provide some examples to the user.
so the ide complains may about the missing ```library.properties``` and that there is no src files.

asetyde, 2015-11-04:
Thx Markus but i want more stability as possible , i ll come back.

tommyfranke, 2015-11-04:
Ooooooh pain, thanx for the answer. Would be good to anounce that Issue to save others time !!!!!

igrr, 2015-11-04:
Readme has quite specific instructions about IDE version — it says "Install
Arduino 1.6.5 from the Arduino website."

asetyde, 2015-11-04:
i use board manager to install version and library , to make code visualstudio with visual micro

tommyfranke, 2015-11-05:
@igrr  Mea Culpa, sorry. I thought it is supported since the Boardmanager appeared in 1.6.5 and then on and on...

igrr, 2015-11-05:
Keeping this open for now, because it has to be fixed anyway.

tommyfranke, 2015-11-06:
Seems like Esp8266 is not the only core that suffers from that new version. Core 'Arduino ARM (32-bits) Boards' do also, see post 18 from Arduino Forum:
https://forum.arduino.cc/index.php?topic=357312.15

marvinroger, 2015-11-06:
About the warnings:

- The category WARNINGS are fixed upstream since 49bc812ccd4d466dcce0e67aa095242c8e228707 and cc828e34a38ff183aa0f0ad1adba4cd7a1a4fe9d
- The invalid library warnings can be fixed easily by putting a `library.properties` containing for example:

```
name=ESP8266
version=1.0
author=Ivan Grokhotkov
maintainer=Ivan Grokhtkov <ivan@esp8266.com>
sentence=ESP8266 sketches examples
paragraph=
category=Other
url=
architectures=esp8266
```
And by creating an `src` folder with a dummy blank header file in it.
The problem is it allows the dev to import the dummy `.h` file from the `Include library` menu. Not sure there's a better way.

- The recipe.ar.pattern warning can be fixed by replacing `recipe.ar.pattern` content with `"{compiler.path}{compiler.ar.cmd}" {compiler.ar.flags} {compiler.ar.extra_flags} "{archive_file_path}" "{object_file}"`.
`archive_file_path` was introduced in https://github.com/arduino/arduino-builder/commit/35bdf4095bb722d3ad9d69844401045f0f485317 and is therefore not backward compatible with 1.6.5.

Links2004, 2015-11-06:
the recipe.ar.pattern change my brake eclipse support too, need to do some test on this.

tommyfranke, 2015-11-06:
@marvinroger Thanx, it works! I just did what you wrote above and now it works. I could compile and upload a scetch i did with IDE 1.6.5. More tests to do, but i think i stay with 1.6.5 for now.

rogerclarkmelbourne, 2015-11-07:
FYI.
I had to change

"{build.path}/{archive_file}"
 to
"{archive_file_path}"

On some other cores (not ESP8266) and this seems to get rid of the warning.

igrr, 2015-11-11:
@rogerclarkmelbourne right, and when you do it the core stops working with older (1.6.5) IDEs.

igrr, 2015-11-11:
Archive file path issue should be fixed by 3c14656. Works with both 1.6.5 and 1.6.6.

rogerclarkmelbourne, 2015-11-11:
@igrr 

Sorry. I ended up backing out the change I suggested after I found it was not compatible with 1.6.5. or older.

Its interesting to see your fix that works for both, as I understand that some organizations (Elektor) started to make 2 different versions, one for 1.6.5. and one for 1.6.6

We have loads of issues with libraries on the STM32 platform when using 1.6.6 so I think we will stay on 1.6.5 until either 1.6.7 or some interim release of 1.6.6

I have too many other things to do, rather than trying to cope with bad design decisions from the IDE team ;-)

agent2003, 2015-11-15:
I'm running IDE 1.6.5 on windows 7. I've copied the esp8266 master files and just used get.py method to install.

I initially got an error that 'boards.txt' could not be found, fixed that by copying boards.text from my esp8266com folder into the 'bootloaders' folder in the esp8266com folder (it said it could not find it in any of the folders in esp8266com folder). I can see that the esp8266 board options have installed in the drop down list.

I'm now getting the error below when i try to compile anything.

"Third-party platform.txt does not define compiler.path. Please report this to the third-party hardware maintainer.

Error while compiling: missing 'recipe.cpp.o.pattern' configuration parameter"

Please can somebody explain marvinroger's fixes in more detail? 

I'm not sure i need the invalid library fix but a search for library.properties brings up a load in "arduino-1.6.5-r5//hardware/esp8266com/libraries" and "arduino-1.6.5-r5//hardware/libraries". Which library.properties should be amended?

Is the src folder with dummy header required for this error? If so, where should this be created and how do I create a dummy header?

For the recipe.ar.pattern warning, I find recipe.ar.pattern in the platform.text files in both the /hardaware/esp8266com folder and /hardware/arduino/avr folder

I find the below recipe.ar.pattern info in the platform.txt in the esp8266com folder:

"## Create archives
recipe.ar.pattern="{compiler.path}{compiler.ar.cmd}" {compiler.ar.flags} {compiler.ar.extra_flags} "{build.path}/core.a" "{object_file}""

and i find this in the avr folder:

## Create archives
recipe.ar.pattern="{compiler.path}{compiler.ar.cmd}" {compiler.ar.flags} {compiler.ar.extra_flags} "{build.path}/{archive_file}" "{object_file}"

I have attached both my platform.text files,from the avr and esp8266com folder.

Thanks in advance.

[platform - avr.txt](https://github.com/esp8266/Arduino/files/34797/platform.-.avr.txt)
[platform - esp8266com.txt](https://github.com/esp8266/Arduino/files/34798/platform.-.esp8266com.txt)

agent2003, 2015-11-15:
My recipe.cpp.o.pattern is below in the esp8266com platform.txt, seems identical to platform.txt on website

## Compile c++ files
recipe.cpp.o.pattern="{compiler.path}{compiler.cpp.cmd}" {compiler.cpreprocessor.flags} {compiler.cpp.flags} -DF_CPU={build.f_cpu} -DARDUINO={runtime.ide.version} -DARDUINO_{build.board} -DARDUINO_ARCH_{build.arch} {compiler.cpp.extra_flags} {build.extra_flags} {includes} "{source_file}" -o "{object_file}"

agent2003, 2015-11-15:
The bottom shows the compiler.path definition in my platform.txt. Unsure if runtime.platform.path is the arduino installation root directory or not. 

The compiler seems to be installed in "arduino-1.6.5-r5\hardware\esp8266com\tools\xtensa-lx106-elf\bin". I've also tried moving compiler files to "arduino-1.6.5-r5-windows\arduino-1.6.5-r5\tools\xtensa-lx106-elf\bin" but no success.

runtime.tools.xtensa-lx106-elf-gcc.path={runtime.platform.path}/tools/xtensa-lx106-elf
runtime.tools.esptool.path={runtime.platform.path}/tools/esptool

compiler.path={runtime.tools.xtensa-lx106-elf-gcc.path}/bin/
compiler.sdk.path={runtime.platform.path}/tools/sdk

igrr, 2015-11-15:
@agent2003 you get all those errors because your installation path is incorrect. Everything should go into hardware/esp8266com/esp8266, not hardware/esp8266com.

closed 2015-11-30 by igrr