esp8266/Arduino #966: 512k (no SPIFFS) breaks OTAs

sven337, 2015-11-04:
OTAs in Updater.cpp are broken with _SPIFFS_start = 0. The following seems to fix them:

-PROVIDE ( _SPIFFS_start = 0 );
-PROVIDE ( _SPIFFS_end = 0 );
+PROVIDE ( _SPIFFS_start = 0x4027B000 );
+PROVIDE ( _SPIFFS_end = 0x4027B000 );

igrr, 2015-11-05:
Thanks, well noted.
I didn't think OTAs were possible with 512k flash at all.

sven337, 2015-11-05:
They are possible - I got them working. But I don't think the program can be very big :)

reaper7, 2015-11-05:
yes, it's works with 512k module, for eg sketch ArduinoOTA->OTALeds
with mod:
PROVIDE ( _SPIFFS_start = 0x4027B000 );
PROVIDE ( _SPIFFS_end = 0x4027B000 ); 
inside eagle.flash.512k0.ld

igrr, 2015-11-05:
New ESP-01 modules come with bigger flash, because AT firmware doesn't fit into 512k any more.
These new modules can be distinguished by black PCB solder mask.

closed 2015-11-05 by igrr