esp8266/Arduino #969: Any reason to stop UDP port when Wifi is reconnected?

polvora, 2015-11-05:
Thanks you all for your work here.

I've been working with DNSServer library in WIFI_AP_STA mode and noticed that everytime it reconnects to an AP my DNS server stopped working since ESP8266WiFi.cpp stops the incomming UDP segments:
```c++
void ESP8266WiFiClass::_eventCallback(void* arg)
{
    System_Event_t* event = reinterpret_cast<System_Event_t*>(arg);
    DEBUGV("wifi evt: %d\r\n", event->event);

    if (event->event == EVENT_STAMODE_DISCONNECTED) {
        WiFiClient::stopAll();
        WiFiUDP::stopAll();
    }
}
```
Is this to prevent anything like a leak or does DNSServer library need a fix?
I ask this because i simply commented the line that stopped udp and it seems to work fine.

pgollor, 2015-11-05:
Could you check if it cause problems if you have an open UDP connection over your client IP and don't stop all UDP connections on reconnect?
If it does not i think we can remove this line.
I hope you understand my question?

Alternatively we have to differ between UDP connections over the STA IP and the UDP connections over the AP IP.

polvora, 2015-11-06:
i connected the ESP module to my home network (WIFI_STA mode) and started listening UDP traffic in the module and in my pc and sent multiple datagrams between them even when both were disconnected from the home network. After multiple reconnections they received the traffic without problems.

[image]

[image]

I used this code to test

```c++
#include <ESP8266WiFi.h>
#include <WiFiUdp.h>
#include <Ticker.h>

const char* ssid     = "GPT2";
const char* password = "otrotipo";

int currentPacketSize;
char* packet;
char temp[1024];

boolean begun, ended;

WiFiUDP udpListen;
WiFiUDP udpSend;
Ticker ticker;

void setup() {
  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid, password);

  Serial.begin(115200);

  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
  }
  
  udpListen.begin(7777);

  ticker.attach(4.0, [](){
    begun = udpSend.beginPacket(IPAddress(192, 168, 1, 101), 55005); // My computer listening
    udpSend.write("ALIVE\n", 6);
    ended = udpSend.endPacket();
    snprintf(temp, sizeof temp, "ALIVE packet : started == %s : sent == %s", begun?"true":"false", ended?"true":"false");
    Serial.println(temp);
  });
}

void loop() {
  currentPacketSize = udpListen.parsePacket();
  if (currentPacketSize){
    packet = (char*) malloc(currentPacketSize * sizeof(char));
    udpListen.read(packet, currentPacketSize);

    Serial.print("Received packet: ");
    Serial.write(packet, currentPacketSize);
    Serial.print("\n");

    // Echo
    begun = udpListen.beginPacket(udpListen.remoteIP(), 55005); // My computer listening
    udpListen.write(packet, currentPacketSize);
    ended = udpListen.endPacket();

    Serial.print("Echo sent to: ");
    Serial.print(udpListen.remoteIP());
    snprintf(temp, sizeof temp, " : started == %s : sent == %s\n", begun?"true":"false", ended?"true":"false");
    Serial.print(temp);
  }
}
```

igrr, 2015-11-06:
TCP pcbs have to be closed because each time WiFi disconnects, interface structure is destroyed, and using the same TCP pcb causes an exception.
Since unlike TCP pcbs, UDP pcbs are not bound to a particular interface, this might not be necessary — I need to check LwIP source code and see.

closed 2015-11-07 by polvora