esp8266/Arduino #970: Detect network port for OTA did not work on windows machine

reaper7, 2015-11-05:
Detect network port for OTA did not work on windows machine
without executed tcp server on port 8266.

solution for me (when ArduinoOTA lib is used):
```
void ArduinoOTA::setup() {
  _udp_ota->begin(_port);
  if (_mdns_host) {
    if (_serial_debug)
      Serial.printf("OTA server at: %s:%u\n",
                    _mdns_host->c_str(),
                    _port);
    MDNS.begin(_mdns_host->c_str());
    MDNS.addService("arduino", "tcp", _port);
    
    WiFiServer TelnetServer(_port);
    TelnetServer.begin();
    TelnetServer.setNoDelay(true);
  }
}
```
with this addon inside ArduinoOTA.cpp port is detected and OTA is usable

webguy16, 2015-11-05:
Thanks. This fixes it for me as well.

igrr, 2015-11-06:
The original idea was to use UDP to listen for the initial handshake, because TCP pcbs are expensive, and there are only 5 of them. Adding this TelnetServer pretty much defeats the purpose :)
It would be nice to find out why this is necessary on windows. Is it Arduino's 'feature' to check if the port is available, or is it an issue with jmdns library which arduino uses?

reaper7, 2015-11-06:
igrr - true, I know...for that this is a issue, not "ready for use" request...
I don't know where is part of code for port detection, where I can find this code? this is python script, java or something else?

igrr, 2015-11-06:
The reason for this issue is here:
https://github.com/arduino/Arduino/blob/master/arduino-core/src/cc/arduino/packages/discoverers/network/BoardReachabilityFilter.java#L72
This seriously breaks our OTA implementation, together with the fact that IDE initiates SSH uploads regardless of the `using_ssh` option now.

mangelajo, 2015-11-10:
Yikes!

closed 2016-02-27 by igrr