esp8266/Arduino #971: Write an FAQ/Troubleshooting document

igrr, 2015-11-05:
A number of issues are popping up periodically, due to common difficulties, lack of documentation, etc.

- "espcomm_sync failed", "no esptool in programmers menu". Need a step-by-step guide to troubleshoot serial connection issues.
- "my ESP crashes running some code". List some common steps to do initial troubleshooting and some info on how to provide a helpful and reproducible issue report.
- "some Arduino library doesn't work". Give a list of some common issues (avr/pgmspace.h, port definitions, usage of AVR intrinsics, etc) and hints on how to fix them.

pgollor, 2015-11-05:
here are some troubleshooting information which works many times ;)
https://github.com/pgollor/esp8266-webconf-mDNS-OTA#troubleshooting

pgollor, 2015-11-06:
Should we use the wiki function from github for the FAQ stuff?

igrr, 2015-11-06:
I would prefer to have it in doc folder inside git, because this way it
will get bundled with HTML version of docs.
But the initial work can happen in wiki, if that's more convenient.

pgollor, 2015-11-06:
For my opinion it is a bit easier to work with the github wiki if you have multiple pages. If we ship a new release we can download the latest markdown file from github wiki and put it into the doc folder.

It's only a thought and it would not be bad if we use directly md files in docs. ;)

Links2004, 2015-11-06:
sine the wiki itself is also a git is easy to move the doc dir simple to it.

and then change the script to make a clone of it.
https://github.com/esp8266/Arduino/blob/master/package/esp8266-arudino-doc.bash#L61

Links2004, 2015-11-06:
igrr has raised an important point. 
If we move the doc to the wiki git then we need to match the versions some how.
for the moment i not have any good idea to handle this in a easy way.

krzychb, 2016-01-20:
@igrr - this is on excellent idea to add FAQ / Troubleshooting :+1:

I will give a shoot to the fist topic from the initial list basing on [real life example](https://github.com/esp8266/Arduino/issues/1471#issuecomment-173150635).

Krzysztof

igrr, 2016-06-08:
FYI, Espressif has posted an $200 bounty on this issue, payable via bountysource.

themindfactory, 2016-06-08:
FAQ and Troubleshooting for what, can not follow this...

RichardS

closed 2016-06-30 by igrr