esp8266/Arduino #973: Check compatibility with Arduino 1.6.6

pgollor, 2015-11-05:
The Arduino community released version 1.6.6 the day before yesterday (2015-11-03).

Could somebody check the compatibility with your code version on your operating system?
If you could, please post the following things:
* operating system (Windows 7,8,10, Linux version, Mac OSX)
* architecture 32bit, 64bit
* Arduino version (stable, staging or the git hash)
* used Arduino package manager or not
* Hardware (ESP8266 Generic, NodeMCU, etc.)???
* the working example sketch
* which works not?

Previously thanks for your posts.

My tests:
* Ubunut 15.04 x64 and 14.04 x64
* Arduino generic board
* no package manager
* version 2f3f1f5a7e6319f796a61f257676366a6550ace5
* `ESP8266WebServer/HelloServer` example sketch

asetyde, 2015-11-06:
sorry but , you test an arduino ? it's not only esp ?

My tests:

Windows 7 & Windows 10 Visual Studio with visual micro with arduino library 1.6.6 , 1.6.7 , 1.6.8 
Nodemcu 1.0 esp826612-e
Stable and Staging boardmanager(latest release )
my project (3) with DHT library 1.1.1 , Mysql library , Wifiudp library  (network/math/eeprom/analoginput/digitalinput) no issue

pgollor, 2015-11-06:
@asetyde thanks for your tests.
No I tested the ESP8266-01. It was my fault. I corrected this.

marvinroger, 2015-11-06:
- **OS:** Ubuntu 14.04
- **Arch:** x64
- **OS:** Ubuntu 14.04
- **Arduino version:** Latest staging
- **Hardware**: NodeMCU v1.0
- **Working sketch**: All
- **Non-working sketch**: None

The only visual problems being the Warnings (fixed upstream), and the invalid library messages.

---

- **OS:** Ubuntu 14.04
- **Arch:** x64
- **OS:** Ubuntu 14.04
- **Arduino version:** ba3af99a04fd015aa9747e6acf2af44dbce4ae33
- **Hardware**: NodeMCU v1.0
- **Working sketch**: All
- **Non-working sketch**: None

The only visual problems being the recipa.ar.pattern warning , and the invalid library messages.

sej7278, 2015-11-07:
what are you testing, 1.6.6 ide totally breaks esp8266 compatibility due to package/platform naming and the use of arduino-builder, that's a known issue.

or do you mean the esp8266 build of the arduino ide (not just the core) not using boards manager?

marvinroger, 2015-11-07:
No it does not. Updating from 1.6.5 to 1.6.6 using the same boards manager files doesn't work. But if you use boards manager with a clean 1.6.6 install, it does work.

sej7278, 2015-11-07:
ah that's interesting, might explain why using the git version just worked for me after i deleted all the packages and staging stuff in ~/.arduino15

so boards manager doesn't even manage to fulfil its entire point - making updating cores easier!

pgollor, 2015-11-07:
I mean the current Arduino 1.6.6 release from arduino.cc

rogerclarkmelbourne, 2015-11-07:
We're having similar issues with 1.6.6 on the STM32 core that I maintain.

One thing that you need to change is 

in platform.txt 

This

```"{build.path}/{archive_file}"

needs to be changed to

```"{archive_file_path}"

marvinroger, 2015-11-07:
This issue is only for test reports, there is a little more detailed issue about this, with your mentionned fix : https://github.com/esp8266/Arduino/issues/965

pgollor, 2015-11-08:
Thanks guys for your posts. So we need some tests with windows and mac os i think.

marvinroger, 2015-11-08:
I will report for Windows soon.

closed 2016-02-27 by igrr