esp8266/Arduino #975: `Failed` typo

marvinroger, 2015-11-06:
There is a redundant typo with the word `failed` misspelled `faild`. See https://github.com/esp8266/Arduino/search?utf8=%E2%9C%93&q=faild

The problem is correcting the constants to `failed` would be a breaking change.
There are two solutions:
- As 2.0.0 is going to be a breaking change anyway, we can make the change now
- Simply let the old `FAILD` constants in place, but also create `FAILED` constants

Let me know, I can make a little PR if you want to.

igrr, 2015-11-06:
We're breaking a number of things anyway, i think this will be a relatively minor issue. If you do a PR, that would be cool.

marvinroger, 2015-11-06:
PR submitted. Just my two cents, I can't contribute much more because of my limited C++ skills.

closed 2015-11-06 by marvinroger