esp8266/Arduino #977: CLI for uploading returns 0 even if upload failed

marvinroger, 2015-11-07:
Using for example `arduino --upload --board esp8266com:esp8266:nodemcuv2 --port /dev/ttyUSB0 sketch.ino` returns 0 even if the upload fails. I don't know if this is a problem from espcomm or from Arduino itself.

igrr, 2015-11-08:
When esptool fails, it returns 255:
```
$ ./esptool -cp /dev/tty.SLAB_USBtoUART -cb 921600 -ca 0 -cf TimeTest.cpp.bin; echo $?
warning: serialport_set_baudrate: baud rate 921600 may not work
warning: espcomm_sync failed
error: espcomm_open failed
error: espcomm_upload_mem failed
255
```
Likely an issue with Arduino itself.

marvinroger, 2015-11-10:
That's strange. The line responsible for upload in the IDE is https://github.com/arduino/Arduino/blob/8e5b3553fc8b17c9bb96b56a5065ba90c7564816/arduino-core/src/processing/app/BaseNoGui.java#L491

The value of success is the one returned [here](https://github.com/arduino/Arduino/blob/435fc323e0ab3b03bc5835e5309b5022bf57388a/arduino-core/src/cc/arduino/packages/Uploader.java#L129), so basically, if exit code of the uploader != 0, call [showError(tr("Error while uploading"), tr("An error occurred while uploading the sketch"), null)](https://github.com/arduino/Arduino/blob/8e5b3553fc8b17c9bb96b56a5065ba90c7564816/arduino-core/src/processing/app/BaseNoGui.java#L1127) which calls [System.exit(1)](https://github.com/arduino/Arduino/blob/435fc323e0ab3b03bc5835e5309b5022bf57388a/arduino-core/src/processing/app/helpers/BasicUserNotifier.java#L18).

The problem doesn't seem to come from the IDE either. I will further test.

marvinroger, 2015-11-10:
The problem is definitely not from esptool. But looking at the Arduino code, I don't find what's wrong.

marvinroger, 2015-11-10:
Issue opened on the Arduino repo: https://github.com/arduino/Arduino/issues/4127

igrr, 2015-11-10:
Perhaps 255 is a 'bad' exit code? Maybe try returning `1` from esptool?

marvinroger, 2015-11-10:
255 seems to be a special exit code meaning "out of range". Maybe Java handles this code in an unconventional way? I guess yes, trying `1` would be a good idea.

marvinroger, 2015-11-10:
Just tried: it doesn't work.

closed 2016-02-29 by igrr