esp8266/Arduino #979: Can't properly reboot esp8266 after reflash.

kommancs96, 2015-11-07:
My configuration is the following:
* ESP8266-01
* Silabs USB to UART 
* Arduino IDE to compile code
* esptool.py to upload the binary at 9600 baud rate.

Connections:
esp8266                      USBtoUART
GND----------------------GND
VCC-----------------------3.3V
TX-------------------------RX
RX-------------------------TX (with voltage divider to 3.3v)
GPIO0--------------------GND
CH_PD-------------------3.3V

I upload my code to the esp8266 and it works fine, I can control an LED from my browser.
However , when I unplug the USB from my laptop and plug it in to a power supply or simply replug to my MacBook , it seems that the chip looses the firmware. After I unplug the chip I also free GPIO0 or connect to high with a resistor.

How do I properly reboot my chip after reflashing firmware?

Links2004, 2015-11-07:
if GPIO0 is at GND the ESP boot always to bootloader.
you need to add add a pull up for normal use.

more info see here:
https://github.com/esp8266/Arduino/blob/master/doc/boards.md#generic-esp8266-modules

devyte, 2017-10-19:
Closing as user error.

closed 2017-10-19 by devyte