esp8266/Arduino #982: OTA Network Port does not Show up for Release 1.6.5-1160-gef26c5f

krzychb, 2015-11-08:
I am using OTA successfully in the following configuration :ok_hand: :
1.	Arduino IDE version: 1.6.5
2.	ESP package release: esp8266-1.6.5-947-g39819f0
3.	Example sketch: DNS_SD_Arduino_OTA.ino 
4.	WiFi Router Thomson TWG870UG

After upgrading ESP package to release **1.6.5-1160-gef26c5f** the OTA network port does not show up in IDE anymore. The usual sketch functionality like joining the WiFi works correctly (but I need to flash using serial :disappointed: ).

To make the network port show up I had to change the router to my old Linksys WRT-150N-EU.

I did it basing on observation of time it takes ESP module to join the WiFi network. The module joins Linksys’ WiFi in about 0.5s. To join Thomson’s WiFi it takes 2 – 3s.

- What has been changed in OTA handling between 947 and 1160 releases that may cause such issue?
- Maybe there is something in router configuration I can tweak?

I am stuck with this issue for second day already and any help is welcome :smile: 

Krzysztof

me-no-dev, 2015-11-08:
Nothing in OTA will do that. The board is discovered by the mDNS service which is a separate thing and has not been changed. As far as network issues go, I have had my fair share of those as well. Even to the point where a smart switch is clogging my connection (not even a router).
If we look at home router on the other hand, they are nothing else but small underpowered computers, so based on the implementation, one router can perform lots better than another one. That is the reason there are custom firmwares like OpenWRT :)

krzychb, 2015-11-08:
Wholly carp :flushed: 

I went back to 1.6.5-947 release and the issue was gone as expected.
Then uninstalled 1.6.5-947 and installed again esp8266-1.6.5-1160.

And guess what - the OTA network port shows up as required.

I think I might screw something up by adding two josn links (to the stable and staging release) in File > Preferences. At certain point I saw two directories “1.6.5-1160-gef26c5f” and “1.6.5-1160-gef26c5f” under “…\AppData\Roaming\Arduino15\packages\esp8266\hardware\esp8266”.

Thank you for quickly responding to this issue :+1: 

Krzysztof

closed 2015-11-08 by krzychb