esp8266/Arduino #983: esp8266 sketch data upload error with nodemcu

wemos, 2015-11-08:
when I use "esp8266 sketch data upload", choose "NodeMCU", there is error:
---
error: unknown board: nodemcu
known boards are: none ck wifio 
error: Failed to open COM16
error: espcomm_open failed
---
when I choose other boards, It work well.
Colud you add nodemcu to "esp8266 sketch data upload"

igrr, 2015-11-08:
Apparently you are using esptool 0.4.5? Nodemcu upload method was added in
0.4.6.

wemos, 2015-11-08:
I use esptool 0.4.6, upload is work well.
Only "esp8266 sketch data upload" is error.

wemos, 2015-11-08:
Do you have new version "esp8266 sketch data upload", I use this:

http://arduino.esp8266.com/ESP8266FS-1.6.5-1105-g98d2458.zip

igrr, 2015-11-08:
ESP8266FS tool simply runs esptool, so if you get an error saying that
nodemcu reset method is not supported, it means that esptool 0.4.5 is still
present somewhere. It could be in hardware/esp8266com/esp8266/tools or in
some stray boards manager package copy you might have on your computer.
Seems like a paths issue to me.

wemos, 2015-11-09:
OK, I remove all old esptool.
It's work well now.
Thank you @igrr

closed 2015-11-09 by wemos