esp8266/Arduino #986: WiFiClient driving me nuts

adrianrb, 2015-11-09:
Hi all...
I´m trying to make a simple weather client app, but can´t get response from many sites.
The idea is use a 16x2 LCD to show data/time/weather as follows:
```
DD/MM/YYYY HH:NN
Temp: XX Celsius
```
Tried to change "Host: " and "GET" strings extensively based in URLs below, but I always got "<nothing>... closing connection".
All URLs works OK in browser.

Someone can point me what´s wrong?
(Same using just the WiFiClient sample...)

Weather sources I´ve tried (I´m in Maringa, Brazil):

+ www.aviationweather.gov/adds/metars/?station_ids=SBMG&std_trans=standard&chk_metars=on
+ http://api.wunderground.com/api/9021af226189803f/conditions/q/BRAZIL/MARINGA.json
+ http://www.liveatc.net/search/?icao=sbmg
+ http://adrian-barbosa.net/w.php <<< this in my site; just make the request and "echo" the response 

PS: Yes, much code is for "guess" the right time in DST (daylight savings time), a.k.a. Summer Time... 8-(

sketch:
```
#include <ESP8266WiFi.h>
#include <WiFiUdp.h>
#include <Time.h>
#include <Wire.h>
#include <LiquidCrystal_I2C.h>

// ########### CHANGE THIS TO YOUR SETUP ###########
#define ssid "GVT-1314"
#define pass "0016432841"
//#define ssid "wifiradio"
//#define pass "12345678"
#define ntpServerName "time.nist.gov" //"a.ntp.br" //"time.nist.gov"
#define timezone -3.0F
#define SerialSpeed 115200
#define seconds2poll 60
#define HTTP_Port 80
// ########### CHANGE THIS TO YOUR SETUP ###########

unsigned int localPort = 2390; // local port to listen for UDP packets

LiquidCrystal_I2C lcd(0x27, 2, 1, 0, 4, 5, 6, 7, 3, POSITIVE);

IPAddress timeServerIP; // will receive server IP

const int NTP_PACKET_SIZE = 48; // NTP time stamp is in the first 48 bytes of the message

byte packetBuffer[NTP_PACKET_SIZE]; //buffer to hold incoming and outgoing packets

// A UDP instance to let us send and receive packets over UDP
WiFiUDP udp;

const char* host = "adrian-barbosa.net";
const char* query = "/w.html";

unsigned long epoch;

int cont = 0;

void ICACHE_FLASH_ATTR setup()
{
    lcd.begin(16, 2);
    lcd.backlight();
    
    Serial.begin(SerialSpeed);

    WiFi.mode(WIFI_AP_STA);
[... 429 lines cut ...]
Starting UDP
Local port: 2390
Signal strength: -71 dBm.
Opening HTTP connection with adrian-barbosa.net
Query: GET /w.html HTTP/1.1
HTTP/1.1 200 OK
Server: nginx
Date: Mon, 09 Nov 2015 00:39:10 GMT
Content-Type: text/html
Content-Length: 0
Connection: close

closing connection
packet received, length=48
Seconds since Jan 1 1900 = 3656018350
Unix time = 1447029550
The UTC time is 0:39:10

 ### ini: 1445126400
 ### fim: 1456012800

The Local time is 22:39:10
and today is (YYYY-MM-DD): 2015-11-8

```

ibaranov-cp, 2015-11-09:
This drove me crazy as well. I finally stumbled onto a correct solution:

```C
/*
 *  This sketch sends data via HTTP GET requests to data.sparkfun.com service.
 *
 *  You need to get streamId and privateKey at data.sparkfun.com and paste them
 *  below. Or just customize this script to talk to other HTTP servers.
 *
 */

#include <ESP8266WiFi.h>
#include <EEPROM.h>

#include <NeoPixelBus.h>

#define pixelCount 12
#define pixelPin 8  // must be pin 8

NeoPixelBus strip = NeoPixelBus(pixelCount, pixelPin);
NeoPixelAnimator animations(&strip); // NeoPixel animation management object

const char* ssid     = "****";
const char* password = "*****";

const char* host = "api.openweathermap.org"; //where we get the data from
const char* location   = "Kitchener,ca"; //your city and two letter contry code
const char* API_key = "********"; //your API key from openweathermap (signup and check login page)
const char* units = "metric"; //ie Celcius
const char* days = "1"; //consecutive days of data pulled

//Time for Leds to be active between, 6AM to 9PM
const byte startT = 6;
const byte stopT = 21;

//Dreaming mode
const bool dream = true;

//Weather Condition Codes:
//200s is Thunderstorms - Drizzle animation with flashes
//300s is Drizzle - Drizzle animation
//500s is rain - Fast drizzle Animation
//600s is snow - Slow Drizzle animation
uint16_t effectState = 0;  // general purpose varible used to store effect state

void setup() {

  EEPROM.begin(10);

  Serial.begin(115200);
  delay(10);

  // We start by connecting to a WiFi network

  Serial.println();
  Serial.println();
  Serial.print("Connecting to ");
  Serial.println(ssid);

  WiFi.begin(ssid, password);

  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }

  Serial.println("");
  Serial.println("WiFi connected");
  Serial.println("IP address: ");
  Serial.println(WiFi.localIP());

  strip.Begin();
  strip.Show();
}

void loop() {

  // visual chunk
  for (int i = 0; i < pixelCount; i++) {
    strip.SetPixelColor(i, RgbColor(random(0, 64), random(0, 64), random(0, 64)));
  }
  strip.Show();

  delay(5000);
  getWeather();
}

//Get current weather, compare to stored weather.
void getWeather()
{
  float yesterday = EEPROM.read(0) + EEPROM.read(1)/100.0;

  //EEPROM.write(0,(int)temp);
  //EEPROM.write(1,(temp - (int)temp)*100);
  
  //String request = host;
  String request = "/data/2.5/forecast/daily?q=";
  request += location;
  request += "&mode=xml&units=";
  request += units;
  request += "&cnt=";
  request += days;
  request += "&APPID=";
  request += API_key;

  char* buf = "";
  request.toCharArray(buf, 160);

  // Use WiFiClient class to create TCP connections
  WiFiClient client;
  const int httpPort = 80;
  if (!client.connect(host, httpPort)) {
    Serial.println("connection failed");
    return;
  }

  // This will send the request to the server +url after GET
  client.print(String("GET ") + buf + " HTTP/1.1\r\n" +
               "Host: " + host + "\r\n" +
               "Connection: close\r\n\r\n");

  //Wait up to 5 seconds for server to respond then read response
  int i = 0;
  while ((!client.available()) && (i < 500)) {
    delay(10);
    i++;
  }

  // Read all the lines of the reply from server and print them to Serial
  while (client.available()) {
    String line = client.readStringUntil('\r');  
    Serial.println(line);
  }

  Serial.println();
  Serial.println("closing connection");
}
```

adrianrb, 2015-11-10:
Thx, it´s working now... (almost one whole week stumbling across this issue!)

But, for igrr and others: couldn´t exists a (optional) "timeout" parameter in client.available()?
It will not require that hack...

For example:
```
1. if (client.available())... // default behavior, as today
2. if (client.available(5000))... // 5 seconds timeout for reading
```

igrr, 2015-11-11:
@adrianrb I think `client.setTimeout` will do the trick if you use one of `readString/readStringUntil` functions:
```c++
client.setTimeout(5000);
String line = client.readStringUntil('\r');
if (!line.length) { // nothing was received in 5 seconds, abort }
client.setTimeout(1000);
// do something with the line and read the remaining headers
```

It would be awesome if someone would convert their HTTP request sketch into HTTPClient library so that everyone doesn't have to write their own HTTP request parsers...

closed 2015-11-11 by igrr

omaro2009, 2019-02-21:
Iam having the same problem with my code Iam using NodeMcu to be able to Send a message to dweet Host
[code]

#include <ESP8266WiFi.h>
//const int ledpin1 = 5;
//const int ledpin2 = 4;
//const int ledpin3 = 15;
//const int input1 = 13;
//const int input2 = 14;
//const int input3 = 12;
//int counter = 0;
const char* host =  "dweet.io";
//// WiFi parameters
//const char* ssid = "Tedata2017";
//const char* password = "10009000*" ;
const char* ssid = "Omar";
const char* password = "11112222" ;

//const char* key="omar2009";
String thingName = "39f2be89-e54a-4148-a168-caa8dba77";
String hi = "I";
void setup() {
  // put your setup code here, to run once:

  Serial.begin(115200);
  delay(100);

  pinMode(5, OUTPUT);
  //Connect to WiFi Network
  Serial.println();
  Serial.println();
  Serial.print("Connecting to ");
  Serial.print(ssid);
  Serial.println("...");

  WiFi.begin(ssid, password);
  int retries = 0;
  while ((WiFi.status() != WL_CONNECTED) && (retries < 15)) {
    retries++;
    delay(500);
    Serial.print(".");
  }
  if (retries > 14) {
    Serial.println(F("WiFi conenction FAILED"));
  }
  if (WiFi.status() == WL_CONNECTED) {
    Serial.println(F("WiFi connected!"));
    Serial.println("IP address: ");
    Serial.println(WiFi.localIP());
  }
  Serial.println(F("Setup ready"));

}

String getDweetString() {
  WiFiClient client;
  String  dweetHttpGet = "GET/dweet/for/";
  dweetHttpGet = dweetHttpGet + String(thingName) + "?";
  dweetHttpGet = dweetHttpGet + String(hi) ;

  dweetHttpGet = dweetHttpGet + " HTTP/1.1\r\n" +
                 "Host: " +
                 host +
                 "\r\n" +
                 "Connection: close\r\n\r\n\r\n";
 
  //String url = "/dweet/for/";
  //  url += thingName;
  //  url += "?";
  //  url += "hi=";
  //  url += hi;
  //
  //
  //  Serial.print("Requesting URL: ");
  //  Serial.println(url);
  //
  //  // This will send the request to the server
  //  client.print(String("GET ") + url + " HTTP/1.1\r\n" +
  //               "Host: " + host + "\r\n" +
  //               "Connection: close\r\n\r\n");

 // delay(10);
  return dweetHttpGet;//this is our freshly made http string request

}

void sendDweet() {
  WiFiClient client;
  const int httpPort = 80;

  //connect to dweet.io
  if (!client.connect(host, httpPort)) {
    Serial.println("connection failed");
    return;
  }
  else {
    Serial.println("connection 1");

  }
  getDweetString();
  client.print(getDweetString());
  delay(10); //wait...
  client.setTimeout(5000);
String line = client.readStringUntil('\r');

if (line.length()==0) { 
 // nothing was received in 5 seconds, abort 
}
client.setTimeout(1000);
Serial.println("NAN");
 
            
}
 // Serial.println();
 // Serial.println("closing connection");

void loop() {
  // put your main code here, to run repeatedly:
  Serial.print("Sending dweet to ");
  Serial.print(host);
  Serial.print("/follow/");
  Serial.print(thingName);
  Serial.println();

  sendDweet(); //send data to dweet.io
  delay(2000);

}
[/code]

The output is:
Sending dweet to dweet.io/follow/39f2be89-e54a-4148-a168-caa8dba77
connection 1
NAN
Thanks in advanced

d-a-v, 2019-02-21:
```
String getDweetString() {
  WiFiClient client;
  ...
```
If used by commented code after it, this client is not initialized.
You don't print anything else (line?) than NAN.