esp8266/Arduino #991: "Error: esptool/mkspiffs not found!" for ESP8266FS.jar on Windows

h4rm0n1c, 2015-11-09:
The only way I could get ESP8266FS.jar working on windows was to put the esptool and mkspiffs executables in the root of the esp8266 tools folder.
The path in question being {runtime.platform.path}/tools/ which resolves to C:\Program Files (x86)\Arduino\hardware\esp8266com\esp8266\tools

After doing this, the Arduino IDE upload function worked without errors.

I'm not sure if this is an issue with ESP8266FS.jar or the IDE.

igrr, 2015-11-09:
1.6.5 or 1.6.6?
Boards manager package or git tree in hardware/esp8266com/esp8266?

h4rm0n1c, 2015-11-09:
1.6.5 with the latest git as of about half an hour ago. (810ab68)

igrr, 2015-11-09:
It's an issue with ESP8266FS tool, I haven't adapted it to the new
directory layout. Will fix.

igrr, 2015-11-12:
@h4rm0n1c please try the new release: https://github.com/esp8266/arduino-esp8266fs-plugin/releases/tag/0.1.3

h4rm0n1c, 2015-11-13:
@igrr All good, cleared out tools folder and tried with new version, tool runs without errors "out of the box".

Thank you! :+1:

closed 2015-11-30 by igrr