esp8266/Arduino #992: WDT reset when client disconnects.

drmpf, 2015-11-09:
When client disconnects from Server, get WDT
debug info is
```
:sent 1460
:ww
:wr
:sent 1460
:ww
:wr
:er -9 1460 1
:abort
:ww
ConnectionClosed
:ur 1
WS:dis

 ets Jan  8 2013,rst cause:4, boot mode:(3,7)

wdt reset
load 0x4010f000, len 1264, room 16 
tail 0
chksum 0x42
csum 0x42
~ld

......
Connected!
Start Server
Server Started
10.1.1.200:23
Listening for connections...
```
Tested using TeraTerm connect then disconnect
TeraTerm sends RST, ESP responds with RST, then watch dog times out.

Test sketch is
```
#include <ESP8266WiFi.h>
#include <WiFiClient.h>
#include <ESP8266WebServer.h>

char ssid[] = "xxxx";
char password[] = "xxxxx";

const int portNo = 23;
WiFiServer server(portNo);
WiFiClient client;

void setup ( void ) {
  Serial.begin (115200);  // NOTE: do not set above 9600 to prevent loosing data on send.
  delay(10);
  Serial.println();
  IPAddress ip(10, 1, 1, 200);
  IPAddress gateway(10, 1, 1, 1); // set gatway to ... 1
  IPAddress subnet(255, 255, 255, 0);
  WiFi.config(ip, gateway, subnet);

  WiFi.begin(ssid, password);
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
  Serial.println();
  Serial.println(F("Connected!"));

  server.begin();
  server.setNoDelay(true); // does not do much if anything
  Serial.println(F("Server Started"));
  Serial.print(WiFi.localIP());
  Serial.print(':');
  Serial.println(portNo);
  Serial.println(F("Listening for connections..."));
  client = server.available();
  Serial.print("+++"); // end of setup start listening now
}

static const size_t SEND_BUFFER_SIZE = 1460; // Max data size for standard TCP/IP packet
static uint8_t sendBuffer[SEND_BUFFER_SIZE]; //
size_t sendBufferIdx = 0;
bool alreadyConnected = false;
// the loop routine runs over and over again forever:
void loop() {
  if (!client) { // see if a client is available
    client = server.available(); // evaluates to false if no connection
  } else {
    // have client
    if (!client.connected()) {
      if (alreadyConnected) {
        // client closed so clean up
        closeConnection();
      }
    } else {
      // have connected client
      if (!alreadyConnected) {
        Serial.println("ConnectionOpened");
        client.setNoDelay(true); // does not do much if anything
        alreadyConnected = true;
        sendBufferIdx = 0;
      }
    }
  }
  sendBuffer[sendBufferIdx] = '\n';
  sendBufferIdx ++;
  if (sendBufferIdx == SEND_BUFFER_SIZE) {
    if (client && client.connected()) {
      // buffer full
      client.write((const uint8_t *)sendBuffer, sendBufferIdx);
    } else {
      // just throw this data away
    }
    sendBufferIdx = 0;
  }
  delay(0); // yield
  if (client && client.connected()) {
    while ((client.available() > 0) &&  (Serial.availableForWrite() > 0) && client.connected()) {
      Serial.write(client.read());
      delay(0); // yield
    }
  }
}

void closeConnection() {
  Serial.println("ConnectionClosed");
  alreadyConnected = false;
  WiFiClient::stopAll(); // stop all clients and release memory.
  client = server.available(); // evaluates to false if no connection
}

```

me-no-dev, 2015-11-09:
Looking at it, WiFiClient::stopAll(); will kill your server as well.

drmpf, 2015-11-09:
Looking at the code stopAll() only works on WiFiClients.
But in any case I tried removing that line and still get WDT reboot

me-no-dev, 2015-11-10:
try calling client.stop() then. It might help. Or also maybe clearing your local "client" object.

drmpf, 2015-11-10:
Tried example WiFiTelnetToSerial sketch
sending 1460 size buffer at a time by modifying the sketch as follows
```
//  if(Serial.available()){
    size_t len = Serial.available();
    uint8_t sbuf[len];
    Serial.readBytes(sbuf, len);
    //push UART data to all connected telnet clients
    for(i = 0; i < MAX_SRV_CLIENTS; i++){
      if (serverClients[i] && serverClients[i].connected()){
   //     serverClients[i].write(sbuf, len);
        serverClients[i].write((char *)buffer,1460);
        delay(1);
      }
    }
//  }
```
buffer is defined as
char buffer[1460] = "\n";

same result, get WDT when Windows TeraTerm disconnects.
```
Ready! Use 'telnet 10.1.1.10 23' to connect
New client: 0ÿûÿýÿûÿýÿû
 ets Jan  8 2013,rst cause:4, boot mode:(1,7)

wdt reset
```

dhanyweb, 2015-11-13:
i have the same problem, any solutions for this?

devyte, 2017-12-17:
A lot has changed since the last relevant comment, including the possibility to use lwip2 in latest git.
Closing due to age and lack of feedback.

closed 2017-12-17 by devyte