esp8266/Arduino #993: Invalid library found in (Arduino 1.6.6)

domonetic, 2015-11-09:
in 1.6.6 version on select genericesp8266 board 
Invalid library found in C:\Users\maxid\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\1.6.5-1160-gef26c5f\libraries\esp8266: C:\Users\maxid\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\1.6.5-1160-gef26c5f\libraries\esp8266

grahamehorner, 2015-11-09:
also getting this with a clean install of arduino.cc IDE v1.6.6 and esp8266 from git hub following instructions on page

domonetic, 2015-11-09:
I have a clean install and use the staging repository

WheezeNL, 2015-11-09:
Yesterday I had the same problem.
IDE 1.6.6 on Win10 gives a load of warnings and errors.
IDE 1.6.5 works perfectly.

Links2004, 2015-11-09:
the problem is noted done here: #965
currently we not support 1.6.6 it may work in some cases
but OTA is for example broken (work is going in there).

domonetic, 2015-11-09:
ok ok back to the past!!!
Thanks

igrr, 2015-11-15:
Fixed with 74aec43.

closed 2015-11-30 by igrr

NimdaEng, 2017-06-15:
delete all folders in C:\Users\<USER>\Documents\Arduino