esp8266/Arduino #994: help with multiple DS18B20 bus position

grahamehorner, 2015-11-09:
I'm attempting identify the position on a daisy chained DS18B20 sensor in the OneWire bus; is this possible? if so how can I do this ?

chaeplin, 2015-11-09:
Each DS18B20 has uniq serial number.
check example of lib https://github.com/milesburton/Arduino-Temperature-Control-Library or https://arduino-info.wikispaces.com/MultipleTemperatureSensorsToLCD

Humancell, 2015-11-09:
Discovery of the physical order or "position" is a non-trivial task.  Some of the newer components allow for "chaining" that you can read about here:  https://www.maximintegrated.com/en/app-notes/index.mvp/id/4037

Without chaining, the normal "search" algorithm follows an address based search documented here:  https://www.maximintegrated.com/en/app-notes/index.mvp/id/187

jeffryr, 2015-11-09:
Before soldering them in place, attach them one at a time and run something like the [example](https://github.com/esp8266/Arduino/blob/master/libraries/OneWire/examples/DS18x20_Temperature/DS18x20_Temperature.pde) sketch so you can associate the unique serial number with each sensor.

grahamehorner, 2015-11-09:
@chaeplin yes I'm aware the DS18B20s have a unique identifier; but I need to know where in the daisy chain the sensor is located.
@jeffryr I can't do what you suggest as the daisy chain is built by the end user, pluging together 3.5mm jack plugs together and thus the DS18B20s could change and/or be replaced; the OneWire bus is not fixed in length or which OneWire devices get attached.  

@Humancell thanks for the links; I'll read up and see if I can figure out how to determine the location of the sensor in the linear chain.

Links2004, 2015-11-09:
The OneWire bus is not designed to do this, the response order is linked to the unique identifier not to the order on the bus.

the easiest way is a kind of setup process for the end-user where he must say I now plug sensor 1,
if he plug it in you see by scanning there is one new device. which this information you now can match the unique identifier to sensor 1.
if this is repeated for every sensor on the bus you end up with a little database to match unique identifier to sensor id / position.

grahamehorner, 2015-11-09:
@links2004 thanks but this is not a limit of the onewire bus but the sensors connected, I have abandoned the idea of using the ds18b20 for the ds28ea00

duncan-a, 2015-11-10:
As I understand the ds28ea00, what you're wanting to do is possible (with
an extra pair of lines for each sensor to allow the 'daisy chaining') but
it will fail if any intermediate sensor is removed, because the 'chain'
will be broken i.e. unplug 2 and 3 & 4 no longer have the chain
information...

grahamehorner, 2015-11-10:
@duncan-a this differs to my understanding;my understanding is that the OneWire bus is able to detect the position on the sensor by signal which sensors respond to in sequence; the bus still uses the same number of wires however the sensor have additional pins to allow this to operate.

https://www.maximintegrated.com/en/products/analog/sensors-and-sensor-interface/DS28EA00.html

igrr, 2015-11-10:
According to the page you have linked, DS28EA00 *is* using extra lines for sequence detection.

`Unlike other 1-Wire thermometers, the DS28EA00 has two additional pins to implement a sequence detect function.`

grahamehorner, 2015-11-10:
@igrr yes the sensor has two addition pins; but the bus still only has 3 wires VCC, GN, DQ ? or am I reading this wrong

igrr, 2015-11-10:
Check the schematic image at the page you have linked. Additional line is shown between PIOA pin of one sensor and PIOB of the next one.

grahamehorner, 2015-11-10:
@igrr I have; yes I can see PIOB and PIOA; PIOA is connected to the next sensor in the chain, and I understand that; but the bus still only has three wires? no VDD is connection to the sensors (shown as dotted line) PIOB-> PIOA connection, GND and IO?

igrr, 2015-11-11:
If you don't need VDD connection, then yes, this makes it just three lines.

closed 2015-11-11 by igrr

grahamehorner, 2015-11-11:
@igrr @Humancell thanks for everyone that helped; thought in I was going mad; and needed someone to verify ;)