esp8266/Arduino #1000: Add multiple request handlers for onnotfound and onfileupload

sticilface, 2015-11-11:
I've been writing a lib that uses both of these functions.  However, it is possible / likely that different libraries may which to have access to these functions and not just one. 

For example I'm using the FSbrowser, but also HTTP upload libs.  At the moment I have this in my setup 
```
  HTTP.onFileUpload([&](){ 
    handleFileUpdate();
    handleUpgrade(); }
    );
```

but this requires the user to correctly place the right functions in their setup function to use the libs. 

Would it be possible to have an AddonFileUpload function to the ESP8266WebServer that would append callbacks to a list in the onfileupload or onnotfound functions (in a similar manor to the handle function),  these functions could return a bool which would then cause the web server to stop or continue through the functions. 

Bit of a confusing explanation there, but hopefully it makes sense.

igrr, 2015-11-11:
Ideally, I would like to move `handleFileUpload` and `handleNotFound` to RequestHandler interface.
Every library would inherit from RequestHandler, and WebServer would walk all registered request handlers in sequence, looking for the first one which is able to handle particular request.
Currently RequestHandler is used only for "normal" requests but adding another two methods is certainly possible.

sticilface, 2015-11-11:
That would be great.  Its probably a bit/quite beyond me to attempt this.  at least in a way that would not just require loads more work from you guys to debug it.

igrr, 2015-11-11:
Ok, not sure this falls into next Monday release, but I'll put this on my list.

igrr, 2015-11-30:
RequestHandler class now includes upload handlers.

closed 2015-11-30 by igrr