esp8266/Arduino #998: CRC error on installing "Stable" Esp8266  from boards manager

agent2003, 2015-11-11:
Hi, similar to the topic below:
https://github.com/esp8266/Arduino/issues/749

I get an error saying CRC error, I think specifically related to the file:
http://arduino.esp8266.com/win32-xtensa-lx106-elf-gb404fb9.tar.gz 

Which when downloaded using a browser gives an 'unexpected end of archive' error in winrar.

I'm using Arduino IDE 1.6.6, which I have just read may cause problems during use, but I'm sure this would not affect download and verification of the file from the download manager

I've tried the stable and staging .json links and get the same problem:

Any help would be appreciated

Thanks

igrr, 2015-11-11:
I have downloaded the package manually and verified it matches SHA sum given in package.json.
Could you please check your network connectivity? If you are unable to download the file manually via browser then the issue is likely unrelated to Arduino IDE and the package file.

agent2003, 2015-11-15:
I fixed by manually downloading the package and running get.py in the tools folder to download. Thanks

closed 2015-11-15 by igrr