esp8266/Arduino #1010: Wifi.APmode() crash dump on 1MB ESP07 : suspect 1M+64k SPIFFS config

hallard, 2015-11-12:
Hi there,

I'm knocking my head since 2 days, I've got ESP07 1MB that is working fine (Wifi Web server, bla bla all I need is fine) until I decide to set it as an AP (1st config UI), at this point it came direct to crash.

So I tested several things, including creating new test sketch with lot of information and setup to Access point (full demo at the end of this issue). 
it show that `WiFi.softAP(ssid,password)` goes direct to a core dump whatever param I pass `WiFi.softAP(ssid)` is the same so `WiFi.softAP("Hello","World")`

FYI, I do not use SPIFFS, so I don't need it.

What is very Strange is that all is working fine as soon as I do not setup AP. When setup to 1MB 64 SPIFFS (so no AP) OTA is working fine, so I think it's really an 1MB Flash chip on board ;-)

here the results I tried :    

**Setting 512K board (NO OTA possible)**
- set board to **512K No SPIFFS** => My sketch works, test sketch with set AP works
- set board to **512K 64K SPIFFS** => My sketch works, test sketch with set AP works    

**Setting 1M board**
- set board to **1MB No SPIFFS** => This one I would like to try
- set board to **1MB 64K SPIFFS** => My sketch works, test sketch with **set AP crash**

I'm suspecting conflict with SDK config/flash config or whatever, I really would like to try with config **1MB No SPIFFS** as a test just to see but we do not have this option.

You can see crash results below :+1: 
```
========== Flash config start
Flash real id   : 001440C8
Flash real size : 1048576

Flash ide  size : 1048576
Flash ide speed : 40000000
Flash ide mode  : DIO
Flash Chip conf : Good!
========== Flash config End
========== SDK Saved parameters Start
Mode: STA
PHY mode: N
Channel: 1
AP id: 0
Status: 0
Auto connect: 0
SSID (7): Livebox
Passphrase (18): ******************
BSSID set: 0
========== SDK Saved parameters End
Configuring access point...

Exception (29):
epc1=0x4000e1c3 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000018 depc=0x00000000

ctx: cont 
sp: 3fff0830 end: 3fff0be0 offset: 01a0

>>>stack>>>
3fff09d0:  40223fae 4021744c 00000018 3fff1cc0  
3fff09e0:  00000001 3fff2188 3fff2188 40224194  
3fff09f0:  3fff2188 3fff1cc0 00000000 00000001  
3fff0a00:  3ffee704 3ffee518 223a22fc 00000000  
3fff0a10:  4022f498 00000064 fc34fe1a fffeffff  
3fff0a20:  00000018 0000ffff 00000000 00000001  
3fff0a30:  3fff0a48 40220748 3ffee494 3fff1cc0  
3fff0a40:  fc34fe1a 3fff01b4 00000000 00000000  
3fff0a50:  00000000 00000000 00000000 00000000  
3fff0a60:  00000000 00000000 00000000 00000000  
3fff0a70:  00000000 00000000 4022143f 3fff1cc0  
3fff0a80:  3ffefa1c 3fff1cc0 3fff1cc0 40221487  
3fff0a90:  00000000 3ffee518 3ffe93bc 00000000  
3fff0aa0:  40210271 00000003 00000003 00000000  
3fff0ab0:  40210389 00000003 00000003 00000030  
3fff0ac0:  00000001 402103e2 00000003 00000000  
3fff0ad0:  4020232d 00000001 3ffe8c9d 4020554c  
3fff0ae0:  3fff0c00 0000000a 3ffef9fc 40202377  
3fff0af0:  3ffefa54 00000001 3fff0c0c 4020518d  
3fff0b00:  3ffe8c9c 0000000a 3fff0c0c 40204d95  
3fff0b10:  3fff0c0c 00000001 3ffe8c9d 4020554c  
3fff0b20:  3fff0c0c 0000001a 3ffe8636 3ffef9fc  
3fff0b30:  00000000 00000000 3ffefbc0 00000000  
3fff0b40:  3ffefb08 401011f3 3ffefbc0 3ffefb08  
3fff0b50:  3ffefa1c 3ffef9fc 40204858 3ffefbc0  
3fff0b60:  402010ae 000001f4 000001f4 3ffef9fc  
3fff0b70:  3ffefa1c 3ffefa9c 3fff0c0c 4020222f  
3fff0b80:  00000000 00000000 00000000 00000000  
3fff0b90:  00000000 00000000 00000000 00000000  
3fff0ba0:  00000002 00000000 00000000 00000000  
3fff0bb0:  00000000 00000000 00000000 3ffefbac  
3fff0bc0:  3fffdc20 00000000 3ffefba4 40204896  
3fff0bd0:  00000000 00000000 3ffefbc0 40100114  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(1,6)
```

Here the full test demo sketch

```c++
#include <ESP8266WiFi.h>
#include <WiFiClient.h> 
#include <ESP8266WebServer.h>

char ssid[16] ;
char password[32];

#define SERIAL_DEBUG	Serial1

ESP8266WebServer server(80);

void handleRoot() {
	server.send(200, "text/html", "<h1>Hello World!</h1>");
}

void setup() {
	delay(1000);
	SERIAL_DEBUG.begin(115200);

  SERIAL_DEBUG.println("========== Flash config start"); 
  uint32_t realSize = ESP.getFlashChipRealSize();
  uint32_t ideSize = ESP.getFlashChipSize();
  FlashMode_t ideMode = ESP.getFlashChipMode();

  SERIAL_DEBUG.printf("Flash real id   : %08X\n", ESP.getFlashChipId());
  SERIAL_DEBUG.printf("Flash real size : %u\n\n", realSize);
  SERIAL_DEBUG.printf("Flash ide  size : %u\n", ideSize);
  SERIAL_DEBUG.printf("Flash ide speed : %u\n", ESP.getFlashChipSpeed());
  SERIAL_DEBUG.printf("Flash ide mode  : %s\n", (ideMode == FM_QIO ? "QIO" : ideMode == FM_QOUT ? "QOUT" : ideMode == FM_DIO ? "DIO" : ideMode == FM_DOUT ? "DOUT" : "UNKNOWN"));
  SERIAL_DEBUG.printf("Flash Chip conf : %s!\n", ideSize==realSize ? "Good" : "**** Wrong ****");
  SERIAL_DEBUG.println("========== Flash config End"); 

  SERIAL_DEBUG.println("========== SDK Saved parameters Start"); 
  WiFi.printDiag(SERIAL_DEBUG);
  SERIAL_DEBUG.println("========== SDK Saved parameters End"); 
  strcpy(ssid,"ESP_Test");
  strcpy(password,"allowme");

	SERIAL_DEBUG.println("Configuring access point...");
  SERIAL_DEBUG.flush();
  delay(500);
	WiFi.softAP(ssid,password);
  SERIAL_DEBUG.print("WiFi.softAP() called");
  SERIAL_DEBUG.flush();

	IPAddress myIP = WiFi.softAPIP();
	SERIAL_DEBUG.print("AP IP address: ");
	SERIAL_DEBUG.println(myIP);
	server.on("/", handleRoot);
	server.begin();
	SERIAL_DEBUG.println("HTTP server started");
}

void loop() {
	server.handleClient();
}
```

If anybody has an idea, I'll take, this is driving me mad !!!

hallard, 2015-11-12:
Got some more information, I played changing flash size from ide then flash same board, and of course changing this break SDK config parameters. And it seems SDK does not do any check on what it find in config (at least a checksum on the config structure would have been good, I'm saving my own config in flash and I check it's correct, if not default settings).

Saying that, in the example below, look, config is corrupted (after played changing flash size), and I've got a crash a boot

Is there anyway to clear this config at boot ? I don't need it and seems I don't want it !

```
========== SDK Saved parameters Start
Mode: AP
PHY mode: N
Channel: 1
AP id: 0
Status: 255
Auto connect: 1
SSID (2): 
Passphrase (31): �†r €À×]
ÐYƒÝPê€åÀç»ö�îÿÀ 
BSSID set: 0
========== SDK Saved parameters End
Connecting to: Livebox with key '4rpJ**********YsH4'...
Exception (29):
epc1=0x4000e1c3 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000018 depc=0x00000000

ctx: cont 
sp: 3fff1070 end: 3fff14a0 offset: 01a0

>>>stack>>>
3fff1210:  4022880e 4021bcb0 00000018 3fff2aa8  
3fff1220:  00000001 3fff3048 3fff3048 402289f4  
3fff1230:  3fff3048 3fff2aa8 3ffef308 00000000  
3fff1240:  3ffee8d4 3ffee6e8 72860ca0 3ffef308  
3fff1250:  40236ac8 00000064 fc34fe1a fffeffff  
3fff1260:  00000018 0000ffff 00000000 00000001  
3fff1270:  3fff1288 40224fa8 3ffee664 3fff2aa8  
3fff1280:  fc34fe1a 3ffffbb6 00000000 00000000  
3fff1290:  00000000 00000000 00000000 00000000  
3fff12a0:  00000000 00000000 00000000 00000000  
3fff12b0:  00000000 00000000 40225c9f 3fff2aa8  
3fff12c0:  3ffef328 3fff2aa8 3fff2aa8 40225ce7  
3fff12d0:  00000000 3ffee6e8 3ffe958c 00000000  
3fff12e0:  40214ae9 00000003 00000003 3ffef308  
3fff12f0:  40214c01 00000003 00000003 00000030  
3fff1300:  00000001 40214c5a 00000003 3ffef308  
3fff1310:  40203de5 00000030 3fff133e 40209fe9  
3fff1320:  402080a5 0000000a 3fff021c 4020427d  
3fff1330:  3fff14d0 00000001 3ffe8609 40209de8  
3fff1340:  3fff1400 0000000a 3ffe8806 3ffe87e8  
3fff1350:  3fff0104 00000001 3fff14d0 40209515  
3fff1360:  3ffe8608 0000000a 3fff14d0 40208edd  
3fff1370:  3fff14d0 00000001 3ffe8609 40209de8  
3fff1380:  3fff14d0 00000011 3ffef339 40209de8  
3fff1390:  00000000 00000000 3fff0480 00000000  
3fff13a0:  0000007f 00000000 40208898 3fff0480  
3fff13b0:  3ffef328 00000022 3fff14d0 3ffef788  
3fff13c0:  3ffef308 3fff021c 3fff14d0 40204348  
3fff13d0:  3ffef308 3fff021c 3fff14d0 40202ab3  
3fff13e0:  3ffe8608 3ffef308 00000400 00000003  
3fff13f0:  3fff14d0 00000002 3fff14d0 40209ebc  
3fff1400:  3ffef308 00000001 3fff14d0 40209edd  
3fff1410:  00000006 3ffef328 3fff14d0 40202233  
3fff1420:  3fff1440 00000040 4023357c 7fffffff  
3fff1430:  3ffef764 3ffef308 3fff14d0 40202e23  
3fff1440:  65736552 6f742074 66656420 746c7561  
3fff1450:  00000000 00000000 00000000 00000000  
3fff1460:  00000000 00000000 00000000 00000000  
3fff1470:  00000000 00000000 00000000 3fff0470  
3fff1480:  3fffdc20 00000000 3fff0468 402088d6  
3fff1490:  00000000 00000000 3fff0480 40100114  
<<<stack<<<
 ets Jan  8 2013,rst cause:2, boot mode:(1,6)

```

hallard, 2015-11-12:
Another problem with corrupted SDK config data (I just had the case) is that it crash before you get hand before setup(), which can totally "brick" your device. 1st line in my setup I tried WiFi.disconnect to reset config, but it crashed before.

Only way, I had to do, reflash with setting flash size back to 512Kb (it worked, lucky or not) and then flash back with 1Mb

May be a 512KB sketch resetting flash of stored SDK config for 1MB config would help in this type of problem. Not sure it's possible. 
Anyway moving to different flash size on same device is not a good idea....

Links2004, 2015-11-12:
there is a function for deleting the config.
https://github.com/esp8266/Arduino/blob/master/cores/esp8266/Esp.cpp#L304

but may the SDK work already with it before the ```setup``` call.
the first possible call will be the usage of 
```
void initVariant() {
}
```

your dumps look like the usage of an null ptr.
the bigger problem that the error happen in the hard coded rom (0x4000e1c3).

the nearest function that we now is ```memset``` (0x4000e190) 
https://github.com/esp8266/Arduino/blob/master/tools/sdk/ld/eagle.rom.addr.v6.ld#L197

but from where the null pointer is coming is hard to say.

igrr, 2015-11-13:
You're totally right, moving to different flash size on the same device is not a good idea.
SDK uses data from config sectors early at boot and there are only some limited integrity checks.
So the only reliable way to switch to a different flash size is by erasing flash first using esptool.py.

hallard, 2015-11-13:
@Links2004 , @igrr 
thanks for these information.
You know, sometimes we don't know modules flash size, so the 1st time is tricky and I use 512Kb NOSPIFFSf with test sketch ;-)
Nice trip esptool to clear flash sounds good, any example of command on how can I do this to clear also config ? And may be know flash size with esp tools ?
thanks for your help

Charles

igrr, 2015-11-13:
`esptool.py -p /dev/ttyUSB0 erase_flash` will erase the whole chip, including config.

By the way, I made a similar feature for esptool-ck, but apparently it segfaults on windows: https://github.com/igrr/esptool-ck/pull/18
Didn't have a chance to debug this yet on a windows machine. 
If you're on windows, perhaps you could take a look?

hallard, 2015-11-13:
@igrr 
I'm under windows so I don't have `esptool.py` so tried the `esptool.exe` from esp8266 repo and and the one you have on your repos esptool-ck (renamed to esptool-ck.exe)

It seems both are doing nothing with erase_flash, even with other commands, like not even trying to communicate with the module

what I'm doing wrong ?

[image]

igrr, 2015-11-13:
@hallard erase_flash is a `esptool.py` option. esptool-ck doesn't have it in 0.4.6 (which you are using) yet.

I think your best bet is to download esptool.py and use it to erase flash (you have python, don't you?)

Regarding erase_flash option in esptool-ck, i have only added it in this pull request: https://github.com/igrr/esptool-ck/pull/18. It has been reported to not work on windows for some reason. I don't have a windows machine now so not able to debug.

hallard, 2015-11-13:
@igrr 
Ah ok, I tough `esptool.py` was only in arduino-esp8266 linux/mac distrib and on windows we only have `esptool.exe`

So I got it launched and running, but It can't connect on my ESP board, I launch `esptool.py` then press flash button and push/release reset while keeping flash pressed. Is that procedure is correct ?
I can program my board with this sequence from arduino IDE.

I tried with cygwin and with cmd line. Notice cygwin trick, port is `/dev/ttySd` where d digit is com_port - 1 

[image]

hallard, 2015-11-13:
@igrr 
I switched to a well made for this, Raspberry PI with ssh, connected my USB.Serial on PI
This seems to have worked, esptool saying nothing on this parameter `erase_flash` mean "success". Good to know!

My ESP07 came back to life with 1M 64K SPIFFS setting no more crash switching from AP to STA. This was so as I thought, related to corrupted SDK saved parameters.

Anyway this procedure can be setup to "unbrick" devices ;-)

PS : If anyone succeed to works with `esptool.py` from Windows I'd like to know the procedure :-)

Links2004, 2015-11-13:
a compiled python there:
https://github.com/Links2004/esptool/blob/master/bin/esptool.exe

I had simply installed py2.7 and it is working.
for simpler usage i create a exe version of it, I dont want to install python on every pc i use ;)

hallard, 2015-11-13:
@Links2004 

I've just tried, with no luck, first shot, 1st time it told me I was missing c:\python27\python27.dll but found it in system folders, so copied it from c:\windows\syswow64\ didn't worked, so tried 32bits version, copied from c:\windows\system32\ then I can execute the exe but got this :

[image]

hallard, 2015-11-13:
I'm anticipating a questio, of course I get installed pyserial module (esptool.py find it);-)

Links2004, 2015-11-13:
you need to download the full bin dir :)
https://github.com/Links2004/esptool/tree/master/bin

hallard, 2015-11-13:
Same thing ;-(

[image]

hallard, 2015-11-13:
Oh sorry access denied, got it !!!

Links2004, 2015-11-13:
this time the serial port is not found, the rest seams to work.

hallard, 2015-11-13:
rocks !!!, should I have closed Serial Monitor on another screen before it would works betters, same thing if I had read correctly error message, my fault ;-)

[image]

Links2004, 2015-11-13:
this epstool version has a little differences to the standard one.
It has an auto retry, and if one baud rate is not working its 
reduce the rate automatically.
this is very useful if you do some fast breadboard test and the wires are all over.

closed 2015-11-15 by igrr