esp8266/Arduino #1011: Disconnected postmortem capability

cpacey, 2015-11-12:
The [built-in postmortem](https://github.com/esp8266/Arduino/blob/master/cores/esp8266/core_esp8266_postmortem.c#L54) logging has been very useful.  However, the data is always output to UART0, which functionally means that I need to be physically connected to the ESP to get that data.  More and more, I'm wanting to get this information from a device in the field which has nothing attached to UART0.

I'm not entirely sure the best way to do this yet - among the options I've considered:
1. Copy the crash information (`rst_info`, `sp`, the stack contents, etc) to flash before restarting, and then process as desired at next start up.  My biggest worry about this is that we might do too many flash writes to a single sector, especially if the device gets into a tight crash loop.
2. Send the info somewhere else, e.g. a quick UDP transmit.  I've not tried this yet, so I don't know if it's feasible.

Regardless of the specific implementation, I could do it myself if I could register my own callback for when a crash occurs.

Links2004, 2015-11-12:
the reset reason is allready stored you can use
```cpp
Serial.println(ESP.getResetInfo());
```
https://github.com/esp8266/Arduino/blob/master/cores/esp8266/Esp.cpp#L291

to get basic info, no full stack dump but epc1, epc2, epc3, excvaddr and depc
which deliver a good start for a debugging.

cpacey, 2015-11-12:
We're already using ESP.getResetInfo(), but aside from the missing stack trace (which has been very useful for me), ESP.getResetInfo() doesn't give us anything for Exceptions or Soft WDT resets, at least for the Huzzah boards we're using.  Essentially, I can only use ESP.getResetInfo() right now to find out if there was a hard wdt reset.

closed 2015-11-30 by igrr