esp8266/Arduino #1013: Documentation on "rst cause" and "boot mode"

funvill, 2015-11-13:
I am getting errors while building firmware for the ESP8266 using a nodemcu 1.0. Are the errors define or documentation anywhere so I can try to resolve my own issues. 

rst cause:2, boot mode:(3,6)
rst cause:4, boot mode:(1,6)

etc..

h4rm0n1c, 2015-11-13:
Here's documentation on that:

http://www.esp8266.com/viewtopic.php?p=2096#p2112

https://github.com/esp8266/esp8266-wiki/wiki/Boot-Process#esp-boot-modes

Your first one appears to be caused by the reset pin (could indicate power supply issues if nothing has actually happened to the pin itself.), the boot mode for this one is boot from spi flash memory.

The second one was caused by the watchdog timer, with the boot mode set to download new code from the serial port.

You get a freebie this time, but for next time:

I'd strongly advise signing up for the esp8266 forums and getting familiar with the search function they provide, most of the answers you seek are already available there, since this is an issues tracker with scope limited to the Arduino IDE project rather than a general help forum.

closed 2015-11-13 by funvill

andig, 2016-01-09:
Do we have an api to retrieve the reset cause/ boot mode?

Links2004, 2016-01-09:
not direct the closest we have  reset cause for is the ```resetInfo.reason ```
https://github.com/esp8266/Arduino/blob/master/cores/esp8266/Esp.cpp#L332

for boot mode the access is unknown

andig, 2016-01-09:
Excellent, just what I was looking for.

FredFirestone, 2016-06-26:
I had the exact same errors with a simple C program I modified. Turned out my user_init function ran to long: Adding or removing os_delay_us produced or removed the issue.

MrD2, 2017-08-11:
This is always the first link on google.... so i'll just put here the results of the above refered 1st file:

    reset causes:
        0: 
        1: normal boot
        2: reset pin
        3: software reset
        4: watchdog reset

    boot device:
        0:
        1: ram
        3: flash

marcmerlin, 2018-05-02:
@h4rm0n1c  your point about "use the forums" is valid. I tried doing that on my last problem. Every message I posted took 2 to 3 days to be manually approved.
That's a total joke, what forum requires this and requires so much delay when you need help?
Sadly as a result, I don't use the forums there anymore, they're pointless as is.
Just an FYI to others so that they don't waste their time like I did :(