esp8266/Arduino #1014: Upload menu - some speeds not shown

rogerclarkmelbourne, 2015-11-13:
On IDE 1.6.5 (Windows), some upload speeds for the Generic ESP8266 which are in boards.txt do not appear in the menu.

This appears to be cause the name of the upload speed is not defined for all speeds
e.g. The following line appears to be missing
generic.menu.UploadSpeed.230400=230400

to make the following (existing) settings become visable

generic.menu.UploadSpeed.230400.linux=230400
generic.menu.UploadSpeed.230400.macosx=230400
generic.menu.UploadSpeed.230400.upload.speed=230400

igrr, 2015-11-13:
I consciously added these speeds only for Linux and OS X because most Windows serial port drivers do not provide them. Windows drivers usually offer 256000 and 512000, which OS X doesn't have, for instance. At least that was my experience with FTDI driver on Windows 7.

So 256000 serves as a replacement for 230400 on windows.

rogerclarkmelbourne, 2015-11-13:
Ah.

OK. I'm not using FTDI I've tried using Prolific and also Silicon Labs (CP2102) and both of these seem to support the speeds you have removed from the list.

I'm currently using 460800 with Prolific as it seems to be quite reliable. The Silicon labs CP2102 didn't seem to be able to operate much faster than 115200.

I have a FTDI, (it may be a clone), but I have not tried it.

It looks like the CP2102 doesnt support some of the speeds that the FTDI chip does e.g.

https://cdn.sparkfun.com/datasheets/BreakoutBoards/CP2102_v1.2.pdf

```
300, 600, 1200, 1800, 2400,
4000, 4800, 7200, 9600, 14400, 16000, 19200, 28800,
38400, 51200, 56000, 57600, 64000, 76800,
115200, 128000, 153600, 230400, 250000,
256000, 460800, 500000, 576000, 921600
```

(but 921600 is only for 5 bit transfers)

So I think if I want to use the CP2102 I need to make some more settings like 576000 ;-)

igrr, 2015-11-15:
> (but 921600 is only for 5 bit transfers)

That's funny, because NodeMCU 1.0 has CP2102, and 921600 works fine (at least on OS X).

Do you want me to make any changes to boards.txt? I.e. should I add 230400 and 460800 to the list for Windows?

rogerclarkmelbourne, 2015-11-15:
Ivan

I was just referring to the hardware spec for the CP2102, but its strange that it works for NodeMCU.

I've noticed that the esptool doesn't support 57600 on Windows, but I think it supports 230400. (I'll need to check 460800)

Either way, adding 230400 on Windows seems very low risk.  I've modified my own boards.txt file to include it, however at the moment, all uploads faster than 115200 are unreliable for me.

Uploads complete without errors, but the code is not actually updated.

I now add this to all my sketches

Serial.println(__DATE__ " " __TIME__);

So I can confirm the date and time of the compile when the sketch starts, to make sure it did really upload the current sketch.

tablatronix, 2016-03-05:
I use 921600 on nodemcu and ftdi on osx works great, cept the autoreset fails now and then. I think the code is using some bit duration measurment that fails at that speed sometimes.

Duality4Y, 2016-03-09:
higher baud don't always work, and i have noticed that this is in most cases due to wires not being shielded properly.

lrmoreno007, 2016-03-10:
I have Windows 7, ESP8266 Generic and FTDI FT232RL upload ok at all speeds. Only sometime I must hold GPIO0 with GND, but I think it's problem of FTDI drivers.

With NodeMCU Amica v.1.0 (ESP-12E  CP2102):
256000 upload OK, run OK.
512000 upload OK, but sometime show some errors in IDE. Always run OK (inclusive when errors in IDE).
921600 upload OK without problems in Arduino IDE, but never run OK:
ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1264, room 16 
tail 0
chksum 0x42
csum 0x42
~ld
e:
 ets Jan  8 2013,rst cause:3, boot mode:(3,6)

ets_main.c

devyte, 2017-11-21:
@d-a-v is this somehow addressed with the boards.txt generator?

d-a-v, 2017-11-23:
We can make the generator flexible.
It currently reflects the same serial conf as before (some speeds for windows only, some others for osx/linux only). This whole serial speeds and stability question might be only a question of driver stability and suitable hardware setup according to the speed (wires length etc...).
What can be done in the generator is optionally offering all serial speeds for all OSes, so the user can easily test and report.

tablatronix, 2017-11-23:
I vote for spending dev resources on a package ini file so users can customize baud speed menus, there are too many variables, sometimes the bottleneck is the transistors use for the autoreset mechanism.

d-a-v, 2017-11-24:
@tablatronix granted #3722

d-a-v, 2017-12-13:
@igrr since this depends now on #3722, should milestone be changed to 2.5.0 ?

devyte, 2017-12-17:
Pushing back to 2.5.0 due to dependency.

d-a-v, 2018-01-08:
@tablatronix can you try the now merged board generator and customize your baud rates ?

d-a-v, 2018-09-03:
Closing due to age and now customizable baud rates.

closed 2018-09-03 by d-a-v