esp8266/Arduino #1015: Fix RF24 library

eliabieri, 2015-11-13:
When I compile the RF24 library I get this error:
```
In file included from /Users/eliabieri/Documents/Arduino/libraries/RF24/RF24.h:18:0,
                 from Color_Lamp.ino:5:
/Users/eliabieri/Documents/Arduino/libraries/RF24/RF24_config.h:119:0: warning: "PSTR" redefined [enabled by default]
   #define PSTR(x) (x)
 ^
In file included from /Users/eliabieri/Library/Arduino15/packages/esp8266/hardware/esp8266/1.6.5-1160-gef26c5f/cores/esp8266/WString.h:29:0,
                 from /Users/eliabieri/Library/Arduino15/packages/esp8266/hardware/esp8266/1.6.5-1160-gef26c5f/cores/esp8266/IPAddress.h:24,
                 from /Users/eliabieri/Library/Arduino15/packages/esp8266/hardware/esp8266/1.6.5-1160-gef26c5f/libraries/ESP8266WiFi/src/ESP8266WiFi.h:31,
                 from Color_Lamp.ino:1:
/Users/eliabieri/Library/Arduino15/packages/esp8266/hardware/esp8266/1.6.5-1160-gef26c5f/cores/esp8266/pgmspace.h:18:0: note: this is the location of the previous definition
 #define PSTR(s) (__extension__({static const char __c[] PROGMEM = (s); &__c[0];}))
 ^
In file included from /Users/eliabieri/Documents/Arduino/libraries/RF24/RF24.h:18:0,
                 from Color_Lamp.ino:5:
/Users/eliabieri/Documents/Arduino/libraries/RF24/RF24_config.h:123:0: warning: "strlen_P" redefined [enabled by default]
   #define strlen_P strlen  
 ^
In file included from /Users/eliabieri/Library/Arduino15/packages/esp8266/hardware/esp8266/1.6.5-1160-gef26c5f/cores/esp8266/WString.h:29:0,
                 from /Users/eliabieri/Library/Arduino15/packages/esp8266/hardware/esp8266/1.6.5-1160-gef26c5f/cores/esp8266/IPAddress.h:24,
                 from /Users/eliabieri/Library/Arduino15/packages/esp8266/hardware/esp8266/1.6.5-1160-gef26c5f/libraries/ESP8266WiFi/src/ESP8266WiFi.h:31,
                 from Color_Lamp.ino:1:
/Users/eliabieri/Library/Arduino15/packages/esp8266/hardware/esp8266/1.6.5-1160-gef26c5f/cores/esp8266/pgmspace.h:63:0: note: this is the location of the previous definition
 #define strlen_P(strP)          strnlen_P((strP), SIZE_IRRELEVANT)
 ^
In file included from /Users/eliabieri/Documents/Arduino/libraries/RF24/RF24.h:18:0,
                 from Color_Lamp.ino:5:
/Users/eliabieri/Documents/Arduino/libraries/RF24/RF24_config.h:124:0: warning: "PROGMEM" redefined [enabled by default]
   #define PROGMEM
 ^
In file included from /Users/eliabieri/Library/Arduino15/packages/esp8266/hardware/esp8266/1.6.5-1160-gef26c5f/cores/esp8266/WString.h:29:0,
                 from /Users/eliabieri/Library/Arduino15/packages/esp8266/hardware/esp8266/1.6.5-1160-gef26c5f/cores/esp8266/IPAddress.h:24,
                 from /Users/eliabieri/Library/Arduino15/packages/esp8266/hardware/esp8266/1.6.5-1160-gef26c5f/libraries/ESP8266WiFi/src/ESP8266WiFi.h:31,
                 from Color_Lamp.ino:1:
/Users/eliabieri/Library/Arduino15/packages/esp8266/hardware/esp8266/1.6.5-1160-gef26c5f/cores/esp8266/pgmspace.h:15:0: note: this is the location of the previous definition
 #define PROGMEM     ICACHE_RODATA_ATTR
 ^
In file included from /Users/eliabieri/Documents/Arduino/libraries/RF24/RF24.h:18:0,
                 from Color_Lamp.ino:5:
/Users/eliabieri/Documents/Arduino/libraries/RF24/RF24_config.h:125:0: warning: "pgm_read_byte" redefined [enabled by default]
   #define pgm_read_byte(addr) (*(const unsigned char *)(addr))
 ^
In file included from /Users/eliabieri/Library/Arduino15/packages/esp8266/hardware/esp8266/1.6.5-1160-gef26c5f/cores/esp8266/WString.h:29:0,
                 from /Users/eliabieri/Library/Arduino15/packages/esp8266/hardware/esp8266/1.6.5-1160-gef26c5f/cores/esp8266/IPAddress.h:24,
                 from /Users/eliabieri/Library/Arduino15/packages/esp8266/hardware/esp8266/1.6.5-1160-gef26c5f/libraries/ESP8266WiFi/src/ESP8266WiFi.h:31,
                 from Color_Lamp.ino:1:
/Users/eliabieri/Library/Arduino15/packages/esp8266/hardware/esp8266/1.6.5-1160-gef26c5f/cores/esp8266/pgmspace.h:77:0: note: this is the location of the previous definition
 #define pgm_read_byte(addr)                                              \
 ^
In file included from /Users/eliabieri/Documents/Arduino/libraries/RF24/RF24.h:18:0,
                 from Color_Lamp.ino:5:
/Users/eliabieri/Documents/Arduino/libraries/RF24/RF24_config.h:126:0: warning: "pgm_read_word" redefined [enabled by default]
   #define pgm_read_word(p) (*(p))
 ^
In file included from /Users/eliabieri/Library/Arduino15/packages/esp8266/hardware/esp8266/1.6.5-1160-gef26c5f/cores/esp8266/WString.h:29:0,
                 from /Users/eliabieri/Library/Arduino15/packages/esp8266/hardware/esp8266/1.6.5-1160-gef26c5f/cores/esp8266/IPAddress.h:24,
                 from /Users/eliabieri/Library/Arduino15/packages/esp8266/hardware/esp8266/1.6.5-1160-gef26c5f/libraries/ESP8266WiFi/src/ESP8266WiFi.h:31,
                 from Color_Lamp.ino:1:
/Users/eliabieri/Library/Arduino15/packages/esp8266/hardware/esp8266/1.6.5-1160-gef26c5f/cores/esp8266/pgmspace.h:86:0: note: this is the location of the previous definition
 #define pgm_read_word(addr)                                              \
 ^

```

me-no-dev, 2015-11-13:
replace #include "avr/progmem.h" with #include "progmem.h"

reaper7, 2015-11-13:
working version is available on my github too

igrr, 2015-11-15:
Closing as this seems to be resolved by @reaper7.

closed 2015-11-15 by igrr