esp8266/Arduino #1016: SPIFFS Error: esptool not found!

hallard, 2015-11-14:
Strange thing I never saw, today I tried to put on SPIFFS the content of "data" folder and when I select "ESP8266 Sketch Data upload" entry on tool menu of Arduino IDE (1.6.5) I've got 

`SPIFFS Error: esptool not found!`, but flash sketch works fine and with esptool (in blue below), so may be wrong tool in definition and should use SPIFFS tool ?

[image]

It was working last time I used it (may be a month ago before big build repo change)

wemos, 2015-11-15:
Try to copy  tools\esptool\esptool.exe to tools\

wemos, 2015-11-15:
I have same problem, I use "Using git version" install hardware package in README.md.

copy tools\esptool\esptool.exe to tools\
copy tools\mkspiffs\mkspiffs.exe to tools\

It's work for me.

hallard, 2015-11-15:
@wemos 
Thanks, that did the trick ;-)

By the way I ordered 5 Wimos Mini D1 boards, they look fine and I like the size ;-)
Will it be possible to have schematics of the main board (not the module with ESP8266) or is it private ?

igrr, 2015-11-15:
@hallard I suppose you could have been using an older release of ESP8266FS tool, which didn't take new tools layout into account.
Please check if https://github.com/esp8266/arduino-esp8266fs-plugin/releases/tag/0.1.3 works for you.

hallard, 2015-11-15:
@igrr 
Correct updating this file works also (of course I deleted the both files copied earlier to test)

Another trick worth putting into documentation. You're telling to put he jar file here
`<home_dir>/Arduino/tools/ESP8266FS/tool/esp8266fs.jar)` which is the classic way. But if you're like me, you have multiple version of Arduino IDE you need each time to copy files from each other.
To avoid this, there is another solution I'm using for long now is to setup my sketchbook location away from IDE setup like the following screen
[image]
Like this you can put tool into 
`<sketchbook_home>/tools/ESP8266FS/tool/esp8266fs.jar`
and this work with every version of IDE you installed. An more than this it works also for libraries and hardware folder
in my case I've got cloned esp8266 git repo into
```
s:\Skydrive\devt\Arduino\hardware\esp8266com\esp8266`
and other classics are in
`s:\Skydrive\devt\Arduino\libraries
s:\Skydrive\devt\Arduino\tools
```
Each time I test dev Arduino IDE (1.6.7 for example) or latest (1.6.6) I just need to setup my sketchbook folder, fine ;-)

closed 2015-11-30 by igrr

itsashishupadhyay, 2019-11-24:
> 
> 
> I have same problem, I use "Using git version" install hardware package in README.md.
> 
> ```
> copy tools\esptool\esptool.exe to tools\
> copy tools\mkspiffs\mkspiffs.exe to tools\
> ```
> 
> It's work for me.

works for me too on ESP32 Huzzah

sr3d, 2020-11-19:
On OSX, to fix this, install the esptool.py using `pip install esptool` or `pip3 install esptool`.  Restart Arduino IDE and the esptool will now be detected.