esp8266/Arduino #1017: Automatic reboot after wdt reset

SensorsIot, 2015-11-14:
My boards crash from time to time and I do not know why. After such crash the wd reset shows up (boot mode:(1,6)), but the board does not restart automatically. How can I make the boards reset/restart automatically after a wdt reset?

Links2004, 2015-11-14:
mode:(1,6) means bootloader mode, you need to add resistors like here:
https://github.com/esp8266/Arduino/blob/master/doc/boards.md#minimal-hardware-setup-for-bootloading-and-usage

SensorsIot, 2015-11-14:
Thank you for this info. I added all these resistors.
After knowing that the ESP sits in the "bootloader" I have to rephrase my question: How can I change its behavior that the board restarts the normal "setup()" routine after the watchdog was initiated? I wrote a recovery routine to my sketch which can handle this situation. When I reset it manually, it works. But this will not be possible if I deployed it ;-)

Links2004, 2015-11-14:
if you have added all the resistors, and you call ```ESP.restart();``` the bord will reboot to the "setup".
keep in mind if you connect other sensors / hardware the GPIO0 need to be HIGH and GPIO15 need to be LOW when you call ```ESP.restart();```

SensorsIot, 2015-11-14:
Thank you for your fast reply. I was not clear enough: When my board crashes, it shows the message wdt reset   mode:(1,6) and does nothing anymore until I reset it manually. With your explanation, this behavior is clear: The bootloader waits for something to happen. In this situation, my sketch has no control over the board anymore and I cannot issue the command ESP.restart(). Only manual intervention (ground the reset pin) helps to restart the sketch. But I would like to avoid this manual reset and do it automatically after the watchdog showed its message.

Links2004, 2015-11-14:
ok now i get it,
if the wdt reset  is triggert the board reboots,
when it reboots it is checking the state of GPIO0 and GPIO15 and GPIO2

| GPIO15 | GPIO0 | GPIO2 | Mode                             |
| ------ | ----- | ----- | -------------------------------- |
| 0V     | 0V    | 3.3V  | Uart Bootloader                  |
| 0V     | 3.3V  | 3.3V  | Boot sketch                      |
| 3.3V   | x     | x     | SDIO mode (not used for Arduino) |

you need to make sure that the pins in the right state when a wtd happens or you call ESP.restart()

better you add some delays to prevent the wtd from triggering.

igrr, 2015-11-14:
This requires some investigation. I have noticed that sometimes my nodemcu
enters bootloader after wdt reset. All the pull up resistors are in place,
so not sure what is happening. I'll take a look at this, but not sure when
:)

SensorsIot, 2015-11-14:
You were absolutely right! The problem is my GPIO0 pin. I pull it up with an Attiny85. I use this method to generate the necessary timing (together with REST).
I did a very small sketch to generate a WD reset:
void setup() {
  Serial.begin(115200);
  Serial.println("Begin");
}
void loop() {
  while (1 == 1);
}

Always after I do the uploading, GPIO0 starts to oscillate from 3.1 to around 2 volts (with 26 MHz!). And in this state, the ESP goes into bootloader mode and stays there:

Soft WDT reset
ctx: cont 
sp: 3ffe96c0 end: 3ffe9890 offset: 01b0

>>>stack>>>
3ffe9870:  00000000 00000000 3ffe98b4 4020186a  
3ffe9880:  00000000 00000000 3ffe8870 40100114  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(1,7)

 ets Jan  8 2013,rst cause:4, boot mode:(1,7)

wdt reset

 If I do a (manual) hard reset with everything exactly the same as before, this happens repetedly:

Begin

Soft WDT reset

ctx: cont 
sp: 3ffe96d0 end: 3ffe9890 offset: 01b0

>>>stack>>>
3ffe9880:  00000000 00000000 3ffe8870 40100114  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1264, room 16 
tail 0
chksum 0x42
csum 0x42
~ld

and GPIO0 does not oscillate at all.

So, I have to invetigate into my uploading logic.
Thank you for your help!

asetyde, 2015-11-14:
With mynodemcu 1.0 esp12e  after flash system is blocked and I must cut off power . After this every time I do reset system work perfect

SensorsIot, 2015-11-14:
An additional info: I added a small capacitor to GPIO0 (47nf to GND) and the level of the pin stays now above 3V. The same behavior appears. So, it might have  another source. The board is an ESP-7.
After the post of asetyde I checked the same coding with my nodemcu ESP12E board. It behaves correctly and resets repedetly. 
Mybe asetyde can check my sketch on his board. Then we would know more.

asetyde, 2015-11-14:
At now i can not test , but why use a while(1) in function loop ? It's same

SensorsIot, 2015-11-14:
You can use while (1) . You are right. It is the same.

asetyde, 2015-11-14:
ok but loop in a loop ?

SensorsIot, 2015-11-14:
without the while loop the ESP does not crash, at least not in my case. And I wanted to see if the module restarts into the setup() after a crash or if it stays in the bootloader loop (see descriptions above).

drmpf, 2015-11-14:
` I added a small capacitor to GPIO0 (47nf to GND) and the level of the pin stays now above 3V`
Since you are trying to keep GPIO0 high, try a 47nf or higher to 3.3v instead.

igrr, 2015-11-15:
Linking #793, which is pretty much the same issue.

avarachanvc, 2015-11-20:
I have the same issue (ESP.reset() causes wdt reset and hangs). Here are my test findings. I am using nodemcu board. GPIO0,GPIO0 are pulled up, and GPIO15 is pulled down. Test code
```
void setup() {
  Serial.begin(115200);
  Serial.println("");
  delay(1000);
  //wdt_disable(); No difference with this line active or not.
  
}

void loop() {
  Serial.print("Start");
  delay(3000);
  //ESP.reset();
  ESP.restart();//Same results as ESP.reset()
}
```
First time run:
```
 Start
 ets Jan  8 2013,rst cause:2, boot mode:(1,6)

 ets Jan  8 2013,rst cause:4, boot mode:(1,6)

wdt reset
```

And is stuck here.

If I repower the device, then Every run then on shows
```
 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x4010f000, len 1264, room 16 
tail 0
chksum 0x42
csum 0x42
~ld
```

So, something is happening after the first repowering? Because, after that it runs smooth. as expected.

Links2004, 2015-11-20:
```
 ets Jan  8 2013,rst cause:2, boot mode:(1,6) <-- reboot by reset pin    - Mode Uart
 ets Jan  8 2013,rst cause:4, boot mode:(1,6) <-- reboot by watchdog  -  Mode Uart

 ets Jan  8 2013,rst cause:2, boot mode:(3,7) <--  reboot by reset pin           - Mode Flash ( boot the sketch)
```

more info about the boot messages see here:
https://github.com/esp8266/Arduino/blob/master/doc/boards.md#boot-messages-and-modes

the boot mode looks in the first case, check the GPIO0 its is at 0V to get mode 1.

avarachanvc, 2015-11-21:
@Links2004 given GPIO0 is pulled up to 3.3V on my board and the fact that I need to press a button (from GPIO0 to GND) under normal circumstances  to program it, any clue why this is happening?

avarachanvc, 2015-11-21:
@Links2004 I did some additional digging with a scope to see what is happening. During programming, if you look at GPIO0, you can see a very high frequency Sin wave.
I have attached pictures. please take a look. 
[ESP WDT RESET and GPIO0 states on first run.pdf](https://github.com/esp8266/Arduino/files/40737/ESP.WDT.RESET.and.GPIO0.states.on.first.run.pdf)

drmpf, 2015-11-21:
I use 3K3 pullups, does lower values reduce the sine wave?

avarachanvc, 2015-11-21:
@drmpf I did not change the resistor, but I tried multiple caps from GPIO0 to GND (ie, across the switch as a simple debouncer), and that did bring the oscillation value down considerably. I tried 1uf, and 10pf.

drmpf, 2015-11-21:
Also a 0.1uF from VCC to GND, close to the ESP supply pins (actually mine is not that close)

avarachanvc, 2015-11-21:
Yes, thats there (10uF||0.1uF)

drmpf, 2015-11-21:
I am not have a problem, I think.  Occasionally I don't get a clean power up, but not often.
Another possibility is a local ham radio transmitter getting into the leads.  What is the frequency of the oscillation?

avarachanvc, 2015-11-21:
As per scope, 26.0003MHz

avarachanvc, 2015-11-21:
I am using the staging version. I think I should try the stable version and see if this goes away.

SensorsIot, 2015-11-21:
I made exactly the same observations as avarachanvc. Also the frequency I discovered was 26 MHz. The behavior sometimes even survived a few power downs. Neither small resistors nor capacitors helped.
I think, the ESP gets into a mode where the GPIO0 is used as and OUTPUT and does not "listen" anymore. This could explain this behavior.
I do not yet understand the circumstances how to get the modules "reliably" in this mode. I think, we should investigatein this direction. If we know hot to put the module in this mode, the guys understanding firmware would be better able to investigate...

Links2004, 2015-11-21:
I think this is coming from the boot loader that is hard coded in the silicon. 
may the output the Crystal frequency (26 MHz) during boot loader.

the first boot to the sketch is triggert by a command to the boot loader. 
https://github.com/igrr/esptool-ck/blob/master/espcomm/espcomm.c#L453

may we need to change the esptool to do a normal reset.

igrr, 2015-11-21:
@Links2004 as per @SensorsIot 's comment,
> The behavior sometimes even survived a few power downs.

So i don't think this is related to the state in which esptool leaves the chip.

igrr, 2015-11-21:
Looking at ESP8266_Pin_List_141105.xls, i see the following comment regarding GPIO0 pin:
> after reset, the default is function5 to export the clock

[image]

avarachanvc, 2015-11-21:
I have not seen a case where this behaviour continues after power reset. It is back to normal boot everytime.

asetyde, 2015-11-21:
With nodemcu 1.0 and new staging 2.0.x no problem .

Stable Yes but only after flash . No big problem

avarachanvc, 2015-11-22:
Any further thoughts on this? The issue is prevelant between flashing the code and the first power recycle. So, is the reboot after programming different from booting from power?

Test code 
```
void setup() {
  pinMode(2, OUTPUT);     // Initialize the BUILTIN_LED pin as an output
}

void loop() {
  while(!digitalRead(0)){    
              //pulling gpio0 to gnd for few seconds resets the device due to wdt reset.
  }
  digitalWrite(2, LOW);   // Turn the LED on (Note that LOW is the voltage level                                   
  delay(1000);                      // Wait for a second
  digitalWrite(2, HIGH);  // Turn the LED off by making the voltage HIGH
  delay(2000);                      // Wait for two seconds (to demonstrate the active low LED)
}
```

If I flash it, and repower, pulling

Links2004, 2015-11-22:
yes the "reboot" after flashing is a command send to the bootloader and not a real reboot by reset pin.
https://github.com/igrr/esptool-ck/blob/master/espcomm/espcomm.c#L453

avarachanvc, 2015-11-22:
Ok. Thinking more about it, it is **NOT** the reboot after flashing that is in question here, because reboot after flash works just fine. Device boots up, and runs good.  It is the 

reboot or reset or crash (whatever) **AFTER**

"reboot after flash" and **BEFORE**

power recycle

that goes to the stuck mode

igrr, 2015-11-22:
There is no reboot after flashing. Flashing leaves the device in a specific
state, which causes an issue during subsequent reset.

asetyde, 2015-11-23:
it seems be ok in 2.0.0 staging but , i think 2.0.0 has many big problem .

ktallevi, 2015-11-29:
Continuing discussion from #1107.

If I physically power cycle the ESP after flashing a sketch (both OTA and non-OTA), the ESP will restart OK via code ESP.restart();

If using the OTA bootloader, the sketch has to be power cycled before attempting an update... otherwise the following occurs after the update:

ets Jan 8 2013,rst cause:1, boot mode:(1,0)

ets Jan 8 2013,rst cause:4, boot mode:(1,0)

wdt reset

Using Olimex MOD-WIFI-ESP8266-DEV board.

me-no-dev, 2015-11-29:
I can also confirm this with NodeMCU and standalone modules.
When you first flash using the serial port, the sketch is uploaded and the bootloader is told to start it.
So when the bootloader is told that after upload and subsequent reset is called, the bootloader just resets itself and not the sketch.
Manual reset does not trigger the bootloader and that is why it's working as expected from then on.
One possible solution I see is esptool to attempt to reset the board after upload and not just ump to the sketch. Second is to live with it and use OTA ;) And third is to find where the reset jump location is and change that to the sketch instead of the bootloader.

me-no-dev, 2015-11-29:
https://github.com/igrr/esptool-ck/blob/master/main.c#L71 changing to 1 does not help...

ktallevi, 2015-11-29:
I'm ok to live with it for the time being. I'll keep an eye on this thread for any fix.

asetyde, 2015-12-01:
i think it is something not set correct , when last week come 2.0.0 rc staging my board reboot correctly but with stable , that work so good not reboot .. something changed

zenmanenergy, 2015-12-10:
can someone smarter than me try this...

in eboot.c line 129:

ets_putc('c'); ets_putc('p'); ets_putc(':');
ets_wdt_disable();

change to:

ets_wdt_disable();
ets_putc('c'); ets_putc('p'); ets_putc(':');

I don't know how to debug this file, so I just started guessing. Moving the disable above the putc() methods made it work for me.

zenmanenergy, 2015-12-10:
I was getting the error after doing an httpUpdate. Links' mods to the Updater fixed my problems with the updater but I started get WDT errors upon reboot.

Links2004, 2015-12-10:
this problem here has noting to do with eboot.
here eboot is never started. 
```
 ets Jan  8 2013,rst cause:4, boot mode:(1,6)
```
the device reboot to the serial bootloader, for some unknown reason.

asetyde, 2016-01-06:
someone it test with 2.1.0 rc ? now work on nodemcu v1 ?

ry60003333, 2016-01-16:
I seem to be running into the same issue; If I call ESP.restart() after flashing the board over serial or an OTA update, the ESP chip hangs and goes into bootloader mode. Before rebooting I am setting GPIO 0 to HIGH and GPIO15 to LOW.

I am using the Adafruit HUZZAH ESP8266 board.

Any progress on this or an idea of when it will be fixed?

Links2004, 2016-01-16:
no progress right now, it is not easy to fix since we have no idea where the info is stored.
the best work around is to reset the ESP over the reset pin once after flashing.

ry60003333, 2016-01-18:
Thanks @Links2004; I'll do that for now and keep watch on this issue for a fix.

hvintik, 2016-01-24:
Try to put CH_PD pin directly to 3.3v without pull up resistor

redge76, 2016-01-27:
Hi,
For those who see this thread because of the whatchdog reset " ets Jan  8 2013,rst cause:4, boot mode:(1,6)" message while doing an OTA, the solution is to use the staging version of esp8266-arduino.
If you use platformio, check this https://github.com/platformio/platformio/issues/401#issuecomment-166958952

Landixus, 2016-03-16:
Same with Arduino IDE 1.6.6 and updated libraries:
after reset/ powercycle:

`Exception (0):
epc1=0x4021479d epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

ctx: sys 
sp: 3ffffd80 end: 3fffffb0 offset: 01a0

>>>stack>>>
3fffff20:  40209126 00000000 00000022 00000000  
3fffff30:  4020828d 00000000 4020b519 3ffec6a0  
3fffff40:  3fff0a34 40211c77 3ffea7ac 3ffea7b8  
3fffff50:  3ffea7b8 0000008c 00000000 00000022  
3fffff60:  00000002 0000001a 402070f3 3ffec6a0  
3fffff70:  3ffea7ac 3fffdcc0 3ffe8d40 3ffe8d40  
3fffff80:  00000080 3ffec6a0 3fffdab0 00000000  
3fffff90:  40206a1b 3fffdab0 00000000 402035ef  
3fffffa0:  3ffe8d40 40000f49 3fffdab0 40000f49  
<<<stack<<<

 ets Jan  8 2013,rst cause:4, boot mode:(3,6)

wdt reset
load 0x4010f000, len 1264, room 16 
tail 0
chksum 0x42
csum 0x42
~ld
Connecting to AP..
`
 I use a nodemcu 1.0 with esp8266 12E

with simple sketch:

`#include <ESP8266WiFi.h>
#include "EIoTCloudRestApi.h"

// EasyIoT Cloud definitions - change EIOT_CLOUD_INSTANCE_PARAM_ID
#define EIOT_CLOUD_INSTANCE_PARAM_ID    "xxxxxxxxxxxxxxxxxxxxxx"

#define INPUT_PIN        4

EIoTCloudRestApi eiotcloud;
bool oldInputState;

void setup() {
  Serial.begin(115200);
  eiotcloud.begin();

  pinMode(INPUT_PIN, INPUT_PULLUP);
    
  oldInputState = !digitalRead(INPUT_PIN);
//  delay(5000);
}

void loop() {
  int inputState = digitalRead(INPUT_PIN);;  
  
  if (inputState != oldInputState)
  {
    eiotcloud.sendParameter(EIOT_CLOUD_INSTANCE_PARAM_ID, (inputState == 1));
    oldInputState = inputState;
    delay(1000);
  }
}`

My Hardware setup is to feed the PIR with 3,3Volt from nodeMCU, (PIR have 3,3 volt input).
PIR output is 3,3Volt to GPIO 4 (so i not feed the GPIO 0,2,15).

Powersupply serve 5.1V to USB Port of the NodeMCU and have 2.5A

I test this on 3 NodeMCU also with 3 PIR.
Did not have a solution yet.

igrr, 2016-03-16:
@Landixus What you described is not the same as the issue discussed here. The issue discussed here is about ESP.restart() not performing what it should if the chip has not been reset after serial upload.

Landixus, 2016-03-17:
@igrr ahh sorry. i experiment much days with the nodeMCU, and i was happy to this first place that looks like "it s my error".
Sorry, i investigate further.

igrr, 2016-03-24:
It turns out that the bootloader doesn't sample GPIO pins to determine boot mode at startup. Instead, it reads high 16 bits of GPIO_IN register (0x6000318). In ESP8266_GPIO_Register_141105.xls, these 16 bits are referred to as follows:
```
GPIO_STRAPPING	[31:16]		The values of the strapping pins.
```
So it seems that the values of GPIO_IN are copied into GPIO_STRAPPING on physical reset. When we upload the sketch, this register gets loaded with the value indicating GPIO0 low. When any software reset happens (ESP.restart(), watchdog reset, etc), bootloader checks this strapping pin register and goes into UART mode, again. This register appears to be read-only, unfortunately. I didn't find a way to re-load this strapping pin values register.

It seems to me that the only workaround for this issue is to make esptool perform physical reset after uploading the sketch (via DTR/RTS pins).

davidgraeff, 2016-03-24:
Awesome finding. Physical reset isn't a solution for over the air updates of course. Have you contacted Espressif regarding GPIO_STRAPPING bits? They provide a cloud based over the air update solution so should face the same problem while soft reseting.

igrr, 2016-03-24:
This issue affects the device only immediately after the serial upload. I doubt this is an issue for any real-world application — you will certainly power the device down and up at least once after programming it. After you perform physical reset (e.g. power-on reset), the strapping register will be in correct state, and soft reset will work fine.

dheerajdake, 2016-06-03:
@igrr Hello. I am able to perform OTA for the ESP8266 using the HttpUpdate. But the ESP-12E module stays in the bootloader mode and doesn't exit out of that mode. If i reset the device manually, it works fine. I am using the ESP8266 as a standalone. How can i do a software reset? Code after the OTA update doesn't execute. So, ESP.restart() doesn't work.

igrr, 2016-06-03:
The two comments above actually provide an answer to your question. The only time software reset doesn't work is immediately following a serial upload. If you reset the module physically after serial upload and place it into normal (non-bootloader) mode, software reset will work.

dheerajdake, 2016-06-03:
@igrr But i am performing an OTA, i don't have access to the module. I can't press the reset button. How is the module able to reset itself when i do serial upload using the USB cable? I want that functionality in OTA. Is there a way to do it?

[70 more replies]

PR #5433 (2018-12-04): Change to hard reset after flashing
Fixes #1017 
Add flag to esptool to do a hard reset after flashing.
closed 2019-10-15 by devyte