esp8266/Arduino #1019: RTS level

corna, 2015-11-14:
Hi
I've wired my ESP-12 + adapter board (like [this one](http://eud.dx.com/product/esp-12-esp8266-serial-wifi-wireless-module-w-pcb-antenna-adapter-board-for-arduino-raspberry-pi-844379296#.Vkc6vmTiveQ)) following [this](https://github.com/esp8266/Arduino/blob/master/doc/ESP_to_serial.png) schematic but I kept receiving the "warning: espcomm_sync failed" error.
If I disconnect the RTS and DTR pins and I manually put the ESP8266 in the programming mode (GPIO0 low, low pulse on REST), the programming works.
I've analyzed the waveforms using an oscilloscope and this is the result (yellow = RTS, blue = TX of the PC)

[image]

Why is the ESP kept resetted during the transfer? Is it normal?
I'm on Debian Testing and I'm using the stable (Jul 23, 2015) version of Arduino-esp8266, Arduino 1.6.5-r5. My hardware is a ESP-12 board, a CH340G adapter and some MOSFETs as level adapters.
Thanks

Links2004, 2015-11-14:
no this is not normal, the  RTS and DTR stuff is done at the beginning, 
during the data transmission reset need to be high.

how have you wired the MOSFETs?
may there is a inversion in the signal.

corna, 2015-11-14:
I've connected the CH340G directly to the oscilloscope (as you can see the high level is at 5V)
However the MOSFETs are in a non-inverting configuration

[image]

Links2004, 2015-11-14:
the schematic looks good.
what have you selected as board in the IDE generic or nodeMCU?

corna, 2015-11-14:
Generic

Links2004, 2015-11-14:
on a FTDI it look like this:

Reset:
[image]

Data:
[image]

tested on latest git with arduino 1.6.5
can you use something other then the CH430?

corna, 2015-11-14:
I've tried with a CP2102 and it works perfectly
Here's the output of the oscilloscope (without the ESP8266, yellow = RTS, blue = TX)

[image]

So now the question is: why doesn't it work with the CH340G?
I've also tried with the git version of ESP8266-arduino with Arduino 1.6.6, but the results are the same

Links2004, 2015-11-15:
is DTR working normal?
if yes you can try to add a 10uF capacitor in the RTS line then you shut only get short pulses on the ESP.

but I have no idea why the CH340 make this strange thing.
may this a other reason for the transistors in the nodeMCU schematic.

[image]

corna, 2015-11-15:
I've tried the CH340 on windows and it works
Where can I find the code of esptool?

Links2004, 2015-11-15:
there are two esptools one is based on python the other in C (the C version is included in the IDE)

esptool.py https://github.com/Links2004/esptool
esptool.exe https://github.com/igrr/esptool-ck

hallard, 2015-11-17:
I've just wrote an article on simple auto-reset with FTDI (no transistors), it will not solve your problem but could be worth [reading](http://hallard.me/esp8266-autoreset/) it

devyte, 2017-10-19:
Closing per #3655 .

closed 2017-10-19 by devyte