esp8266/Arduino #1020: Crashes and libraries that use interrupts

adrionics, 2015-11-14:
I wanted to let others know that if you use a library that uses interrupts, those interrupt events may cause your ESP to crash if you call WiFi or WiFiClient functions and an interrupt happens while the call is in progress.

I experienced this with the excellent SoftwareSerial port from here: https://github.com/plerup/espsoftwareserial.

The solution was to add functions to the library to disable the interrupts and then call that disable function before calling the WiFi and Client functions, then re-enable interrupts after.

Hope this helps some who may be experiencing crashes that may be hard to stop or explain!

me-no-dev, 2015-11-14:
maybe if the library starts to use our attachInterrupt/detachInterrupt routines, then it might work.

adrionics, 2015-11-14:
Thanks for the tip, I will look at that.

igrr, 2015-11-15:
I have taken a look at this library and couldn't find any obvious reason why it would have to crash, especially during WiFi events. This needs some investigation, perhaps @plerup can give some pointers.

kdsoo, 2015-11-17:
I'm having similar symptom not with some specific library but attachInterrupt().
First I suspected #797 issue but problem still remains after I changed to staged branch.
Strange thing is when I get rid of whole code except attachinterrupt() and interrupt handler callback function, the problem disappears.
The extra code I get rid of is only about esp8266WebServer server(80)... server.handleClient() and EEPROM code which is only called on http request.
esp node crashes and reboot when external interrupt happens.

adrionics, 2015-11-17:
I had to disable the interrupts while processing WebServer requests too.

plerup, 2015-11-17:
@adrionics What baud rate are you using?

adrionics, 2015-11-17:
9600 or 2400, both have the problem.

kdsoo, 2015-11-18:
I tried disabling interrupt while handling http request and enabling back afterward but still crashing.
I tried like follows..
......

const int PIR = 12;

void loop(void) {
    disableInterrupt();
    server.handleClient();
    enableInterrupt();
}

void enableInterrupt() {
    attachInterrupt(PIR, processPresence, RISING);
}

void disableInterrupt() {
    detachInterrupt(PIR);
}

void processPresence() {
    digitalWrite(13, 1);
}

....

adrionics, 2015-11-18:
I fixed it by adding these two functions to the ESP SoftwareSerial library:
void SoftwareSerial::disable() {
   ETS_GPIO_INTR_DISABLE();
}
void SoftwareSerial::enable() {
   ETS_GPIO_INTR_ENABLE();
}

Then, call them before and after.

kdsoo, 2015-11-18:
I found a glitch in my code. I forgot to comment out a single debugging message using softwareserial....
anyway it seems to be working fine with @adrionics 's workaround for now.
cheers!

plerup, 2015-11-18:
If it helps just to disable the interrupts then there must be something triggering them when you get the crash. Maybe noise on the used rx pin.

As I stated in the comment in the source code the interrupt seems to trigger on any change on the pin although GPIO_PIN_INTR_NEGEDGE has been set.

I assume it's not real serial traffic in your case because then you would have missed it by disabling the interrupts.

adrionics, 2015-11-18:
plerup, I suspect it may be noise or real serial traffic that triggers the crash - as you say, any change on the pin. I sometimes do miss serial packets while the interrupt is disabled, but luckily in my case this does not matter.

I believe that if an interrupt is triggered for any reason, and the device is in the process of handling some routine like WiFi.begin, it crashes.

adrionics, 2015-11-18:
PS, it becomes more apparent with a fast stream of serial data. If the serial data is at a low rate, you may need to wait for an exact "while the WiFi / server" functions are called interrupt crash.

supersjimmie, 2015-12-15:
I too can confirm this issue. My code keeps crashing during wificlient sessions with SofwareSerial enabled. After using @adrionics disable/enable functions it seems to be working fine.  

I will now try to turn all around, by creating the SoftwareSerial mySerial(..) within the function where I will read the data, instead of creating it globally.  That way a softwareserial won't even exist during any wificlient operation.  (I can affort missing serial data during other functions)

adrionics, 2016-02-25:
@plerup has a new version of the library that seems to work really well and incorporates more features of the original like inverted serial and multiple ports. The enable / disable function seems to fix the interrupt problems. Well done @plerup!!

closed 2016-02-27 by igrr

chakri1vr1, 2017-01-13:
@adrionics You sir, are a life saver !!

Was using a interrupt based power measurement on ESP while making HTTP requests. It used to crash randomly and was unable to debug. Boy oh boy did this post come handy !! Switched to non-interrupt based power measurement and its working fine now :)

dforco, 2017-03-16:
Im having similar problem, using WIFI_AP_STA mode, WebServer on port 80, and simultaneous connecting to a local network. When enabling a timer 0 or 1 interrupt to blink a led, ESP start to crash randomly.

gcgallo, 2017-05-27:
Are you using the generic C wifi functions or the Arduino ESP8266WiFi header? I had a dreadful problem where my ESP8266 kept crashing on an interrupt. The interrupt was triggering a sequence of wifi, webserver, and dnsserver setup functions. At first I assumed it was just something with the interrupt. However, after narrowing down the possibilities, I discovered it was caused by mix and matching generic C wifi functions and Arduino WiFi methods. I had something like this:
```
//THIS IS WRONG
wifi_station_disconnect();
wifi_set_opmode(WIFI_AP);
WiFi.softAPConfig(apIP, apIP, IPAddress(255, 255, 255, 0));
WiFi.softAP(ap);
```
instead of something like this:
```
//THIS IS RIGHT
WiFi.disconnect();
WiFi.mode(WIFI_AP);
WiFi.softAPConfig(apIP, apIP, IPAddress(255, 255, 255, 0));
WiFi.softAP(ap);
```
Hope this helps some people!

faraonc, 2017-12-06:
AltSoftSerial solved the problem for me.

I initially had SoftwareSerial and my AJAX requests would get errors and disrupts Serial communications.

https://www.pjrc.com/teensy/td_libs_AltSoftSerial.html