esp8266/Arduino #1021: OTA update from github url

mmiscool, 2015-11-14:
Hello,

Git hub seems to do every thing over https and i was hoping to use the ota functionality to download from there. Is there a way to make ota work over https or a methos to get a regular http download from git hub?

```
        t_httpUpdate_return  ret = ESPhttpUpdate.update("github.com", 80, "/esp8266/Basic/raw/master/Flasher/Build/4M/ESP8266Basic.cpp.bin");
        
        switch (ret) {
          case HTTP_UPDATE_FAILD:
            Serial.println("HTTP_UPDATE_FAILD");
            break;

          case HTTP_UPDATE_NO_UPDATES:
            Serial.println("HTTP_UPDATE_NO_UPDATES");
            break;

          case HTTP_UPDATE_OK:
            Serial.println("HTTP_UPDATE_OK");
            break;
        }
```

sticilface, 2015-11-14:
I think the short answer is no.  
I was trying to download an 18Kb file from github, and it failed, @igrr debugged it failed as the TLS fragment size is too large.  I reckon OTA is a no go.  Something i wanted to try to do too.

Links2004, 2015-11-14:
github.io shut work it allow http requests. 

more info here:
https://pages.github.com/

igrr, 2015-11-15:
@mmiscool yup, pretty much what @sticilface said.
My suggestion is to either:
- set up TLS proxy (i.e. Apache Traffic Server) and configure it for smaller fragment size
- host a file with SHA sum + link to you OTA binary on github, and then download OTA binary over plain HTTP. Verify SHA sum as you are downloading.
Sorry but i don't think we can do anything about this issue given the RAM available to us on the ESP8266. Also unfortunately most TLS servers don't support TLS fragment size negotiation (that's an extension to TLS protocol), so we can't make them switch to smaller fragment sizes.

igrr, 2015-11-17:
Okay, spoke too soon.
I've improved TLS library so now it can handle full 16k fragments.
Here's an example which does OTA over HTTPS from github:
https://gist.github.com/igrr/24dd2138e9c8a7daa1b4

mmiscool, 2015-11-17:
Just tested this. 

Turned the code in to  function. Minimal modifications to remove the wifi connecting stuff as that is already handled in the  sketch.

Added this function.

'''
const char* host = "raw.githubusercontent.com";
const int httpsPort = 443;

// Use web browser to view and copy
// SHA1 fingerprint of the certificate
const char* fingerprint = "B0 74 BB EF 10 C2 DD 70 89 C8 EA 58 A2 F9 E1 41 00 D3 38 82";

const char* url = "/esp8266/Basic/master/Flasher/Build/4M/ESP8266Basic.cpp.bin";

bool BasicOTAupgrade()
{
  configTime(3 * 3600, 0, "pool.ntp.org");

  // Use WiFiClientSecure class to create TLS connection
  WiFiClientSecure client;
  Serial.print("connecting to ");
  Serial.println(host);
  if (!client.connect(host, httpsPort)) {
    Serial.println("connection failed");
    return 0;
  }

  if (client.verify(fingerprint, host)) {
    Serial.println("certificate matches");
  } else {
    Serial.println("certificate doesn't match");
    return 0;
  }

  Serial.print("Starting OTA from: ");
  Serial.println(url);

  auto ret = ESPhttpUpdate.update(client, host, url);
  // if successful, ESP will restart
  Serial.println("update failed");
  Serial.println((int) ret);
}
'''

I get the following output on the serial terminal.

igrr, 2015-11-17:
Looks like you are running out of RAM. Check with nothing except this function in your sketch (i.e. no SPIFFS, no web server).

Links2004, 2015-11-27:
check out the new API of ESPhttpUpdate
       
now it shut work with one line like this:
ESPhttpUpdate.update("https://github.com/esp8266/Basic/raw/master/Flasher/Build/4M/ESP8266Basic.cpp.bin");      

https://github.com/Links2004/Arduino/blob/httpClient/libraries/ESP8266httpUpdate/examples/httpUpdate/httpUpdate.ino

closed 2016-02-27 by igrr