esp8266/Arduino #1026: OTA update successfull but device runs the old program.

EUA, 2015-11-15:
Using ArduinoOTA,
I don't know what is happening actually, but my device run old program after proper OTA session.
It doesn't happen easy. But when it start to show such a behavior, you can't update it via OTA anymore. Need to remove device from board and flash via serial port.

I was changed EEPROM of my ESP11 device with 8MB flash (32Mbit). I suspect this issue arise due the EEPROM that I installed but after changing another flash with 4MB size, it still could show such a behaviour after some flashing session.

Can you guess what could be results such a problem?

vicatcu, 2015-11-16:
Not sure if this is your problem but 8MB == 64Mbit (not 32Mbit).

EUA, 2015-11-16:
Yes you are right, but problem is not related with it. Using 1M+64k SPIFFS flash size...

I enabled OTA_DEBUG and trying to reproduce same behavior.

EUA, 2015-11-17:
It happen again. Here is the debug log.
It reports write is successful but it is not.

```
scandone
f 0, scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 3
pm open phy_2,type:2 0 0
cnt 

connected with DeathNet High, channel 1
dhcp client start...
ip:192.168.1.109,mask:255.255.255.0,gw:192.168.1.1

WiFi connected
IP address: 192.168.1.109
OTA server at: esp8266-ElevatorCALL.local:8266
sleep enable,type: 2
Ready
Update Start: ip:192.168.1.2, port:30836, size:301776, md5:fb1f34ad4d87e7540752be68a2f52c6c
sleep disable
Update Success
Rebooting...
state: 5 -> 0 (0)
rm 0
pm close 7 0 0/8566135
del if0
usl
sul 0 0
�@�rjrA(!�S�u�К��XJ�~��*!�UI����I*E,�-1��Ъ¡��A�!��n�*E"��I�A,T[M�vA,T[�`��,p*�scandone
f 0, scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 3
pm open phy_2,type:2 0 0
cnt 

connected with DeathNet High, channel 1
dhcp client start...
ip:192.168.1.109,mask:255.255.255.0,gw:192.168.1.1

WiFi connected
IP address: 192.168.1.109
OTA server at: esp8266-ElevatorCALL.local:8266
sleep enable,type: 2
Ready
```

%100 sure about second updated scheme is different than this.

Also I found that, If I remove eeprom to another esp8266, you cannot program it via OTA also.
But you can program using serial interface.

igrr, 2015-11-17:
Wow, this garbage during reset looks weird. Normally when ESP8266 restarts during OTA I see all messages from the bootloader printed at 115200 (same baud rate as the sketch itself).

Update actually happens after reset. Bootloader gets a command via RTC memory, and copies new firmware in place of the old one. If RTC memory is not preserved during reset, bootloader will not perform update.

It would be great to figure out why the ESP is behaving like it is going through hard reset (as opposed to restart).

EUA, 2015-11-22:
Reset garbage due issue #1068 .