esp8266/Arduino #1028: Lost of SoftAP binding when starting a new STA binding

mytrain, 2015-11-16:
My application is using both SoftAP and STA as following:
-Smartphone connected via SoftAP to ESP8266
-Then each 2 minutres ESP connects using STA to my home WiFi router.

This issue is that conneting to STA kick out the Smartphone form the SoftAP as show in system debug

f -180,
scandone
state: 0 -> 2 (b0)
station: 7c:7d:3d:bb:73:e3 leave, AID = 1    <<<< THIS SHOULD NOT HAPPEN
rm 1
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 5
cnt 
chg_B:-120

Any tips to solve the issue ?

Used version : 1.6.5-1160-gef26c5f

igrr, 2015-11-16:
@Links2004 gave an explanation of this issue a few days back: https://github.com/esp8266/Arduino/issues/119#issuecomment-155856654
(i think previous comment down that link describes the issue similar to what you are seing).

mytrain, 2015-11-17:
@igrr 
Thanks for the explanation. I assume that this is a hardware constraint and there is no change that ESPRESSIF is solving the issue one day.
I'll revisit my design to find a work around
.

closed 2015-11-17 by igrr